fl4p/batmon-ha #106: Jk Jk-b2a24s20p disconnection issues..

Catchmydrift, 2023-05-27:
Hi, having connection issues with JK bmses 2 x Jk-b2a24s20p.     They connect initially, but for only about 1 minute, then disconnect.  They will reconnect a few minutes later, then the cycle repeats..    Any clues from the code below? 

Thanks 👍

(it was all working fine untill i had to upgrade home assistant due to putting it onto ssd instead of sd card..  Wasnt ever going to upgrade, but was forced too..) 

19:08:15 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.66
19:08:15 INFO [bt] BMS JKBt(C8:47:8C:E5:AB:CB) keep alive enabled
19:08:15 INFO [bt] BMS JKBt(C8:47:8C:EA:FF:1E) keep alive enabled
19:08:15 INFO [main] connecting mqtt mqtt_user@core-mosquitto
19:08:15 INFO [main] Fetching 2 BMS + 0 others serially, period=1.00s, keep_alive=True
19:08:15 INFO [sampling] connecting bms JKBt(C8:47:8C:E5:AB:CB)
19:08:15 INFO [main] mqtt watchdog loop started with timeout 900.0s
19:08:19 INFO [jikong] normal connect failed ([org.bluez.Error.Failed] le-connection-abort-by-local), connecting with scanner
19:08:46 ERROR [sampling] JK-B2A24S20P error: [org.bluez.Error.Failed] le-connection-abort-by-local
19:08:46 INFO [sampling] Bleak version 0.13.1a1
19:08:46 INFO [sampling] connecting bms JKBt(C8:47:8C:EA:FF:1E)
19:08:46 INFO [jikong] normal connect failed ([org.bluez.Error.Failed] le-connection-abort-by-local), connecting with scanner
19:09:23 ERROR [sampling] Bottom-Bank error: [org.bluez.Error.Failed] le-connection-abort-by-local
19:09:23 INFO [sampling] Bleak version 0.13.1a1
19:09:23 INFO [sampling] connecting bms JKBt(C8:47:8C:EA:FF:1E)
19:09:23 INFO [jikong] normal connect failed ([org.bluez.Error.Failed] le-connection-abort-by-local), connecting with scanner
19:09:57 INFO [sampling] connected bms JKBt(C8:47:8C:EA:FF:1E)!
19:09:57 INFO [sampling] Bottom-Bank subscribing for {'charge': True, 'discharge': True} switch change
19:09:57 INFO [mqtt_util] subscribe homeassistant/switch/Bottom-Bank/charge/set
19:09:57 INFO [mqtt_util] subscribe homeassistant/switch/Bottom-Bank/discharge/set
19:09:58 INFO [sampling] Bottom-Bank: BmsSampl(0.0%,U=0.0V,I=0.00A,P=0W,q=13500.4Ah/0,mos=0.0°C)
19:09:58 INFO [sampling] Bottom-Bank volt=3300,3299,3299,3298,3299,3298,3298,3299 temp=[0.0, 0.0]
19:09:58 INFO [sampling] JKBt(C8:47:8C:EA:FF:1E) times: connect=34.44s fetch=0.27s
19:09:58 INFO [sampling] connecting bms JKBt(C8:47:8C:E5:AB:CB)
19:09:59 INFO [jikong] normal connect failed ([org.bluez.Error.Failed] le-connection-abort-by-local), connecting with scanner
19:10:14 INFO [sampling] connected bms JKBt(C8:47:8C:E5:AB:CB)!
19:10:15 INFO [sampling] JK-B2A24S20P subscribing for {'charge': True, 'discharge': True} switch change
19:10:15 INFO [mqtt_util] subscribe homeassistant/switch/JK-B2A24S20P/charge/set
19:10:15 INFO [mqtt_util] subscribe homeassistant/switch/JK-B2A24S20P/discharge/set
19:10:15 INFO [sampling] JK-B2A24S20P: BmsSampl(96.0%,U=26.4V,I=-7.00A,P=-185W,q=268.7Ah/280,mos=23.1°C)
19:10:15 INFO [sampling] JK-B2A24S20P volt=3306,3303,3302,3303,3302,3301,3303,3305 temp=[21.7, 21.1]
19:10:15 INFO [sampling] JKBt(C8:47:8C:E5:AB:CB) times: connect=16.48s fetch=0.66s
19:10:16 INFO [sampling] Bottom-Bank: BmsSampl(0.0%,U=0.0V,I=0.00A,P=0W,q=13500.4Ah/0,mos=0.0°C)
19:10:16 INFO [sampling] Bottom-Bank volt=3299,3298,3299,3299,3299,3298,3298,3298 temp=[0.0, 0.0]
19:10:17 INFO [sampling] JK-B2A24S20P: BmsSampl(96.0%,U=26.4V,I=-7.21A,P=-191W,q=268.7Ah/280,mos=23.1°C)
19:10:17 INFO [sampling] JK-B2A24S20P volt=3306,3305,3302,3302,3303,3302,3303,3302 temp=[21.7, 21.1]
19:10:19 INFO [sampling] Bottom-Bank: BmsSampl(0.0%,U=0.0V,I=0.00A,P=0W,q=13500.4Ah/0,mos=0.0°C)
19:10:19 INFO [sampling] Bottom-Bank volt=3300,3299,3298,3299,3299,3298,3298,3299 temp=[0.0, 0.0]
19:10:20 INFO [sampling] JK-B2A24S20P: BmsSampl(96.0%,U=26.4V,I=-7.00A,P=-185W,q=268.7Ah/280,mos=23.1°C)
19:10:20 INFO [sampling] JK-B2A24S20P volt=3305,3305,3302,3302,3302,3302,3303,3303 temp=[21.7, 21.1]
19:10:22 INFO [sampling] Bottom-Bank: BmsSampl(0.0%,U=0.0V,I=0.00A,P=0W,q=13500.4Ah/0,mos=0.0°C)
19:10:22 INFO [sampling] Bottom-Bank volt=3300,3299,3300,3300,3300,3298,3298,3299 temp=[0.0, 0.0]
19:10:23 INFO [sampling] JK-B2A24S20P: BmsSampl(96.0%,U=26.4V,I=-7.00A,P=-185W,q=268.6Ah/280,mos=23.1°C)
19:10:23 INFO [sampling] JK-B2A24S20P volt=3306,3305,3303,3302,3302,3301,3303,3303 temp=[21.7, 21.1]
19:10:25 INFO [sampling] Bottom-Bank: BmsSampl(0.0%,U=0.0V,I=0.00A,P=0W,q=13500.4Ah/0,mos=0.0°C)
19:10:25 INFO [sampling] Bottom-Bank volt=3302,3300,3300,3299,3298,3295,3297,3298 temp=[0.0, 0.0]
19:10:28 INFO [sampling] JK-B2A24S20P: BmsSampl(95.9%,U=26.4V,I=-7.21A,P=-191W,q=268.6Ah/280,mos=23.1°C)
19:10:28 INFO [sampling] JK-B2A24S20P volt=3305,3303,3302,3302,3302,3301,3303,3303 temp=[21.7, 21.1]
19:10:28 INFO [sampling] JKBt(C8:47:8C:E5:AB:CB) times: connect=0.00s fetch=1.96s
19:10:29 INFO [sampling] Bottom-Bank: BmsSampl(0.0%,U=0.0V,I=0.00A,P=0W,q=13500.4Ah/0,mos=0.0°C)
19:10:29 INFO [sampling] Bottom-Bank volt=3300,3299,3299,3300,3299,3298,3298,3299 temp=[0.0, 0.0]
19:10:29 INFO [sampling] JKBt(C8:47:8C:EA:FF:1E) times: connect=0.00s fetch=1.22s
19:10:31 INFO [sampling] JK-B2A24S20P: BmsSampl(95.9%,U=26.4V,I=-7.21A,P=-191W,q=268.6Ah/280,mos=23.1°C)
19:10:31 INFO [sampling] JK-B2A24S20P volt=3306,3303,3301,3302,3303,3302,3303,3303 temp=[21.8, 21.1]
19:10:31 INFO [sampling] JKBt(C8:47:8C:E5:AB:CB) times: connect=0.00s fetch=1.08s
19:10:31 INFO [sampling] Bottom-Bank: BmsSampl(0.0%,U=0.0V,I=0.00A,P=0W,q=13500.4Ah/0,mos=0.0°C)
19:10:31 INFO [sampling] Bottom-Bank volt=3299,3299,3299,3299,3299,3297,3298,3299 temp=[0.0, 0.0]
19:10:39 WARNING [bt] BMS JKBt(C8:47:8C:E5:AB:CB) disconnected after 27.0s!
Traceback (most recent call last):
  File "/app/main.py", line 312, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
[... 48 lines cut ...]
19:12:25 INFO [bt] BT Device   JK_B2A24S20P   address=C8:47:8C:EA:FF:1E
19:12:25 INFO [bt] BT Device   30-78-CB-F1-1A-86   address=30:78:CB:F1:1A:86
19:12:25 INFO [bt] BT Device   BTG042   address=38:3B:26:5A:71:69
19:12:25 INFO [bt] BT Device   BTG077   address=38:3B:26:2E:B9:B6
19:12:25 INFO [bt] BT Device   4D-A7-38-B7-FC-24   address=4D:A7:38:B7:FC:24
19:12:25 INFO [bt] BT Device   JK-B2A24S20P   address=C8:47:8C:E5:AB:CB
19:12:25 INFO [bt] BT Device   E5-B6-E6-9F-D4-84   address=E5:B6:E6:9F:D4:84
19:12:25 INFO [bt] BT Device   6A-DE-E9-31-08-06   address=6A:DE:E9:31:08:06
19:12:25 INFO [bt] BT Device   58-8C-58-59-E6-56   address=58:8C:58:59:E6:56
19:12:25 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.66
19:12:25 INFO [bt] BMS JKBt(C8:47:8C:E5:AB:CB) keep alive enabled
19:12:25 INFO [bt] BMS JKBt(C8:47:8C:EA:FF:1E) keep alive enabled
19:12:25 INFO [main] connecting mqtt mqtt_user@core-mosquitto
19:12:25 INFO [main] Fetching 2 BMS + 0 others serially, period=1.00s, keep_alive=True
19:12:25 INFO [sampling] connecting bms JKBt(C8:47:8C:E5:AB:CB)
19:12:25 INFO [main] mqtt watchdog loop started with timeout 900.0s
19:12:27 INFO [jikong] normal connect failed ([org.bluez.Error.Failed] le-connection-abort-by-local), connecting with scanner
19:12:52 ERROR [sampling] JK-B2A24S20P error: [org.bluez.Error.Failed] le-connection-abort-by-local
19:12:52 INFO [sampling] Bleak version 0.13.1a1
19:12:52 INFO [sampling] connecting bms JKBt(C8:47:8C:EA:FF:1E)
19:12:54 INFO [jikong] normal connect failed ([org.bluez.Error.Failed] le-connection-abort-by-local), connecting with scanner
19:13:23 ERROR [sampling] Bottom-Bank error: [org.bluez.Error.Failed] le-connection-abort-by-local
19:13:23 INFO [sampling] Bleak version 0.13.1a1
19:13:23 INFO [sampling] connecting bms JKBt(C8:47:8C:E5:AB:CB)
19:13:28 INFO [jikong] normal connect failed ([org.bluez.Error.Failed] le-connection-abort-by-local), connecting with scanner

fl4p, 2023-08-20:
There is a bug which causes the batmon process to exit, specifically
```
await asyncio.wait(loops, return_when='FIRST_COMPLETED')
```
returns with global variable `shutdown` being False. I have no idea why this is happening on a disconnect

fl4p, 2023-08-20:
When BLE connecting is closed by disabling the bluetooth hardware, the process doesn't exit.
Seems to be related with bleak and/or asyncio. Very weird

Catchmydrift, 2023-08-20:
> When BLE connecting is closed by disabling the bluetooth hardware, the process doesn't exit. Seems to be related with bleak and/or asyncio. Very weird

I have two of the jk bmses, one will usually connect reasonably reliably, but the second one will never stay connected for very long unfortunately. The  bmses are the same model.

fl4p, 2023-10-20:
this shouldve been fixed in the meanwhile

closed 2023-10-20 by fl4p

Catchmydrift, 2023-10-20:
Unfortunately not fixed..  Very unlikely that either bms connects, and both
at the same time is even more unlikely, and when they do, one or both
always disconnect quickly after..