fl4p/batmon-ha #108: MQTT warnings: JSON result was not a dictionary

p-zander, 2023-06-01:
Disclaimer: I know very little about MQTT. We were just toying around to see if we could add the supervolt BMS to the addon, see here [Abraxotl/batmon-ha](https://github.com/Abraxotl/batmon-ha).

It is working so far, but needs a lot of testing and polish. Anyway we noticed a lot of warnings generated in the Homeassistant Core logs, checked with the "dummy" BMS type and that is flooding the log as well with messages like these:

```
2023-06-01 21:19:59.565 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on BMS/meter/total_energy (qos=0): b'0.1119'
2023-06-01 21:19:59.566 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-06-01 21:19:59.582 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on BMS/temperatures/1 (qos=0): b'22'
2023-06-01 21:19:59.582 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-06-01 21:21:19.912 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on BMS/soc/total_voltage (qos=0): b'14.23'
2023-06-01 21:21:19.913 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-06-01 21:21:19.957 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on BMS/soc/current (qos=0): b'0'
2023-06-01 21:21:19.959 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-06-01 21:21:19.997 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on BMS/soc/power (qos=0): b'0.0'
2023-06-01 21:21:20.001 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-06-01 21:21:20.052 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on BMS/soc/capacity (qos=0): b'150.0'
2023-06-01 21:21:20.053 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-06-01 21:21:20.064 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on BMS/mosfet_status/capacity_ah (qos=0): b'149.8'
2023-06-01 21:21:20.065 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-06-01 21:21:20.079 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on BMS/mosfet_status/temperature (qos=0): b'22'
2023-06-01 21:21:20.080 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
```

As I said, no idea why it would be expected that the payload is a dictionary. The warnings are generated here [homeassistant.components.mqtt.mixins@411](https://github.com/home-assistant/core/blob/2023.5.4/homeassistant/components/mqtt/mixins.py#L411). The mqtt_util.py is untouched, it might be a configuration issue or just an unnecessary warning.

Version info:
```
    "installation_type": "Home Assistant OS",
    "version": "2023.5.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.11",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.21-v8",
    "supervisor": "2023.04.1",
    "host_os": "Home Assistant OS 10.2",
    "docker_version": "23.0.6",
    "chassis": "embedded",
    "run_as_root": true
```

MQTT integration uses `paho-mqtt==1.6.1`

fl4p, 2023-06-05:
I fix this with the next release.

fl4p, 2023-06-05:
I've just pushed the fix here

p-zander, 2023-06-08:
Thanks! The warning didn't appear again since "json_attributes_topic" was removed from the discovery, so this should have done the trick. Hope it wasn't needed for anything else.

fl4p, 2023-06-08:
It should be fine without it. `json_attributes_topic` is only needed when the actual state value is nested in a JSON object. Batmon just sends raw values.