fl4p/batmon-ha #109: non more BT connection to 4 x JBD since a system update

ornolio, 2023-06-04:
Describe the issue here.
since last system update my 4 x JBD stopped to be connected by BT.
Anything to do with  Apparmor ?
Thank you for your help.

If you experience bluetooth connection problems, follow these steps before opening an issue:
https://github.com/fl4p/batmon-ha#troubleshooting

```
Paste log output between BEGIN and END:
BEGIN 
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
20:08:48 INFO [install_bleak] bleak 0.13.1a1 installed, want 0.20.2, running pip3 install bleak==0.20.2
Collecting bleak==0.20.2
  Downloading bleak-0.20.2-py3-none-any.whl (135 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 135.6/135.6 kB 233.3 kB/s eta 0:00:00
Collecting dbus-fast<2.0.0,>=1.83.0
  Downloading dbus_fast-1.86.0.tar.gz (65 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 66.0/66.0 kB 163.0 kB/s eta 0:00:00
  Installing build dependencies: started

Installing build dependencies: still running...
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting async-timeout<5,>=3.0.0
  Downloading async_timeout-4.0.2-py3-none-any.whl (5.8 kB)
Building wheels for collected packages: dbus-fast
  Building wheel for dbus-fast (pyproject.toml): started

END
```

ornolio, 2023-06-05:
Solved after removing Terminal & SSH 9.7.1 and Home Assistant Core 2023.5.4 changes.

closed 2023-06-05 by ornolio

fl4p, 2023-08-20:
Thanks for letting us know