fl4p/batmon-ha #110: Can not toggle charge or discharge switch

StormCh, 2023-06-06:
Hi I try to toggle the charge or discharge switch, but it is not working. With the App "https://web.e77code.com/xiaoxiang-bms/" it is working.
```
21:51:02 INFO [sampling] NightPower-Akku volt=3333,3333,3334,3335 temp=[23.2, 22.8, 22.7]
21:51:05 INFO [sampling] NightPower-Akku: BmsSampl(88.7%,U=13.3V,I=0.00A,P=0W,q=87.6Ah/99,mos=nan°C)
21:51:05 INFO [sampling] NightPower-Akku volt=3333,3333,3334,3335 temp=[23.2, 22.8, 22.8]
21:51:05 INFO [sampling] JbdBt(xxxxxxxx) times: connect=0.00s fetch=0.23s
21:51:08 INFO [sampling] NightPower-Akku: BmsSampl(88.7%,U=13.3V,I=0.00A,P=0W,q=87.6Ah/99,mos=nan°C)
21:51:08 INFO [sampling] NightPower-Akku volt=3333,3334,3334,3335 temp=[23.2, 22.8, 22.8]
21:51:11 INFO [sampling] NightPower-Akku: BmsSampl(88.7%,U=13.3V,I=0.00A,P=0W,q=87.6Ah/99,mos=nan°C)
21:51:11 INFO [sampling] NightPower-Akku volt=3333,3333,3334,3335 temp=[23.2, 22.8, 22.7]
21:51:14 INFO [sampling] NightPower-Akku: BmsSampl(88.7%,U=13.3V,I=0.00A,P=0W,q=87.6Ah/99,mos=nan°C)
21:51:14 INFO [sampling] NightPower-Akku volt=3333,3333,3334,3335 temp=[23.2, 22.9, 22.7]
21:51:17 INFO [sampling] NightPower-Akku: BmsSampl(88.7%,U=13.3V,I=0.00A,P=0W,q=87.6Ah/99,mos=nan°C)
21:51:17 INFO [sampling] NightPower-Akku volt=3333,3333,3334,3335 temp=[23.2, 22.8, 22.7]
21:51:20 INFO [sampling] NightPower-Akku: BmsSampl(88.7%,U=13.3V,I=0.00A,P=0W,q=87.6Ah/99,mos=nan°C)
21:51:20 INFO [sampling] NightPower-Akku volt=3333,3333,3334,3335 temp=[23.2, 22.8, 22.7]
21:51:23 INFO [sampling] NightPower-Akku: BmsSampl(88.7%,U=13.3V,I=0.00A,P=0W,q=87.6Ah/99,mos=nan°C)
21:51:23 INFO [sampling] NightPower-Akku volt=3333,3333,3334,3335 temp=[23.2, 22.8, 22.7]
21:51:26 INFO [sampling] NightPower-Akku: BmsSampl(88.7%,U=13.3V,I=0.00A,P=0W,q=87.6Ah/99,mos=nan°C)
21:51:26 INFO [sampling] NightPower-Akku volt=3333,3334,3334,3335 temp=[23.2, 22.8, 22.7]
21:51:29 INFO [sampling] NightPower-Akku: BmsSampl(88.7%,U=13.3V,I=0.00A,P=0W,q=87.6Ah/99,mos=nan°C)
21:51:29 INFO [sampling] NightPower-Akku volt=3333,3333,3334,3335 temp=[23.2, 22.8, 22.7]
21:51:32 INFO [sampling] NightPower-Akku: BmsSampl(88.7%,U=13.3V,I=0.00A,P=0W,q=87.6Ah/99,mos=nan°C)
21:51:32 INFO [sampling] NightPower-Akku volt=3333,3333,3334,3335 temp=[23.2, 22.8, 22.7]
21:51:35 INFO [sampling] NightPower-Akku: BmsSampl(88.7%,U=13.3V,I=0.00A,P=0W,q=87.6Ah/99,mos=nan°C)
21:51:35 INFO [sampling] NightPower-Akku volt=3333,3334,3334,3335 temp=[23.2, 22.8, 22.8]
21:51:38 INFO [sampling] NightPower-Akku: BmsSampl(88.7%,U=13.3V,I=0.00A,P=0W,q=87.6Ah/99,mos=nan°C)
21:51:38 INFO [sampling] NightPower-Akku volt=3333,3334,3334,3335 temp=[23.2, 22.8, 22.7]
21:51:39 INFO [mqtt_util] received msg homeassistant/switch/NightPower-Akku/discharge/set: ON
21:51:40 INFO [mqtt_util] Set NightPower-Akku discharge switch True
21:51:40 INFO [jbd] send switch msg: b'\xddZ\xe1\x02\x00\x00\xff\x1dw'
21:51:41 INFO [sampling] NightPower-Akku: BmsSampl(88.7%,U=13.3V,I=0.00A,P=0W,q=87.6Ah/99,mos=nan°C)
21:51:41 INFO [sampling] NightPower-Akku volt=3333,3333,3334,3335 temp=[23.2, 22.8, 22.8]
21:51:44 INFO [sampling] NightPower-Akku: BmsSampl(88.7%,U=13.3V,I=0.00A,P=0W,q=87.6Ah/99,mos=nan°C)
21:51:44 INFO [sampling] NightPower-Akku volt=3333,3333,3334,3335 temp=[23.2, 22.8, 22.8]
21:51:45 INFO [sampling] JbdBt(xxxxxxx) times: connect=0.00s fetch=0.08s
21:51:47 INFO [sampling] NightPower-Akku: BmsSampl(88.7%,U=13.3V,I=0.00A,P=0W,q=87.6Ah/99,mos=nan°C)
21:51:47 INFO [sampling] NightPower-Akku volt=3333,3334,3334,3335 temp=[23.2, 22.8, 22.7]
```

Sbellins, 2023-07-12:
I have the same issue, i cannot operate the Charge switch (i've not tested the discharge). When i click on it it show it deactivated on home assistant and a couple of seconds later it show it's on again, last week it worked as intend now it does this strange thing (same error in the log as @StormCh )

fl4p, 2023-12-19:
this shold be working now with JBD. re-open if needed

closed 2023-12-19 by fl4p