fl4p/batmon-ha #113: jk bms with faulty temperature

lucize, 2023-06-30:
seems that my bms doesn't show the temperature anymore and the code can't live without it, can this exception be handled correctly ?

```
2820408  ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised when updating state of sensor.280ah_temperatures_1, topic: '280ah/temperatures/1' with payload: b'nan'
2820664  Traceback (most recent call last):
2820838    File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 581, in state
2821143      numerical_value = int(value)
2821358                        ^^^^^^^^^^
2821609  ValueError: invalid literal for int() with base 10: 'nan'
2821770 
2821930  The above exception was the direct cause of the following exception:
2822070 
2822227  Traceback (most recent call last):
2822377    File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 270, in process_write_state_requests
2822561      entity.async_write_ha_state()
2822731    File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 590, in async_write_ha_state
2822955      self._async_write_ha_state()
2823190    File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 654, in _async_write_ha_state
2823352      state = self._stringify_state(available)
2823747              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2823925    File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 596, in _stringify_state
2824069      if (state := self.state) is None:
2824287                   ^^^^^^^^^^
2824431    File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 585, in state
2824590      raise ValueError(
2824798  ValueError: Sensor sensor.280ah_temperatures_1 has device class 'temperature', state class 'None' unit '°C' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: 'nan' (<class 'str'>)#033[0m
2829282  #033[31m2023-06-30 19:20:31.825 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised when updating state of sensor.280ah_temperatures_2, topic: '280ah/temperatures/2' with payload: b'nan'
2829514  Traceback (most recent call last):
2829671    File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 581, in state
2829829      numerical_value = int(value)
2829968                        ^^^^^^^^^^
2830110  ValueError: invalid literal for int() with base 10: 'nan'
2830263 
2830477  The above exception was the direct cause of the following exception:
2830628 
2830787  Traceback (most recent call last):
2830923    File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 270, in process_write_state_requests
2831063      entity.async_write_ha_state()
2831218    File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 590, in async_write_ha_state
2831372      self._async_write_ha_state()
2831557    File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 654, in _async_write_ha_state
2831741      state = self._stringify_state(available)
2831876              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2832027    File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 596, in _stringify_state
2832190      if (state := self.state) is None:
2832405                   ^^^^^^^^^^
2832673    File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 585, in state
2832849      raise ValueError(
2833073  ValueError: Sensor sensor.280ah_temperatures_2 has device class 'temperature', state class 'None' unit '°C' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: 'nan' (<class 'str'>)#033[0m
 homeassistant addon_2af0a32d_batmon[1508]: 19:20:38 WARNING [bt] BMS JKBt(xx) disconnected after 41.7s!

```

fl4p, 2023-07-28:
I'll fix this with the next update

fl4p, 2023-08-20:
Fixed with 4ddc3348ead1c95a912f9b2c6c8e03a77ec7e07a , please wait for the next release and re-open if issue persists.

closed 2023-08-20 by fl4p