fl4p/batmon-ha #114: WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary

kamilzierke, 2023-07-03:
As in the title; MQTT broker warnings about JSON results not being a dictionary ones.
Couple of houndred tousands a day. Had to enable DEBUG level of logs to be able to connect the error with Batmon and DalyBMS.

"WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary"

```
Exemplary Log from Batmon:
BEGIN 
12:38:35 INFO [sampling] DalyBt(C6:6C:09:05:02:EC) times: connect=0.00s fetch=1.20s
12:38:40 INFO [sampling] DalyBMS: BmsSampl(46.7%,U=55.4V,I=0.00A,P=0W,q=0.9Ah/2,mos=nan°C)
12:38:41 INFO [sampling] DalyBMS volt=3937,3972,3942,3982,3947,3984,3932,3977,3945,3974,3933,3985,3952,3982 temp=[26]
12:38:41 INFO [sampling] DalyBt(C6:6C:09:05:02:EC) times: connect=0.00s fetch=1.30s
12:38:46 INFO [sampling] DalyBMS: BmsSampl(46.7%,U=55.4V,I=0.00A,P=0W,q=0.9Ah/2,mos=nan°C)
12:38:47 INFO [sampling] DalyBMS volt=3928,3974,3944,3989,3942,3995,3932,3985,3934,3981,3930,3995,3945,3998 temp=[26]
12:38:47 INFO [sampling] DalyBt(C6:6C:09:05:02:EC) times: connect=0.00s fetch=1.20s
12:38:53 INFO [sampling] DalyBMS: BmsSampl(46.7%,U=55.5V,I=-0.60A,P=-33W,q=0.9Ah/2,mos=nan°C)
12:38:53 INFO [sampling] DalyBMS volt=3940,3972,3953,3977,3953,3981,3952,3975,3949,3976,3952,3990,3957,3987 temp=[26]
12:38:53 INFO [sampling] DalyBt(C6:6C:09:05:02:EC) times: connect=0.00s fetch=1.25s
12:38:59 INFO [sampling] DalyBMS: BmsSampl(46.7%,U=55.4V,I=0.00A,P=0W,q=0.9Ah/2,mos=nan°C)
12:39:00 INFO [sampling] DalyBMS volt=3927,3973,3934,3981,3941,3984,3925,3975,3938,3976,3926,3980,3942,3978 temp=[26]
12:39:00 INFO [sampling] DalyBt(C6:6C:09:05:02:EC) times: connect=0.00s fetch=1.30s
END
```
```
Exemplary Log from HA:

BEGIN 
2023-07-03 12:35:21.117 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/1 (qos=0): b'3.989'
2023-07-03 12:35:21.117 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:21.124 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/2 (qos=0): b'3.95'
2023-07-03 12:35:21.125 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:21.128 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/3 (qos=0): b'3.98'
2023-07-03 12:35:21.128 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:21.130 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/4 (qos=0): b'3.944'
2023-07-03 12:35:21.131 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:21.135 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/5 (qos=0): b'3.976'
2023-07-03 12:35:21.135 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:21.137 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/6 (qos=0): b'3.938'
2023-07-03 12:35:21.138 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:21.141 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/7 (qos=0): b'3.992'
2023-07-03 12:35:21.141 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:21.144 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/8 (qos=0): b'3.945'
2023-07-03 12:35:21.144 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:21.146 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/9 (qos=0): b'3.985'
2023-07-03 12:35:21.146 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:21.148 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/10 (qos=0): b'3.948'
2023-07-03 12:35:21.148 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:21.152 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/11 (qos=0): b'3.993'
2023-07-03 12:35:21.152 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:21.154 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/12 (qos=0): b'3.938'
2023-07-03 12:35:21.154 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:21.156 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/13 (qos=0): b'3.98'
2023-07-03 12:35:21.156 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:21.158 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/14 (qos=0): b'3.941'
2023-07-03 12:35:21.158 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:21.159 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/min (qos=0): b'3.938'
2023-07-03 12:35:21.160 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:21.162 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/min_index (qos=0): b'5'
2023-07-03 12:35:21.162 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:21.163 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/max (qos=0): b'3.993'
2023-07-03 12:35:21.163 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:21.164 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/delta (qos=0): b'0.055'
2023-07-03 12:35:21.164 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:21.165 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/average (qos=0): b'3.964'
2023-07-03 12:35:21.165 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:21.166 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/median (qos=0): b'3.963'
2023-07-03 12:35:21.166 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:21.513 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/temperatures/1 (qos=0): b'26'
2023-07-03 12:35:21.513 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:26.913 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/soc/total_voltage (qos=0): b'55.4'
[... 18 lines cut ...]
2023-07-03 12:35:27.397 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:27.402 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/11 (qos=0): b'3.991'
2023-07-03 12:35:27.402 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:27.404 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/12 (qos=0): b'3.918'
2023-07-03 12:35:27.405 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:27.408 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/13 (qos=0): b'3.969'
2023-07-03 12:35:27.408 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:27.411 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/14 (qos=0): b'3.924'
2023-07-03 12:35:27.411 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
[... cut ...]
2023-07-03 12:35:27.426 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/average (qos=0): b'3.957'
2023-07-03 12:35:27.426 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
2023-07-03 12:35:27.427 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on DalyBMS/cell_voltages/median (qos=0): b'3.9565'
2023-07-03 12:35:27.427 WARNING (MainThread) [homeassistant.components.mqtt.mixins] JSON result was not a dictionary
END
```

kamilzierke, 2023-07-18:
Any insights? :)

fl4p, 2023-07-24:
YOu can ignore the warnings. Will fix it with the next update

fl4p, 2023-07-25:
Fixed with 0.0.64

closed 2023-07-25 by fl4p