fl4p/batmon-ha #116: error

GRV0423, 2023-07-24:
Describe the issue here.
I'm getting this error; how can it be fix?
If you experience bluetooth connection problems, follow these steps before opening an issue:
https://github.com/fl4p/batmon-ha#troubleshooting

```
Paste log output between BEGIN and END:
BEGIN 
[28, 28]
01:00:53 INFO [sampling] batteryi: BmsSampl(45.0%,U=52.4V,I=7.70A,P=403W,q=117.0Ah/260,mos=nan°C)
01:00:53 ERROR [sampling] batteryii error: <class 'asyncio.exceptions.TimeoutError'>
01:00:53 INFO [sampling] batteryii bms debug data: bytearray(b'\x0c\xcd\x0c\xcb\x1e')
01:00:53 INFO [sampling] Bleak version 0.20.2
01:00:53 ERROR [main] Error (num 1, max 200) reading BMS: 
01:00:53 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.10/asyncio/tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/main.py", line 35, in fetch_loop
    await fn()
  File "/app/bmslib/sampling.py", line 67, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 153, in sample
    voltages = await bms.fetch_voltages()
  File "/app/bmslib/daly.py", line 205, in fetch_voltages
    resp = await self._q(0x95, num_responses=num_resp)
  File "/app/bmslib/daly.py", line 114, in _q
    sample = await self._fetch_futures.wait_for(command, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 50, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
  File "/usr/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
END
```

GRV0423, 2023-07-27:
after update still getting on error.
min and max index cell is off by 1
example if it says Cell_voltages min_index 3 but its actually cell 4
thank you

\[31, 30]
23:59:02 ERROR [sampling] batteryii error: timeout awaiting result 95, got 2/6 responses
23:59:02 INFO [sampling] batteryii bms debug data: bytearray(b'\x0c\xf0\x0c\xd2\xff')
23:59:02 INFO [sampling] Bleak version 0.20.2
23:59:02 ERROR [main] Error (num 1, max 200) reading BMS: timeout awaiting result 95, got 2/6 responses
23:59:02 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/bmslib/daly.py", line 114, in _q
    sample = await self._fetch_futures.wait_for(command, self.TIMEOUT)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/__init__.py", line 50, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/main.py", line 35, in fetch_loop
    await fn()
  File "/app/bmslib/sampling.py", line 69, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 158, in sample
    voltages = await bms.fetch_voltages()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/daly.py", line 205, in fetch_voltages
    resp = await self._q(0x95, num_responses=num_resp)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/daly.py", line 117, in _q
    raise TimeoutError(
TimeoutError: timeout awaiting result 95, got 2/6 responses

fl4p, 2023-07-28:
Which Daly BMS model is it?

GRV0423, 2023-07-28:
Its a Daly Bms 16S 150A with CAN & Fan.

fl4p, 2023-07-28:
Did it ever read any cell voltages?

GRV0423, 2023-07-28:
Yes it read all the cells voltage from 1-16, but on the cell index's min and max it start show 0-15, not 0-16 thats why its off by 1.

fl4p, 2023-07-30:
fixed with https://github.com/fl4p/batmon-ha/commit/7792f5c05413577a8f225426f3008215d9e2a177

closed 2023-07-30 by fl4p