fl4p/batmon-ha #117: Daly error 147 - Bluetooth seems to work, but then fails

wimg, 2023-07-24:
The Daly device is detected over Bluetooth, but then Daly error 147 occurs.

Any idea what might be wrong ? I've tried the various Bluetooth troubleshooting steps, but they seem to have no effect.

```
Paste log output between BEGIN and END:
BEGIN 
11:13:45 INFO [main] Power cycle bluetooth hardware
11:13:49 INFO [bt] BT Discovery:
11:13:54 INFO [bt] BT Device   DL-76670203023E   address=76:67:02:03:02:3E
11:13:54 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.66
11:13:54 INFO [bt] BMS DalyBt(76:67:02:03:02:3E) keep alive enabled
11:13:54 INFO [main] connecting mqtt @localhost

11:13:54 INFO [main] Fetching 1 BMS + 0 others serially, period=1.00s, keep_alive=True
11:13:54 INFO [sampling] connecting bms DalyBt(76:67:02:03:02:3E)
11:13:56 INFO [daly] found rx uuid to be working: 17 (tx 15, sx 48)
11:13:56 INFO [sampling] connected bms DalyBt(76:67:02:03:02:3E)!
11:14:04 ERROR [sampling] daly1 error: 147
11:14:04 INFO [sampling] Bleak version 0.13.1a1
11:14:12 ERROR [sampling] daly1 error: 147
11:14:12 INFO [sampling] Bleak version 0.13.1a1
11:14:12 ERROR [main] 1 exceptions occurred fetching BMSs
11:14:12 ERROR [main] Error (num 1, max 0) reading BMS: 147
11:14:12 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/app/bmslib/daly.py", line 114, in _q
    sample = await self._fetch_futures.wait_for(command, self.TIMEOUT)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/__init__.py", line 50, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/main.py", line 35, in fetch_loop
    await fn()
  File "/app/main.py", line 280, in fn
    raise exceptions[0]
  File "/app/main.py", line 275, in fn
    await t()
  File "/app/bmslib/sampling.py", line 69, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 97, in sample
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/daly.py", line 128, in fetch
    status = await self._fetch_status()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/daly.py", line 155, in _fetch_status
    response_data = await self._q(0x93)
                    ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/daly.py", line 116, in _q
    n_recv = num_responses - self._fetch_nr[command].count(None)
                             ~~~~~~~~~~~~~~^^^^^^^^^
KeyError: 147

END
```

fl4p, 2023-07-24:
I'll fix this with the next update

fl4p, 2023-07-25:
0.0.64 should fix it, please re-open if not

fl4p, 2023-07-25:
Duplicate of https://github.com/fl4p/batmon-ha/issues/107

closed 2023-07-25 by fl4p