fl4p/batmon-ha #119: Daly failing after fix in #117

wimg, 2023-07-26:
As a follow-up on #117 - after deploying the fix in there, this is the new issue :  
 

```
Paste log output between BEGIN and END:
BEGIN 
21:13:07 INFO [main] Power cycle bluetooth hardware
21:13:11 INFO [bt] BT Discovery:
21:13:16 INFO [bt] BT Device   DL-76670203023E   address=76:67:02:03:02:3E
21:13:16 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.66
21:13:16 INFO [bt] BMS DalyBt(76:67:02:03:02:3E) keep alive enabled
21:13:16 INFO [main] connecting mqtt @localhost
21:13:16 ERROR [main] mqtt connection error [Errno 99] Address not available
21:13:16 INFO [main] Fetching 1 BMS + 0 others serially, period=1.00s, keep_alive=True
21:13:16 INFO [sampling] connecting bms DalyBt(76:67:02:03:02:3E)
21:13:18 INFO [daly] found rx uuid to be working: 17 (tx 15, sx 48)
21:13:18 INFO [sampling] connected bms DalyBt(76:67:02:03:02:3E)!
21:13:26 ERROR [sampling] daly1 error: timeout awaiting result 93, got 0/1 responses
21:13:26 INFO [sampling] Bleak version 0.13.1a1
21:13:34 ERROR [sampling] daly1 error: timeout awaiting result 93, got 0/1 responses
21:13:34 INFO [sampling] Bleak version 0.13.1a1
21:13:34 ERROR [main] 1 exceptions occurred fetching BMSs
21:13:34 ERROR [main] Error (num 1, max 0) reading BMS: timeout awaiting result 93, got 0/1 responses
21:13:34 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/app/bmslib/daly.py", line 114, in _q
    sample = await self._fetch_futures.wait_for(command, self.TIMEOUT)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/__init__.py", line 50, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/main.py", line 35, in fetch_loop
    await fn()
  File "/app/main.py", line 280, in fn
    raise exceptions[0]
  File "/app/main.py", line 275, in fn
    await t()
  File "/app/bmslib/sampling.py", line 69, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 97, in sample
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/daly.py", line 128, in fetch
    status = await self._fetch_status()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/daly.py", line 155, in _fetch_status
    response_data = await self._q(0x93)
                    ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/daly.py", line 117, in _q
    raise TimeoutError(
TimeoutError: timeout awaiting result 93, got 0/1 responses

21:13:43 ERROR [sampling] daly1 error: timeout awaiting result 93, got 0/1 responses
21:13:43 INFO [sampling] Bleak version 0.13.1a1
21:13:43 ERROR [main] 1 exceptions occurred fetching BMSs
[... 118 lines cut ...]
Traceback (most recent call last):
  File "/app/main.py", line 35, in fetch_loop
    await fn()
  File "/app/main.py", line 280, in fn
    raise exceptions[0]
  File "/app/main.py", line 275, in fn
    await t()
  File "/app/bmslib/sampling.py", line 69, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 97, in sample
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/daly.py", line 128, in fetch
    status = await self._fetch_status()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/daly.py", line 155, in _fetch_status
    response_data = await self._q(0x93)
                    ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/daly.py", line 117, in _q
    raise TimeoutError(
TimeoutError: timeout awaiting result 93, got 0/1 responses

END
```

fl4p, 2023-07-28:
Which Daly BMS model is it?

Sayrin, 2023-08-03:
Hey i have a similar problem 

the software version on the daly ist 210629
and the BT Firmware ver is 2.1.16

if it helps i can connect on the shell via bluetoothctl

$ sudo bluetoothctl
[bluetooth]# power on
[bluetooth]# agent on
[bluetooth]# default-agent
[bluetooth]# scan on
[NEW] Device XX:XX:XX:XX:XX:XX <- this is the daly bms
[bluetooth]# scan off
[bluetooth]# trust XX:XX:XX:XX:XX:XX
[bluetooth]# pair XX:XX:XX:XX:XX:XX
[XX:XX:XX:XX:XX:XX]# pair XX:XX:XX:XX:XX:XX
Attempting to pair with XX:XX:XX:XX:XX:XX
[CHG] Device XX:XX:XX:XX:XX:XX Connected: yes
[bluetooth]# quit

here is the error log

14:12:17 INFO [bt] BT Discovery:
14:12:22 INFO [bt] BT Device   SSBTCAB5   address=XX:XX:XX:XX:XX:XX
14:12:22 INFO [bt] BT Device   SmartSolar HQ2036CD6PS   address=XX:XX:XX:XX:XX:XX
14:12:22 INFO [bt] BT Device   SSBTC968   address=XX:XX:XX:XX:XX:XX
14:12:22 INFO [bt] BT Device   XX:XX:XX:XX:XX:XX   address=XX:XX:XX:XX:XX:XX
14:12:22 INFO [bt] BT Device   XX:XX:XX:XX:XX:XX   address=XX:XX:XX:XX:XX:XX
14:12:22 INFO [bt] BT Device   XX:XX:XX:XX:XX:XX   address=XX:XX:XX:XX:XX:XX
14:12:22 INFO [bt] BT Device   XX:XX:XX:XX:XX:XX   address=XX:XX:XX:XX:XX:XX
14:12:22 INFO [bt] BT Device   XX:XX:XX:XX:XX:XX   address=XX:XX:XX:XX:XX:XX
14:12:22 INFO [bt] BT Device   XX:XX:XX:XX:XX:XX   address=XX:XX:XX:XX:XX:XX. <- das ist das daly
14:12:22 INFO [bt] BT Device   XX:XX:XX:XX:XX:XX  address=XX:XX:XX:XX:XX:XX
14:12:22 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.55
14:12:22 INFO [bt] BMS DalyBt(XX:XX:XX:XX:XX:XX) keep alive enabled
14:12:22 INFO [main] connecting mqtt batmon@192.168.0.39
14:12:22 INFO [main] Fetching 1 BMS + 0 others serially, period=1.00s, keep_alive=True
14:12:22 INFO [sampling] connecting bms DalyBt(XX:XX:XX:XX:XX:XX)
14:12:24 WARNING [daly] tried rx/tx/sx uuids 17/15/48: Characteristic with UUID 17 could not be found!
14:12:24 INFO [daly] found rx uuid to be working: 0000fff1-0000-1000-8000-00805f9b34fb (tx 0000fff2-0000-1000-8000-00805f9b34fb, sx 02f00000-0000-0000-0000-00000000ff01)
14:12:24 INFO [sampling] connected bms DalyBt(XX:XX:XX:XX:XX:XX)!
14:12:36 ERROR [sampling] daly error: <class 'asyncio.exceptions.TimeoutError'>
14:12:36 INFO [sampling] Bleak version 0.13.1a1
14:12:48 ERROR [sampling] daly error: <class 'asyncio.exceptions.TimeoutError'>
14:12:48 INFO [sampling] Bleak version 0.13.1a1
14:12:48 ERROR [main] 1 exceptions occurred fetching BMSs
14:12:48 ERROR [main] Error (num 1, max 0) reading BMS: 
14:12:48 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/batmon-ha/main.py", line 35, in fetch_loop
    await fn()
  File "/home/pi/batmon-ha/main.py", line 280, in fn
    raise exceptions[0]
  File "/home/pi/batmon-ha/main.py", line 275, in fn
    await t()
  File "/home/pi/batmon-ha/bmslib/sampling.py", line 69, in __call__
    return await self.sample()
  File "/home/pi/batmon-ha/bmslib/sampling.py", line 97, in sample
    sample = await bms.fetch()
  File "/home/pi/batmon-ha/bmslib/daly.py", line 128, in fetch
    status = await self._fetch_status()
  File "/home/pi/batmon-ha/bmslib/daly.py", line 155, in _fetch_status
    response_data = await self._q(0x93)
  File "/home/pi/batmon-ha/bmslib/daly.py", line 114, in _q
    sample = await self._fetch_futures.wait_for(command, self.TIMEOUT)
  File "/home/pi/batmon-ha/bmslib/__init__.py", line 50, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

wimg, 2023-08-09:
Sorry for my late reply, I was on holiday.

It's a Daly 4S 12V

wimg, 2023-08-10:
Also interesting in the settings : it seems there's a way to set the protocol and communication method.
I do have a USB cable attached to it, but I have no idea which settings to use :
[image]
[image]

fl4p, 2023-08-20:
I am guessing that it is a newer model which has a slightly different protocol and doesn't react on the request.

see this issue https://github.com/fl4p/batmon-ha/issues/33