fl4p/batmon-ha #122: connection error on Victron Smart Shunt - ERROR [sampling] solarCharger error: <class 'AssertionError'>

febu93, 2023-08-01:
I am successfully monitoring two Liontron batteries at the same time using your app. thanks again for your work!

I now tried to connect a new victron smart shunt that i recently bought. I always get the same error. It seems like its connecting successfully but than fails to parse the data froms the device.

what i tried:

- tried to fetch data from all three devices at the same time
- tried to fetch data only from victron smart shunt
- renamed the device
- change the pin code
- bt_power_cycle: true/false
- install_newer_bleak: true/false

any ideas?

```
Paste log output between BEGIN and END:
BEGIN 

   return runner.run(main)
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
  File "/usr/lib/python3.11/asyncio/base_events.py", line 640, in run_until_complete
    self.run_forever()
  File "/usr/lib/python3.11/asyncio/base_events.py", line 607, in run_forever
    self._run_once()
  File "/usr/lib/python3.11/asyncio/base_events.py", line 1922, in _run_once
    handle._run()
  File "/usr/lib/python3.11/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/app/main.py", line 282, in main
    await fetch_loop(fn, period=sample_period, max_errors=max_errors)
  File "/app/main.py", line 35, in fetch_loop
    await fn()
  File "/app/main.py", line 275, in fn
    await t()
  File "/app/bmslib/sampling.py", line 67, in __call__
    return await self.sample()
  File "/app/bmslib/sampling.py", line 89, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 250, in __aenter__
    await self.connect()
  File "/app/bmslib/victron.py", line 64, in connect
    await super().connect(timeout=timeout)
  File "/app/bmslib/bt.py", line 153, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 138, in _connect_client
    self.logger.debug("Pairing %s using psk '%s'...", self._psk)
Message: "Pairing %s using psk '%s'..."
Arguments: ('123456',)
20:16:19 DEBUG [client] BLE device @ ED:B9:01:26:9D:35 already paired with hci0
20:16:19 DEBUG [victron] write keep_alive b'`\xea'
20:16:19 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_B9_01_26_9D_35/service0029/char003d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\xff\xff\xff\x7f')>}, []]
20:16:19 DEBUG [client] Read Characteristic 6597eeff-4bda-4c1e-af4b-551c4cf74769 | /org/bluez/hci0/dev_ED_B9_01_26_9D_35/service0029/char003d: bytearray(b'\xff\xff\xff\x7f')
20:16:19 DEBUG [client] Write Characteristic 6597ffff-4bda-4c1e-af4b-551c4cf74769 | /org/bluez/hci0/dev_ED_B9_01_26_9D_35/service0029/char002a: b'`\xea'
20:16:19 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_B9_01_26_9D_35/service0029/char003d): ['org.bluez.GattCharacteristic1', {'Notifying': <dbus_next.signature.Variant ('b', True)>}, []]
20:16:19 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_B9_01_26_9D_35/service0029/char0035): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\xf9\xff')>}, []]
20:16:19 DEBUG [client] Read Characteristic 6597ed8e-4bda-4c1e-af4b-551c4cf74769 | /org/bluez/hci0/dev_ED_B9_01_26_9D_35/service0029/char0035: bytearray(b'\xf9\xff')
20:16:19 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_B9_01_26_9D_35/service0029/char0035): ['org.bluez.GattCharacteristic1', {'Notifying': <dbus_next.signature.Variant ('b', True)>}, []]
20:16:19 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_B9_01_26_9D_35/service0029/char0031): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b')\x05')>}, []]
20:16:19 DEBUG [client] Read Characteristic 6597ed8d-4bda-4c1e-af4b-551c4cf74769 | /org/bluez/hci0/dev_ED_B9_01_26_9D_35/service0029/char0031: bytearray(b')\x05')
20:16:19 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_B9_01_26_9D_35/service0029/char0031): ['org.bluez.GattCharacteristic1', {'Notifying': <dbus_next.signature.Variant ('b', True)>}, []]
20:16:19 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_B9_01_26_9D_35/service0029/char0039): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\xf4\xfd\xff\xff')>}, []]
20:16:19 DEBUG [client] Read Characteristic 6597ed8c-4bda-4c1e-af4b-551c4cf74769 | /org/bluez/hci0/dev_ED_B9_01_26_9D_35/service0029/char0039: bytearray(b'\xf4\xfd\xff\xff')
20:16:19 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_B9_01_26_9D_35/service0029/char0039): ['org.bluez.GattCharacteristic1', {'Notifying': <dbus_next.signature.Variant ('b', True)>}, []]
20:16:19 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_B9_01_26_9D_35/service0029/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\xff\xff')>}, []]
20:16:19 DEBUG [client] Read Characteristic 65970fff-4bda-4c1e-af4b-551c4cf74769 | /org/bluez/hci0/dev_ED_B9_01_26_9D_35/service0029/char002d: bytearray(b'\xff\xff')
20:16:19 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_B9_01_26_9D_35/service0029/char002d): ['org.bluez.GattCharacteristic1', {'Notifying': <dbus_next.signature.Variant ('b', True)>}, []]
20:16:19 INFO [sampling] connected bms SmartShuntBt(ED:B9:01:26:9D:35)!
20:16:19 ERROR [sampling] solarCharger error: <class 'AssertionError'>
20:16:19 INFO [sampling] Bleak version 0.13.1a1
20:16:19 ERROR [main] 1 exceptions occurred fetching BMSs
20:16:19 ERROR [main] Error (num 1, max 200) reading BMS: 
20:16:19 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 35, in fetch_loop
    await fn()
  File "/app/main.py", line 280, in fn
    raise exceptions[0]
  File "/app/main.py", line 275, in fn
    await t()
  File "/app/bmslib/sampling.py", line 67, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 95, in sample
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/victron.py", line 86, in fetch
    sample = BmsSample(**values)
             ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/bms.py", line 59, in __init__
    assert math.isfinite(soc)
AssertionError
20:16:20 ERROR [sampling] solarCharger error: <class 'AssertionError'>
20:16:20 INFO [sampling] Bleak version 0.13.1a1
20:16:20 ERROR [main] 1 exceptions occurred fetching BMSs
20:16:20 ERROR [main] Error (num 2, max 200) reading BMS: 
20:16:20 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 35, in fetch_loop
    await fn()
  File "/app/main.py", line 280, in fn
    raise exceptions[0]
  File "/app/main.py", line 275, in fn
    await t()
  File "/app/bmslib/sampling.py", line 67, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 95, in sample
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/victron.py", line 86, in fetch
    sample = BmsSample(**values)
             ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/bms.py", line 59, in __init__
    assert math.isfinite(soc)
AssertionError
20:16:21 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_ED_B9_01_26_9D_35/service0029/char0039): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\x12\xfe\xff\xff')>}, []]
20:16:21 DEBUG [victron] msg current 0.494

END
```

fl4p, 2023-08-01:
I'll fix this with the next update

febu93, 2023-08-07:
I just tried it and it works with the latest version. thanks a lot for your quick response 👍

closed 2023-08-07 by febu93