fl4p/batmon-ha #124: No module named 'crcmod'

Qweua, 2023-08-20:
Hi,

Looks like got 0.65 too fast and now have error on add-on startup:

`
s6-rc: info: service legacy-services successfully started
Traceback (most recent call last):
  File "/app/main.py", line 13, in <module>
    import bmslib.models.ant
  File "/app/bmslib/models/ant.py", line 21, in <module>
    import crcmod as crcmod
ModuleNotFoundError: No module named 'crcmod'
s6-rc: info: service legacy-services: stopping
`

Delphinus44, 2023-08-20:
Same here just upgrade and batmon stop working, same error. Config asked MQTT port, I put 1883 by IANA

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
22:22:20 INFO [install_bleak] bleak 0.13.1a1 installed, want 0.20.2, running pip3 install bleak==0.20.2
Collecting bleak==0.20.2
  Downloading bleak-0.20.2-py3-none-any.whl (135 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 135.6/135.6 kB 2.5 MB/s eta 0:00:00
Collecting dbus-fast<2.0.0,>=1.83.0 (from bleak==0.20.2)
  Downloading dbus_fast-1.92.0-cp311-cp311-musllinux_1_1_x86_64.whl (4.8 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.8/4.8 MB 6.3 MB/s eta 0:00:00
Installing collected packages: dbus-fast, bleak
  Attempting uninstall: bleak
    Found existing installation: bleak 0.13.1a1
    Uninstalling bleak-0.13.1a1:
      Successfully uninstalled bleak-0.13.1a1
Successfully installed bleak-0.20.2 dbus-fast-1.92.0
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Traceback (most recent call last):
  File "/app/main.py", line 13, in <module>
    import bmslib.models.ant
  File "/app/bmslib/models/ant.py", line 21, in <module>
    import crcmod as crcmod
ModuleNotFoundError: No module named 'crcmod'
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

daveprue, 2023-08-21:
Same bug...
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Traceback (most recent call last):
  File "/app/main.py", line 13, in <module>
    import bmslib.models.ant
  File "/app/bmslib/models/ant.py", line 21, in <module>
    import crcmod as crcmod
ModuleNotFoundError: No module named 'crcmod'
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

ToscanoMA, 2023-08-21:
hi, i have the same crc import error, did you fix it?

fl4p, 2023-08-21:
I accidentally pushed an update. You can just downgrade to an earlier version

StormCh, 2023-08-21:
Stop working with update 0.6.5

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
Traceback (most recent call last):
  File "/app/main.py", line 13, in <module>
    import bmslib.models.ant
  File "/app/bmslib/models/ant.py", line 21, in <module>
    import crcmod as crcmod
ModuleNotFoundError: No module named 'crcmod'
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

alejandropradocabrero, 2023-08-21:
any easy way to downgrade? Or a web with the steps to downgrade?

Qweua, 2023-08-21:
> hi, i have the same crc import error, did you fix it?

Restore from automatic backup before update is ok for me.

ToscanoMA, 2023-08-21:
> > hi, i have the same crc import error, did you fix it?
> 
> Restore from automatic backup before update is ok for me.

yeah, i fix it restoring automatic backup

fl4p, 2023-08-21:
I just pushed an update 0.0.66, can't test it because my HA is broken. Let me know if it works

@alejandropradocabrero  https://github.com/fl4p/batmon-ha/blob/master/doc/Downgrade.md

Qweua, 2023-08-21:
@fl4p 
0.0.66 works.
Didn't start automatically, I have to specify mmqt  port (possible better default value). But after that started without issues.

fabioven, 2023-08-21:
Batmon 0.0.66 works.
no problem!

Delphinus44, 2023-08-21:
crcmod fixed with 0.0.66 but I found another problem, going to open a new issue.

closed 2023-08-22 by fl4p