fl4p/batmon-ha #125: Ver 0.0.66 keep restarting if can't find a device

Delphinus44, 2023-08-21:
I have 4 devices and when I need to config one of them I switch it off and on to connect to it with mobile apps, before after I close the mobile connection after some minutes it will show in batmon.
With last release and because I have one of the BMS turn of, batmon keep rebooting and don't connect to the other 3.

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
21:52:55 INFO [install_bleak] bleak 0.13.1a1 installed, want 0.20.2, running pip3 install bleak==0.20.2
Collecting bleak==0.20.2
  Downloading bleak-0.20.2-py3-none-any.whl (135 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 135.6/135.6 kB 2.8 MB/s eta 0:00:00
Collecting dbus-fast<2.0.0,>=1.83.0 (from bleak==0.20.2)
  Downloading dbus_fast-1.93.0-cp311-cp311-musllinux_1_1_x86_64.whl (4.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.7/4.7 MB 6.6 MB/s eta 0:00:00
Installing collected packages: dbus-fast, bleak
  Attempting uninstall: bleak
    Found existing installation: bleak 0.13.1a1
    Uninstalling bleak-0.13.1a1:
      Successfully uninstalled bleak-0.13.1a1
Successfully installed bleak-0.20.2 dbus-fast-1.93.0
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
21:53:04 INFO [main] Power cycle bluetooth hardware
21:53:07 INFO [bt] BT Discovery:
21:53:12 INFO [bt] BT Device   EV105-ESQ   address=C8:47:8C:AAxxxx
21:53:12 INFO [bt] BT Device   LiFePo4   address=F4:13:D2:xxxxxx
21:53:12 INFO [bt] BT Device   EV280-K   address=C8:47:8C:BBxxxx
21:53:12 INFO [main] Bleak version 0.20.2, BtBackend version bluez-v5.66
21:53:12 ERROR [main] Main loop exception: Can't resolve device name C8:47:8C:E8:xx:xx, not discovered
21:53:12 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 318, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/app/main.py", line 157, in main
    name: str = dev.get('alias') or dev_by_addr(addr).name
                                    ^^^^^^^^^^^^^^^^^
  File "/app/main.py", line 119, in dev_by_addr
    raise Exception("Can't resolve device name %s, not discovered" % address)
Exception: Can't resolve device name C8:47:8C:E8:55:7B, not discovered
21:53:12 INFO [main] exit signal handler... (), {}, shutdown already False
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

fabioven, 2023-08-22:
I HAVE THA SAME PROBLEM WITH 2 BMS:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
21:26:09 INFO [main] Power cycle bluetooth hardware
21:26:12 INFO [bt] BT Discovery:
21:26:17 INFO [bt] BT Device   JK1   address=C8:47:8C:E5:XX.XX
21:26:17 INFO [bt] BT Device   JK2   address=C8:47:8C:EE:XX.XX
21:26:17 INFO [bt] BT Device   78-BD-BC-31-XXXX  address=78:BD:BC:31
21:26:17 INFO [bt] BT Device   64-07-F6-AAXXXX  address=64:07:F6:AA
21:26:17 INFO [bt] BT Device   4C-B8-62-69XXXX  address=4C:B8:62:69
21:26:17 INFO [bt] BT Device   CE-D5-DD-F6XXXX   address=CE:D5:DD:F6
21:26:17 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.66
21:26:17 INFO [bt] BMS JKBt(C8:47:8C:E5:XX:XX) keep alive enabled
21:26:17 INFO [bt] BMS JKBt(C8:47:8C:EE:XX:XX) keep alive enabled
21:26:17 ERROR [main] Main loop exception: unknown bms jk_bms in group VirtualGroupBms(battery_group1,[jk_bms,jk_bms2])
21:26:17 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 318, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/app/main.py", line 180, in main
    raise Exception("unknown bms %s in group %s" % (member_ref, group_bms))
Exception: unknown bms jk_bms in group VirtualGroupBms(battery_group1,[jk_bms,jk_bms2])
21:26:17 INFO [main] exit signal handler... (), {}, shutdown already False
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

fl4p, 2023-08-22:
> I HAVE THA SAME PROBLEM WITH 2 BMS:
> 
> s6-rc: info: service s6rc-oneshot-runner: starting s6-rc: info: service s6rc-oneshot-runner successfully started s6-rc: info: service fix-attrs: starting s6-rc: info: service fix-attrs successfully started s6-rc: info: service legacy-cont-init: starting s6-rc: info: service legacy-cont-init successfully started s6-rc: info: service legacy-services: starting s6-rc: info: service legacy-services successfully started 21:26:09 INFO [main] Power cycle bluetooth hardware 21:26:12 INFO [bt] BT Discovery: 21:26:17 INFO [bt] BT Device JK1 address=C8:47:8C:E5:XX.XX 21:26:17 INFO [bt] BT Device JK2 address=C8:47:8C:EE:XX.XX 21:26:17 INFO [bt] BT Device 78-BD-BC-31-XXXX address=78:BD:BC:31 21:26:17 INFO [bt] BT Device 64-07-F6-AAXXXX address=64:07:F6:AA 21:26:17 INFO [bt] BT Device 4C-B8-62-69XXXX address=4C:B8:62:69 21:26:17 INFO [bt] BT Device CE-D5-DD-F6XXXX address=CE:D5:DD:F6 21:26:17 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.66 21:26:17 INFO [bt] BMS JKBt(C8:47:8C:E5:XX:XX) keep alive enabled 21:26:17 INFO [bt] BMS JKBt(C8:47:8C:EE:XX:XX) keep alive enabled 21:26:17 ERROR [main] Main loop exception: unknown bms jk_bms in group VirtualGroupBms(battery_group1,[jk_bms,jk_bms2]) 21:26:17 ERROR [main] Stack: Traceback (most recent call last): File "/app/main.py", line 318, in asyncio.run(main()) File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run return runner.run(main) ^^^^^^^^^^^^^^^^ File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run return self._loop.run_until_complete(task) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete return future.result() ^^^^^^^^^^^^^^^ File "/app/main.py", line 180, in main raise Exception("unknown bms %s in group %s" % (member_ref, group_bms)) Exception: unknown bms jk_bms in group VirtualGroupBms(battery_group1,[jk_bms,jk_bms2]) 21:26:17 INFO [main] exit signal handler... (), {}, shutdown already False s6-rc: info: service legacy-services: stopping s6-rc: info: service legacy-services successfully stopped s6-rc: info: service legacy-cont-init: stopping s6-rc: info: service legacy-cont-init successfully stopped s6-rc: info: service fix-attrs: stopping s6-rc: info: service fix-attrs successfully stopped s6-rc: info: service s6rc-oneshot-runner: stopping s6-rc: info: service s6rc-oneshot-runner successfully stopped

This is another issue. As you can see in the log, you referenced a non-existing BMS in your group. instead of jk_bms1 in should be JK1 and JK2

fabioven, 2023-08-22:
Thank you for your suggestion. fixed the configuration.
and now it works perfectly.
nice app

closed 2023-09-24 by fl4p