fl4p/batmon-ha #127: bms disconnection and batmon stop

ToscanoMA, 2023-08-23:
hy everyone,
when starting batmon it works fine, until it crashes and stops the add-on.
Could you help me?
ty

```
8:22:07 INFO [sampling] DalyBt(A7:7A:12:03:01:25) times: connect=0.00s fetch=0.36s
08:22:08 INFO [sampling] dalyBMS: BmsSampl(30.3%,U=51.7V,I=12.30A,P=636W,q=84.8Ah/280,mos=nan°C)
08:22:09 WARNING [bt] BMS DalyBt(A7:7A:12:03:01:25) disconnected after 117.9s!
Traceback (most recent call last):
  File "/app/main.py", line 318, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/app/main.py", line 288, in main
    await fetch_loop(fn, period=sample_period, max_errors=max_errors)
  File "/app/main.py", line 37, in fetch_loop
    await fn()
  File "/app/main.py", line 281, in fn
    await t()
  File "/app/bmslib/sampling.py", line 68, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 157, in sample
    voltages = await bms.fetch_voltages()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/daly.py", line 225, in fetch_voltages
    resp = await self._q(0x95, num_responses=num_resp)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/daly.py", line 134, in _q
    sample = await self._fetch_futures.wait_for(command, self.TIMEOUT)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/__init__.py", line 50, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/tasks.py", line 479, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError
08:22:09 INFO [main] exit signal handler... (), {}, shutdown already False
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

fl4p, 2023-08-27:
this is similar to https://github.com/fl4p/batmon-ha/issues/123

fl4p, 2023-09-02:
fix with https://github.com/fl4p/batmon-ha/commit/f046c2d63a472b43fef1bf4c6375cc8be0511a84

fl4p, 2023-10-04:
please re-open if neede

closed 2023-10-04 by fl4p