fl4p/batmon-ha #128: Cannot get mqtt to work.

Quattrohead, 2023-08-25:
Can I only have 1 mqtt device ? I already have Solar Assistant running great, but I have 5 JBD BMS on it and cannot add a JK BMS, so was trying to use batmon and bluetooth adapter built into my Home Assistant.
Can someone please be kind enough to suggest how to set this up in plain English to an electro-mechanical guy why can just about figure out some super basic HA stuff....please.

s6-rc: info: service legacy-services successfully started
11:26:55 INFO [bt] BT Discovery:
11:27:00 INFO [bt] BT Device   Felicity 0062   address=70:3E:97:07:BF:EC
11:27:00 INFO [bt] BT Device   Felicity 0065   address=70:3E:97:08:10:A7
11:27:00 INFO [bt] BT Device   PROV_841FFCv   address=B8:F0:09:84:1F:FE
11:27:00 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.66
11:27:00 INFO [bt] BMS JKBt(C8:47:8C:E5:07:87) keep alive enabled
11:27:00 INFO [main] connecting mqtt kevin@core-mosquitto
11:27:03 ERROR [main] mqtt connection error [Errno 113] Host is unreachable
11:27:03 INFO [main] Fetching 1 BMS + 0 others serially, period=5.00s, keep_alive=True
11:27:03 INFO [sampling] connecting bms JKBt(C8:47:8C:E5:07:87)
11:27:03 INFO [main] mqtt watchdog loop started with timeout 900.0s
11:27:03 INFO [jikong] normal connect failed ([org.bluez.Error.Failed] Operation already in progress), connecting with scanner
11:27:12 ERROR [sampling] battery 5 jk error: Device C8:47:8C:E5:07:87 not discovered. Make sure it in range and is not being controled by another application. ({'E0:9F:2A:E4:92:39', 'B8:F0:09:84:1F:FE', '70:3E:97:08:10:A7', '38:3B:26:2E:C1:D1', '70:3E:97:07:BF:EC'})
11:27:12 INFO [sampling] Bleak version 0.13.1a1
11:27:12 INFO [sampling] connecting bms JKBt(C8:47:8C:E5:07:87)

fl4p, 2023-08-27:
from your logs, there is no mqtt issue. try enabling `install_newer_bleak` and see wether it improves bluetooth connection

Quattrohead, 2023-08-29:
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/switch/charge failed: 4 (4, 472)
10:51:28 INFO [sampling] felicity0065: BmsSampl(68.8%,U=53.4V,I=16.58A,P=885W,q=129.4Ah/188,mos=nan°C)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/meter/total_charge failed: 4 (4, 473)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/meter/total_energy failed: 4 (4, 474)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/meter/total_energy_discharge failed: 4 (4, 475)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/meter/total_energy_charge failed: 4 (4, 476)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/meter/total_cycles failed: 4 (4, 477)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/meter/total_abs_diff_charge failed: 4 (4, 478)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/1 failed: 4 (4, 479)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/2 failed: 4 (4, 480)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/3 failed: 4 (4, 481)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/4 failed: 4 (4, 482)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/5 failed: 4 (4, 483)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/6 failed: 4 (4, 484)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/7 failed: 4 (4, 485)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/8 failed: 4 (4, 486)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/9 failed: 4 (4, 487)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/10 failed: 4 (4, 488)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/11 failed: 4 (4, 489)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/12 failed: 4 (4, 490)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/13 failed: 4 (4, 491)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/14 failed: 4 (4, 492)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/15 failed: 4 (4, 493)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/16 failed: 4 (4, 494)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/min failed: 4 (4, 495)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/min_index failed: 4 (4, 496)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/max failed: 4 (4, 497)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/max_index failed: 4 (4, 498)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/delta failed: 4 (4, 499)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/average failed: 4 (4, 500)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/cell_voltages/median failed: 4 (4, 501)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/temperatures/1 failed: 4 (4, 502)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/temperatures/2 failed: 4 (4, 503)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/temperatures/3 failed: 4 (4, 504)
10:51:28 WARNING [mqtt_util] mqtt publish felicity0065/temperatures/4 failed: 4 (4, 505)
10:51:28 INFO [sampling] felicity0065 volt=3332,3335,3333,3335,3337,3334,3333,3333,3336,3334,3333,3338,3337,3333,3334,3336 temp=[34.5, 32.7, 32.6, 32.3]
10:51:30 INFO [sampling] connecting bms JKBt(C8:47:8C:E5:07:87,diy 14k1)
10:51:36 INFO [jikong] normal connect failed (<class 'TimeoutError'>), connecting with scanner
10:52:20 ERROR [sampling] diy 14k1 error: [org.bluez.Error.Failed] le-connection-abort-by-local
10:52:20 INFO [sampling] Bleak version 0.13.1a1
10:52:20 ERROR [main] 1 exceptions occurred fetching BMSs
10:52:20 ERROR [main] Error (num 13, max 200) reading BMS: [org.bluez.Error.Failed] le-connection-abort-by-local
10:52:20 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/dbus_next/aio/message_bus.py", line 305, in call
    await future
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/bmslib/models/jikong.py", line 119, in connect
    await super().connect(timeout=6)
  File "/app/bmslib/bt.py", line 180, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 144, in _connect_client
    await asyncio.wait_for(self.client.connect(timeout=timeout), timeout=timeout + 1)
  File "/usr/lib/python3.11/asyncio/tasks.py", line 479, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 348, in connect
    reply = await asyncio.wait_for(
            ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/main.py", line 37, in fetch_loop
    await fn()
  File "/app/main.py", line 291, in fn
    raise exceptions[0]
  File "/app/main.py", line 286, in fn
    await t()
  File "/app/bmslib/sampling.py", line 68, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
[... cut ...]
    await asyncio.wait_for(self.client.connect(timeout=timeout), timeout=timeout + 1)
  File "/usr/lib/python3.11/asyncio/tasks.py", line 479, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 359, in connect
    assert_reply(reply)
  File "/usr/lib/python3.11/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] le-connection-abort-by-local

Quattrohead, 2023-08-29:
Seems like I have various issues, at least I can see data from the JBD BMS now.

closed 2023-10-04 by fl4p