fl4p/batmon-ha #129: Bluetooth starts and stops everytime

Raydox187, 2023-08-29:
Hey, i checked troubleshooting and played with all the switches since 3 days. and nothing changes. i got 4 daly BMS and it reads all of them but the connection start and stop and i am a absolutly newby sometimes there are all 4 online and then one after the other is kicked out and starts new. changing parameters like sample period ....nothing makes stable.....please Help . ;-)

when u need more Data please say me . it is my first Project with Homeassistant and its little to big for me without any Help 

BEGIN 
 File "/app/bmslib/models/daly.py", line 131, in _q
    await self.client.write_gatt_char(self.UUID_TX, msg)
  File "/usr/lib/python3.11/site-packages/bleak/__init__.py", line 659, in write_gatt_char
    await self._backend.write_gatt_char(char_specifier, data, response)
  File "/usr/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 905, in write_gatt_char
    assert_reply(reply)
  File "/usr/lib/python3.11/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.UnknownObject] Method "WriteValue" with signature "aya{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist
13:33:00 ERROR [sampling] battery_group1 error: can only concatenate list (not "NoneType") to list
13:33:00 INFO [sampling] battery_group1 bms debug data: missing set()
13:33:00 INFO [sampling] Bleak version 0.20.2
13:33:00 ERROR [sampling] BMS unten rechts error: [org.freedesktop.DBus.Error.UnknownObject] Method "WriteValue" with signature "aya{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist
13:33:00 INFO [sampling] BMS unten rechts bms debug data: bytearray(b'\x01@\x00\x00\x00\x00\x00x')
13:33:00 INFO [sampling] Bleak version 0.20.2
13:33:01 INFO [sampling] BMS oben rechts: BmsSampl(83.1%,U=50.2V,I=-3.50A,P=-176W,q=83.1Ah/100,mos=nan°C)
13:33:02 INFO [sampling] BMS oben rechts volt=2281,2285,2287,2284,2283,2281,2281,2277,2285,2280,2279,2288,2287,2287,2285,2289,2285,2288,2289,2286,2285,2287 temp=[24]
13:33:02 INFO [sampling] DalyBt(C6:6C:09:03:0A:83,BMS oben rechts) times: connect=0.00s fetch=1.11s
13:33:02 ERROR [sampling] BMS unten links error: [org.freedesktop.DBus.Error.UnknownObject] Method "WriteValue" with signature "aya{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist
13:33:02 INFO [sampling] BMS unten links bms debug data: bytearray(b'\x01>\x00\x00\x00\x00\x00M')
13:33:02 INFO [sampling] Bleak version 0.20.2
13:33:02 INFO [sampling] BMS oben links: BmsSampl(78.2%,U=50.2V,I=-3.60A,P=-181W,q=78.2Ah/100,mos=nan°C)
13:33:14 ERROR [sampling] BMS oben links error: timeout awaiting result 95, got 6/8 responses
13:33:14 INFO [sampling] BMS oben links bms debug data: bytearray(b'\x00\x00\x00\x00\x0e')
13:33:14 INFO [sampling] Bleak version 0.20.2
13:33:14 ERROR [main] 4 exceptions occurred fetching BMSs
13:33:14 ERROR [main] Error (num 24, max 200) reading BMS: can only concatenate list (not "NoneType") to list
13:33:14 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 37, in fetch_loop
    await fn()
  File "/app/main.py", line 291, in fn
    raise exceptions[0]
  File "/app/main.py", line 286, in fn
    await t()
  File "/app/bmslib/sampling.py", line 68, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 96, in sample
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/group.py", line 63, in fetch
    return self.group.fetch()
           ^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/group.py", line 31, in fetch
    return sum_parallel(self.samples.values())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/group.py", line 119, in sum_parallel
    temperatures=sum((s.temperatures for s in samples), []),
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: can only concatenate list (not "NoneType") to list
13:33:15 ERROR [sampling] battery_group1 error: can only concatenate list (not "NoneType") to list
13:33:15 INFO [sampling] battery_group1 bms debug data: missing set()
13:33:15 INFO [sampling] Bleak version 0.20.2
13:33:15 INFO [sampling] BMS oben links: BmsSampl(78.2%,U=50.2V,I=-2.70A,P=-136W,q=78.2Ah/100,mos=nan°C)
13:33:16 INFO [sampling] BMS oben links volt=2285,2285,2285,2284,2282,2285,2285,2285,2285,2275,2281,2280,2288,2283,2287,2281,2283,2291,2281,2285,2283,2282 temp=[24]
13:33:16 INFO [sampling] DalyBt(C6:6C:09:01:0A:27,BMS oben links) times: connect=0.00s fetch=1.26s
13:33:16 ERROR [sampling] BMS unten links error: [org.freedesktop.DBus.Error.UnknownObject] Method "WriteValue" with signature "aya{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist
13:33:16 INFO [sampling] BMS unten links bms debug data: bytearray(b'\x01>\x00\x00\x00\x00\x00M')
13:33:16 INFO [sampling] Bleak version 0.20.2
13:33:16 ERROR [sampling] BMS unten rechts error: [org.freedesktop.DBus.Error.UnknownObject] Method "WriteValue" with signature "aya{sv}" on interface "org.bluez.GattCharacteristic1" doesn't exist
13:33:16 INFO [sampling] BMS unten rechts bms debug data: bytearray(b'\x01@\x00\x00\x00\x00\x00x')
13:33:16 INFO [sampling] Bleak version 0.20.2
13:33:17 INFO [sampling] BMS oben rechts: BmsSampl(83.1%,U=50.2V,I=-3.70A,P=-186W,q=83.1Ah/100,mos=nan°C)
END
```

fl4p, 2023-08-30:
Is it always the same BMS that fails?
Try to only enable the BMS which is most unstable and see if it works for some hours

Raydox187, 2023-08-31:
I Test with one BMS than with 2 and 3 and 4 ...everytime Same Problem ..... Today my esp8266 is comming and i Test with other i think ..... Because too mich time for me ..... Thanks

fl4p, 2023-10-20:
add-on option `install_newer_bleak` probably fixes this

closed 2023-10-20 by fl4p