fl4p/batmon-ha #131: China noname bms

test3210-d, 2023-09-03:
bms is similar to SuperVoltBt,  
same     UUID_RX = '6e400003-b5a3-f393-e0a9-e50e24dcca9e'
             UUID_TX = '6e400002-b5a3-f393-e0a9-e50e24dcca9e'

different commands 
packet structure 
[01] [03] [data length]      [data ...]      [2 byte checksum] 

-Send  -> 01 03 23 1C 00 04 8E 4B    #<>  bat info cell count, temp count ,  num ? ?
rx>  01 03 08 00 04 00 01 00 06 00 05 CD D5        #<>  4 cell 1 termo sensor , ?? ?? 

-Send  -> 01 03 D0 00 00 26 FC D0    #<>   voltages info
rx>  01 03 4C **0E 2D** 0E 14 0E 2C 0E 2D EE 49 EE 49 EE 49 EE 49 
rx>  EE 49 EE 49 EE 49 EE 49 EE 49 EE 49 EE 49 EE 49 EE 49 EE 49 
rx>  EE 49 EE 49 EE 49 EE 49 EE 49 EE 49 EE 49 EE 49 EE 49 EE 49 
rx>  EE 49 EE 49 EE 49 EE 49 0E 2D 0E 14 00 01 00 02 00 19 **05 A8** 
rx>  68 90 

0e2d = 3629 mv 
05a8= 1448 = 14.48 V battery voltage

-Send  -> 01 03 D0 26 0 19 5D 0B    <>  capacity && temp
rx>  0x01 ,0x03 ,0x32 ,0x02 ,0x80 ,0x0 ,0x0 ,0x0 ,0x0 ,0x0 ,0x0 ,0x0 ,0x0 ,0x0 ,0x0 ,0x0 ,0x0 ,0x0 ,0x0 
rx>  0x00 ,0x00 ,0x02 ,0x80 ,0x02 ,0x80 ,0x2 ,0x80 ,**0x00 ,0x07** ,0xUU ,0xUU ,0x0 ,0x64 ,0x0 ,0x64 ,0x26 ,0xf1 ,0x27 ,0x10 
rx>  0x27 ,0x10 ,0x0 ,0x2 ,0x00 ,0x00 ,0x00 ,0x04 ,0x00 ,0x00 ,0x00 ,0x00 ,0x00 ,0x00 ,0x76 ,0x95 

00 64 = 100% capacity 
2710 = 10 000 = 100 Ah  
00 07 = 0.7 A charge current
next cell marked as 0xUU ,0xUU  - discharge current , 

I think it is possible to make a template for such bms and just choose the position of the variable and its conversion formula in the config file, because the Chinese manufacturers do everything in a new way every time
I check communication using esp32 , it work . But python not for me :(

fl4p, 2023-09-04:
thanks, i'll think about a way to implement this

fl4p, 2026-09-03:
`type: noname_modbus` has been in the add-on since 1.97 and decodes the frames you posted byte for byte. No test report in three months, so closing; reopen with a verbose_log if a value is wrong on the real device, especially the masked discharge-current bytes.

closed 2026-09-03 by fl4p