fl4p/batmon-ha #132: Ant bms

klbsss, 2023-09-05:
ant not working

```
Paste log output between BEGIN and END:
BEGIN 

15:11:29 ERROR [sampling] battery1 error: non-bool switches values {'discharge': 1, 'charge': 1}
15:11:29 INFO [sampling] battery1 bms debug data: bytearray(b'~\xa1\x11\x00\x00|\x05\x02\x02\x07\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc4\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc4\x0e\xc6\x0e\xbe\x0e\xc6\x0e\xc8\x0e\xc2\x0e\xc4\x0e\xd8\xff\xd8\xff\x1e\x00\x1d\x00V\n\xe9\xffE\x00d\x00\x01\x01\x00\x00\x80\xb1O\x01\x04T\xe7\x00\xeb\xd19\x00\xc1\xff\xff\xff\xdd0r\x01\x00\x00\x00\x00\xc8\x0e\x05\x00\xbe\x0e\x03\x00\n\x00\xc4\x0e\x00\x00\x85\x00\x81\x00\x11\x02\xf1\xfa\x84m<\x00R67\x00\xc2Hq\x00\xc4\xa4J\x00\xe7\x9b\xaaU')
15:11:29 INFO [sampling] Bleak version 0.13.1a1
15:11:29 ERROR [main] 1 exceptions occurred fetching BMSs
15:11:29 ERROR [main] Error (num 201, max 200) reading BMS: non-bool switches values {'discharge': 1, 'charge': 1}
15:11:29 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 37, in fetch_loop
    await fn()
  File "/app/main.py", line 291, in fn
    raise exceptions[0]
  File "/app/main.py", line 286, in fn
    await t()
  File "/app/bmslib/sampling.py", line 68, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 96, in sample
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/ant.py", line 194, in fetch
    sample = BmsSample(
             ^^^^^^^^^^
  File "/app/bmslib/bms.py", line 74, in __init__
    assert all(map(lambda x: isinstance(x, bool), switches.values())), "non-bool switches values %s" % switches
AssertionError: non-bool switches values {'discharge': 1, 'charge': 1}

15:11:29 WARNING [main] too many errors, abort
15:11:29 INFO [main] fetch_loop <function main.<locals>.fn at 0x7f16f4362160> ends
15:11:29 INFO [main] All fetch loops ended. shutdown is already False
15:11:29 INFO [main] Disconnecting AntBt(16:AA:22:02:23:45,battery1)
15:11:31 WARNING [bt] BMS AntBt(16:AA:22:02:23:45,battery1) disconnected after 229.1s!
15:11:31 INFO [main] exit signal handler... (), {}, shutdown already True
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
END
```

fl4p, 2023-09-05:
I just pushed a fix, please try again

klbsss, 2023-09-05:
It's working)

fl4p, 2023-09-05:
good

closed 2023-09-05 by fl4p