fl4p/batmon-ha #133: Not Connect new bms jk

grydo, 2023-09-05:
the new jk bms, does not connect with several errors, jk02_32s protocol

fl4p, 2023-09-06:
what is the exact model number?

grydo, 2023-09-06:
JK-B2A20S20P

derek71, 2023-09-17:
mam to samo

derek71, 2023-09-17:
I confirm

fl4p, 2023-09-20:
please update to latest version 0.0.70 and post the error log here

rdaguerre, 2023-09-26:
File "/app/main.py", line 297, in fn
    await t()
  File "/app/bmslib/sampling.py", line 81, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 103, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 293, in __aenter__
    await self.connect()
  File "/app/bmslib/models/jikong.py", line 122, in connect
    await self._connect_with_scanner(timeout=timeout)
  File "/app/bmslib/bt.py", line 217, in _connect_with_scanner
    raise Exception('Device %s not discovered. Make sure it in range and is not being controled by '
Exception: Device C8:47:8C:10:83:85 not discovered. Make sure it in range and is not being controled by another application. ({'7C:C2:94:B6:F4:85', 'B4:E8:42:D7:DB:61'})
22:49:34 INFO [sampling] connecting bms JKBt(C8:47:8C:10:83:85,jk_)
22:49:41 INFO [jikong] normal connect failed (Device with address C8:47:8C:10:83:85 was not found.), connecting with scanner
22:49:50 ERROR [sampling] jk_ error: Device C8:47:8C:10:83:85 not discovered. Make sure it in range and is not being controled by another application. ({'7C:C2:94:B6:F4:85'})
22:49:50 INFO [sampling] Bleak version 0.13.1a1
22:49:50 ERROR [main] 1 exceptions occurred fetching BMSs
22:49:50 ERROR [main] Error (num 3, max 200) reading BMS: Device C8:47:8C:10:83:85 not discovered. Make sure it in range and is not being controled by another application. ({'7C:C2:94:B6:F4:85'})
22:49:50 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/bmslib/models/jikong.py", line 119, in connect
    await super().connect(timeout=6)
  File "/app/bmslib/bt.py", line 189, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 153, in _connect_client
    await asyncio.wait_for(self.client.connect(timeout=timeout), timeout=timeout + 1)
  File "/usr/lib/python3.11/asyncio/tasks.py", line 479, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 198, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address C8:47:8C:10:83:85 was not found.
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/main.py", line 37, in fetch_loop
    await fn()
  File "/app/main.py", line 302, in fn
    raise exceptions[0]
  File "/app/main.py", line 297, in fn
    await t()
  File "/app/bmslib/sampling.py", line 81, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 103, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 293, in __aenter__
    await self.connect()
  File "/app/bmslib/models/jikong.py", line 122, in connect
    await self._connect_with_scanner(timeout=timeout)
  File "/app/bmslib/bt.py", line 217, in _connect_with_scanner
    raise Exception('Device %s not discovered. Make sure it in range and is not being controled by '
Exception: Device C8:47:8C:10:83:85 not discovered. Make sure it in range and is not being controled by another application. ({'7C:C2:94:B6:F4:85'})
22:49:51 INFO [sampling] connecting bms JKBt(C8:47:8C:10:83:85,jk_)
22:49:57 INFO [jikong] normal connect failed (Device with address C8:47:8C:10:83:85 was not found.), connecting with scanner
22:50:07 ERROR [sampling] jk_ error: Device C8:47:8C:10:83:85 not discovered. Make sure it in range and is not being controled by another application. (set())
22:50:07 INFO [sampling] Bleak version 0.13.1a1
22:50:07 ERROR [main] 1 exceptions occurred fetching BMSs
22:50:07 ERROR [main] Error (num 4, max 200) reading BMS: Device C8:47:8C:10:83:85 not discovered. Make sure it in range and is not being controled by another application. (set())
22:50:07 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/bmslib/models/jikong.py", line 119, in connect
    await super().connect(timeout=6)
  File "/app/bmslib/bt.py", line 189, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 153, in _connect_client
    await asyncio.wait_for(self.client.connect(timeout=timeout), timeout=timeout + 1)
  File "/usr/lib/python3.11/asyncio/tasks.py", line 479, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 198, in connect
    raise BleakError(
bleak.exc.BleakError: Device with address C8:47:8C:10:83:85 was not found.
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/main.py", line 37, in fetch_loop
    await fn()
  File "/app/main.py", line 302, in fn
    raise exceptions[0]
  File "/app/main.py", line 297, in fn
    await t()
  File "/app/bmslib/sampling.py", line 81, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 103, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 293, in __aenter__
    await self.connect()
  File "/app/bmslib/models/jikong.py", line 122, in connect
    await self._connect_with_scanner(timeout=timeout)
  File "/app/bmslib/bt.py", line 217, in _connect_with_scanner
    raise Exception('Device %s not discovered. Make sure it in range and is not being controled by '
Exception: Device C8:47:8C:10:83:85 not discovered. Make sure it in range and is not being controled by another application. (set())
22:50:08 INFO [sampling] connecting bms JKBt(C8:47:8C:10:83:85,jk_)

donulf, 2023-10-09:
hello, I'm not able to connect the JK B2A20S20P too. the BMS works fine, I think your addon see the BMS, but doesn't work... 
I updated to the last version. Here the log:

s6-rc: info: service legacy-services successfully started
19:53:38 INFO [install_bleak] bleak 0.13.1a1 installed, want 0.20.2, running pip3 install bleak==0.20.2
Collecting bleak==0.20.2
  Downloading bleak-0.20.2-py3-none-any.whl (135 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 135.6/135.6 kB 424.4 kB/s eta 0:00:00
Collecting dbus-fast<2.0.0,>=1.83.0 (from bleak==0.20.2)
  Downloading dbus_fast-1.95.2-cp311-cp311-musllinux_1_1_x86_64.whl (4.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.7/4.7 MB 892.7 kB/s eta 0:00:00
Installing collected packages: dbus-fast, bleak
  Attempting uninstall: bleak
    Found existing installation: bleak 0.13.1a1
    Uninstalling bleak-0.13.1a1:
      Successfully uninstalled bleak-0.13.1a1
Successfully installed bleak-0.20.2 dbus-fast-1.95.2
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
19:53:51 INFO [main] Power cycle bluetooth hardware
19:53:54 INFO [bt] BT Discovery:
19:53:59 INFO [bt] BT Device   SmartBMV HQ230726UWA   address=E5:99:2F:49:C3:20
19:53:59 INFO [bt] BT Device   JK_B2A20S20P   address=C8:47:8C:E9:29:34
19:53:59 INFO [main] Bleak version 0.20.2, BtBackend version bluez-v5.66
19:53:59 INFO [bt] BMS JKBt(C8:47:8C:E9:29:34,battery) keep alive enabled
19:53:59 INFO [main] connecting mqtt mqtt-user@core-mosquitto
19:53:59 INFO [main] Fetching 1 BMS + 0 others serially, period=1.00s, keep_alive=True
19:53:59 INFO [sampling] connecting bms JKBt(C8:47:8C:E9:29:34,battery)
19:53:59 INFO [main] mqtt watchdog loop started with timeout 900.0s
19:54:08 ERROR [sampling] battery error: <class 'TimeoutError'>
19:54:08 INFO [sampling] battery bms debug data: {3: bytearray(b'U\xaa\xeb\x90\x03HJK_B2A20S20P\x00\x00\x00\x0011.XW\x00\x00\x0011.25\x00\x00\x00H\x8a\x00\x00\x04\x00\x00\x00JK_B2A20S20P\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00231009\x00\x002092245025\x000000\x00Input Userdata\x00\x00123456\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00Input Userdata\x00\x00|\xf8\xff\xff\x1f\r\x00\x00\x00\x00\x00\x00\x90\x0f\x00\x00\x00\x00\xc0\xd8\x03\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\x0f\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x96')}
19:54:08 INFO [sampling] Bleak version 0.20.2
19:54:16 ERROR [sampling] battery error: <class 'TimeoutError'>
19:54:16 INFO [sampling] battery bms debug data: {3: bytearray(b'U\xaa\xeb\x90\x03HJK_B2A20S20P\x00\x00\x00\x0011.XW\x00\x00\x0011.25\x00\x00\x00H\x8a\x00\x00\x04\x00\x00\x00JK_B2A20S20P\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00231009\x00\x002092245025\x000000\x00Input Userdata\x00\x00123456\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00Input Userdata\x00\x00|\xf8\xff\xff\x1f\r\x00\x00\x00\x00\x00\x00\x90\x0f\x00\x00\x00\x00\xc0\xd8\x03\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\x0f\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x96')}
19:54:16 INFO [sampling] battery device info: DeviceInfo(JK_B2A20S20P,hw-11.XW,sw-11.25,JK_B2A20S20P,#2092245025)
19:54:16 INFO [sampling] Bleak version 0.20.2
19:54:16 ERROR [main] 1 exceptions occurred fetching BMSs
19:54:16 ERROR [main] Error (num 1, max 200) reading BMS: 
19:54:16 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/tasks.py", line 500, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/main.py", line 38, in fetch_loop
    await fn()
  File "/app/main.py", line 307, in fn
    raise exceptions[0]
  File "/app/main.py", line 302, in fn
    await t()
  File "/app/bmslib/sampling.py", line 81, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 113, in sample
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/jikong.py", line 221, in fetch
    await self._fetch_futures.wait_for(0x02, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 50, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

fl4p, 2023-10-17:
similar to https://github.com/fl4p/batmon-ha/issues/111, lets continue there.
Enable verbose_log and post logs again

closed 2023-10-17 by fl4p