fl4p/batmon-ha #135: JK BMS (JK_B1A20S15P) TimeoutError

Eddy-Fischer, 2023-09-08:
Hi all,

I am unfortunately not that capable to read this error messages and I am not a linux expert.
I have a JK BMS (JK_B1A20S15P) and the following bluetooth:

[image]

I installed batmon-ha yesterday in my HA OS and it worked but today in the morning (without restarting the HA or my HP Thin Client) I have the following error:
TimeoutError
09:00:17 ERROR [sampling] battery1 error: <class 'TimeoutError'>
09:00:17 INFO [sampling] battery1 bms debug data: {3: bytearray(b"U\xaa\xeb\x90\x03\xe2JK_B1A20S15P\x00\x00\x00\x0011.XW\x00\x00\x0011.26\x00\x00\x00\x98\xa9E\x00\'\x00\x00\x00JK_B1A20S15P\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00230709\x00\x003020743165\x000000\x00Input Userdata\x00\x00141105\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00Input Userdata\x00\x00|\xf8\xff\xff\x1f\r\x00\x00\x00\x00\x00\x00\x90\x0f\x00\x00\x00\x00\xc0\xd8\x03\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe/\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00+")}
09:00:17 INFO [sampling] Bleak version 0.13.1a1
09:00:17 ERROR [main] 1 exceptions occurred fetching BMSs
09:00:17 ERROR [main] Error (num 17, max 200) reading BMS: 
09:00:17 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/main.py", line 37, in fetch_loop
    await fn()
  File "/app/main.py", line 302, in fn
    raise exceptions[0]
  File "/app/main.py", line 297, in fn
    await t()
  File "/app/bmslib/sampling.py", line 80, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 108, in sample
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/jikong.py", line 221, in fetch
    await self._fetch_futures.wait_for(0x02, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 50, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError

Then I tried to switch off, -on and paired the bluetooth:
[image]

and restart batmon, I still get the error:

09:30:56 DEBUG [jikong] bms msg(22) (buf300): 0 0 0 0 0 0 0 0 0 fe 2f 0 0 0 0 0 0 0 0 0 0 4e
09:30:56 DEBUG [jikong] got response 3 (len300)
09:30:56 DEBUG [jikong] write b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10'
09:30:56 DEBUG [client] Write Characteristic 0000ffe1-0000-1000-8000-00805f9b34fb | /org/bluez/hci0/dev_C8_47_8C_EC_EF_69/service000e/char0011: b'\xaaU\x90\xeb\x96\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10'
09:30:56 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_EC_EF_69/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\xaaU\x90\xeb\xc8\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00D')>}, []]
09:30:56 DEBUG [jikong] bms msg(20) (buf20): aa 55 90 eb c8 1 1 0 0 0 0 0 0 0 0 0 0 0 0 44
09:31:04 ERROR [sampling] battery1 error: <class 'TimeoutError'>
09:31:04 INFO [sampling] battery1 bms debug data: {3: bytearray(b"U\xaa\xeb\x90\x03\xf6JK_B1A20S15P\x00\x00\x00\x0011.XW\x00\x00\x0011.26\x00\x00\x00\xa0\xb0E\x00\'\x00\x00\x00JK_B1A20S15P\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00230709\x00\x003020743165\x000000\x00Input Userdata\x00\x00141105\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00Input Userdata\x00\x00|\xf8\xff\xff\x1f\r\x00\x00\x00\x00\x00\x00\x90\x0f\x00\x00\x00\x00\xc0\xd8\x03\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe/\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00N")}
09:31:04 INFO [sampling] Bleak version 0.13.1a1
09:31:12 ERROR [sampling] battery1 error: <class 'TimeoutError'>
09:31:12 INFO [sampling] battery1 bms debug data: {3: bytearray(b"U\xaa\xeb\x90\x03\xf6JK_B1A20S15P\x00\x00\x00\x0011.XW\x00\x00\x0011.26\x00\x00\x00\xa0\xb0E\x00\'\x00\x00\x00JK_B1A20S15P\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00230709\x00\x003020743165\x000000\x00Input Userdata\x00\x00141105\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00Input Userdata\x00\x00|\xf8\xff\xff\x1f\r\x00\x00\x00\x00\x00\x00\x90\x0f\x00\x00\x00\x00\xc0\xd8\x03\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe/\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00N")}
09:31:12 INFO [sampling] Bleak version 0.13.1a1
09:31:12 ERROR [main] 1 exceptions occurred fetching BMSs
09:31:12 ERROR [main] Error (num 1, max 200) reading BMS: 
09:31:12 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/main.py", line 37, in fetch_loop
    await fn()
  File "/app/main.py", line 302, in fn
    raise exceptions[0]
  File "/app/main.py", line 297, in fn
    await t()
  File "/app/bmslib/sampling.py", line 80, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 108, in sample
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/jikong.py", line 221, in fetch
    await self._fetch_futures.wait_for(0x02, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 50, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError

Anyone can assist?

Thanks and regards
Ed

Eddy-Fischer, 2023-09-08:
Info: I just unplugged the USB dongle and pulgged it back in. Now it is showing back the value. Not sure if this is a solution but for sure the error will comeback :)

Eddy-Fischer, 2023-09-08:
[image]

fl4p, 2023-09-09:
a reboot usually helps. 
the linux ble stack appears to be a bit wonky, might be bluez as well

Eddy-Fischer, 2023-09-10:
> a reboot usually helps. the linux ble stack appears to be a bit wonky, might be bluez as well

Hi, thanks for the reply. I did a system reboot but it did not solve the issue. After I unplugged the USB dongle and pulgged it back in, issue is solved. Any idea how I can solve this without going to cellar and unplug/plug in it ?
Kind regards
Ed

Eddy-Fischer, 2023-09-10:
It looks like the error message repeat each time:
15:02:10 DEBUG [client] Sending PINGREQ
15:02:10 DEBUG [client] Received PINGRESP
15:02:14 ERROR [sampling] battery1 error: <class 'TimeoutError'>
15:02:14 INFO [sampling] battery1 bms debug data: {3: bytearray(b"U\xaa\xeb\x90\x03{JK_B1A20S15P\x00\x00\x00\x0011.XW\x00\x00\x0011.26\x00\x00\x00\xf8\xa0H\x00\'\x00\x00\x00JK_B1A20S15P\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00230709\x00\x003020743165\x000000\x00Input Userdata\x00\x00141105\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00Input Userdata\x00\x00|\xf8\xff\xff\x1f\r\x00\x00\x00\x00\x00\x00\x90\x0f\x00\x00\x00\x00\xc0\xd8\x03\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe/\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1e")}
15:02:14 INFO [sampling] Bleak version 0.20.2
15:02:14 ERROR [main] 1 exceptions occurred fetching BMSs
15:02:14 ERROR [main] Error (num 12, max 200) reading BMS: 
15:02:14 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/main.py", line 37, in fetch_loop
    await fn()
  File "/app/main.py", line 302, in fn
    raise exceptions[0]
  File "/app/main.py", line 297, in fn
    await t()
  File "/app/bmslib/sampling.py", line 80, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 108, in sample
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/jikong.py", line 221, in fetch
    await self._fetch_futures.wait_for(0x02, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 50, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError
15:02:23 ERROR [sampling] battery1 error: <class 'TimeoutError'>
15:02:23 INFO [sampling] battery1 bms debug data: {3: bytearray(b"U\xaa\xeb\x90\x03{JK_B1A20S15P\x00\x00\x00\x0011.XW\x00\x00\x0011.26\x00\x00\x00\xf8\xa0H\x00\'\x00\x00\x00JK_B1A20S15P\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00230709\x00\x003020743165\x000000\x00Input Userdata\x00\x00141105\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00Input Userdata\x00\x00|\xf8\xff\xff\x1f\r\x00\x00\x00\x00\x00\x00\x90\x0f\x00\x00\x00\x00\xc0\xd8\x03\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe/\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1e")}
15:02:23 INFO [sampling] Bleak version 0.20.2
15:02:23 ERROR [main] 1 exceptions occurred fetching BMSs
15:02:23 ERROR [main] Error (num 13, max 200) reading BMS: 
15:02:23 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/main.py", line 37, in fetch_loop
    await fn()
  File "/app/main.py", line 302, in fn
    raise exceptions[0]
  File "/app/main.py", line 297, in fn
    await t()
  File "/app/bmslib/sampling.py", line 80, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 108, in sample
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/jikong.py", line 221, in fetch
    await self._fetch_futures.wait_for(0x02, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 50, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError

Eddy-Fischer, 2023-09-13:
Anyone could assist?

fl4p, 2023-09-13:
what is you exact bms model?

Eddy-Fischer, 2023-09-13:
Detail about the bms: https://eu.nkon.nl/jk-bms-b1a20s15p.html

fl4p, 2023-09-15:
do you know the hardware and firmware version?

Eddy-Fischer, 2023-09-15:
> do you know the hardware and firmware version?

[image]

Firmware from?

fl4p, 2023-09-15:
the BMS

fl4p, 2023-09-15:
i think you can find it in the app

Eddy-Fischer, 2023-09-15:
> i think you can find it in the app

If I want to check the bms, I need to unplug the USB dongle. I will do it tonight.
Thanks.

vladyspavlov, 2023-09-23:
I have the same issue with ORICO BTA-403 (CSR8510A10) 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode) and JK_B2A24S15P, hw ver 11.XW, sw ver 11.26
Replugging adapter helps

fl4p, 2023-09-24:
@vladyspavlov did you try to enable `bt_power_cycle` ?

vladyspavlov, 2023-09-24:
> @vladyspavlov did you try to enable `bt_power_cycle` ?

Yes, i tried, nothing changed

fl4p, 2023-09-24:
Appears to be an issue with the driver. Try running batmon outside any container, directly on the host.

Eddy-Fischer, 2023-09-24:
> > @vladyspavlov did you try to enable `bt_power_cycle` ?
> 
> Yes, i tried, nothing changed

I havent got any issue anymore since the last time I rebooted my Home assistance. I think I can live with it. Can I close the ticket?

vladyspavlov, 2023-09-24:
> 
> 
> 
> > > @vladyspavlov did you try to enable `bt_power_cycle` ?
> > 
> > Yes, i tried, nothing changed
> 
> I havent got any issue anymore since the last time I rebooted my Home assistance. I think I can live with it. Can I close the ticket?

Try to reboot again 🙂
It broke every time after the addon reboot until replug the adapter.

fl4p, 2023-09-24:
@Eddy-Fischer yea please close

closed 2023-09-24 by Eddy-Fischer

vladyspavlov, 2023-10-03:
Ok, with verbose logs enabled, I defined that for some reason it was trying to use hci0 (which is built-in RPI 4 Bluetooth) instead of defined in config for the BMS hci1. I fixed it by disabling the built-in Bluetooth to leave only one hci, which is by default hci0. The problem with timeouts still persists, but my next step was installing the newer bleak version, which looks like it solved the issues. I'll monitor it for some time and leave feedback.

fl4p, 2023-10-04:
That's odd. Might be related to the old bleak version.