fl4p/batmon-ha #143: Some values started to show zero

alinfulger, 2023-09-26:
Some of the values started to show ZERO value. They were working before and studently became zero.

JK BMS power / current / voltage / capacity / balance current / temperatures. Rest of parameters are still reported correctly

20:19:38 INFO [sampling] battery1: BmsSampl(0.0%,U=0.0V,I=0.00A,P=0W,q=21102.6Ah/0,mos=0.0°C)
20:19:39 INFO [sampling] battery1 volt=3324,3324,3323,3323,3322,3322,3321,3323,3321,3324,3323,3324,3323,3323,3323,3322 temp=[0.0, 0.0]

fl4p, 2023-09-28:
did this happen after an update?

alinfulger, 2023-09-30:
Hi Fabian,

Issue was fixed by changing / re-set the battery capacity in the BMS and
when reconnected it started to send correct data again.
Not sure what caused it, it was working fine no update was made in that day
and it just suddenly stopped sending some values, was sending zero value
for certain parameters.

Before fixing the issue, I downgraded to previous versions with no succes,
so it was something that happened in the BMS. BMS on local app was showing
correct values though.

Best regards,
Alin.

closed 2023-10-01 by fl4p