fl4p/batmon-ha #144: Feature Request: Set SoC (Daly)

XDGFX, 2023-09-27:
I've just started using this in a standalone Docker container and so far it's working great!

I would like to suggest adding the ability to set state of charge of the BMS.

This would be useful as my Daly BMS ignores current below ~2.5A, and resultantly after a few weeks the reported SoC is up to 20-30% off the real value. When I get a low voltage cut-off, it would be good to be able to reset the SoC to 0%, which I would normally do through the app but obviously cannot if batmon is currently connected to the BMS.

Great work, thanks again!

fl4p, 2023-09-27:
if you disable keep_alive and set sample_period to a couple of seconds you can still connect with your phone

fl4p, 2023-09-28:
doesn't daly set SoC to 0 when it reaches uv protection?

fl4p, 2023-11-14:
Just curious, what are the SoC levels reported by the BMS when UV protection hits? If it is always lets say 5%, you can reduce the capacity in the bms settings and it"ll better predict when battery is empty.

For reference, here is the UART documentation for Daly BMS, it misses a SoC write entpoint (it doesn’t include the codes for settings the switches neither though). For sure there is one, because the app has that function. Need to rever-engineer that, or do you know any app that implements that?

[v1.1 Daly RS485+UART Protocol.pdf](https://github.com/fl4p/batmon-ha/files/13348505/Part.4_.Daly.RS485%2BUART.Protocol.pdf)

[Daly UART_485 Communications Protocol V1.2 (2).pdf](https://github.com/fl4p/batmon-ha/files/13348500/Daly.UART_485.Communications.Protocol.V1.2.2.pdf)

XDGFX, 2023-11-14:
It varies, as it gets more out of sync over time. Normally around 30% SoC as it's been a couple of weeks since a full charge, but I had UV hit this morning at 70% SoC as it's been about a month and I've been balancing the charge level when I've been driving.

I don't know of any other apps that have done it sadly. I'm not too experienced with bluetooth programming, would the options be listen with a third device when that BT command is sent, or connect to and monitor the UART on the BMS end?

verjoz, 2024-02-24:
It would be a very useful feature, i have the same problem :+1:

fl4p, 2026-09-03:
Added a `set SOC` number entity (b8995b1, 74ca014, next release). Daly (`daly`, `daly_uart`): command 0x21 with clock and SOC*10, the frame dbus-serialbattery has used since 2023; the two PDFs above do not document it. JK with firmware 11+ (`jk`, `jk_32s`): register 0x6E, the soc_calibration number in esphome-jk-bms. JBD, ANT and the others have no known SoC write. Set the number in HA and the gauge should follow on the next poll. Untested here, so please report whether it takes, with a verbose_log if not.

fl4p, 2026-09-03:
2.20 is out with the `set SOC` number entity for Daly (`daly`, `daly_uart`) and JK firmware >= 11 (`jk`, `jk_32s`). Untested on hardware, please report whether the gauge takes the value.

XDGFX, 2026-09-03:
Unfortunately I won't be able to test as I've since moved to a new truck with a JK BMS

fl4p, 2026-09-03:
I implemented it for JK BMS too

XDGFX, 2026-09-03:
Ah, well my system is still a WIP (no HA setup yet) but I'm also planning on the bms communicating with my Victron gear through CAN, and bringing it into HA that way. So not something I'll be able to help verify for at least a little while

fl4p, 2026-09-06:
Ok, let me know once you tried it.