fl4p/batmon-ha #145: InfluxDB Error pls help...

haselnuss1971, 2023-09-29:
i get the error:

9:22:59 ERROR [main] Main loop exception: No module named 'influxdb'
09:22:59 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 335, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/app/main.py", line 233, in main
    sinks.append(InfluxDBSink(**{k[9:]: v for k, v in user_config.items() if k.startswith('influxdb_')}))
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sinks.py", line 30, in __init__
    import influxdb
ModuleNotFoundError: No module named 'influxdb'
09:22:59 INFO [main] exit signal handler... (), {}, shutdown already False

What i am missing... Thank you.

fl4p, 2023-10-04:
influxdb is an undocumented feature. batmon will write samples directly to the database. I added the missing dependency `influxdb` to `requirements.txt`.
next release will fix it

fl4p, 2023-10-04:
see https://github.com/fl4p/batmon-ha/blob/master/doc/InfluxDB.md

fl4p, 2023-10-05:
should've been fixed with v0.0.71, please re-open if needed

closed 2023-10-05 by fl4p