fl4p/batmon-ha #147: Update is not working 0.071

legolas3107, 2023-10-06:
I've been trying since it came out but I can't get this error out... can you help me ?

Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:226
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 14:42:47 (1 occurrences)
Last logged: 14:42:47

[2607201512] Error updating Batmon: The command '/bin/ash -o pipefail -c pip3 install -r requirements.txt' returned a non-zero code: 1
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 167, in async_install
    await async_update_addon(self.hass, slug=self._addon_slug, backup=backup)
  File "/usr/src/homeassistant/homeassistant/components/hassio/handler.py", line 53, in _wrapper
    raise HassioAPIError(data["message"])
homeassistant.components.hassio.handler.HassioAPIError: The command '/bin/ash -o pipefail -c pip3 install -r requirements.txt' returned a non-zero code: 1

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 226, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 2012, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2049, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 876, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 948, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 157, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 432, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 169, in async_install
    raise HomeAssistantError(f"Error updating {self.title}: {err}") from err
homeassistant.exceptions.HomeAssistantError: Error updating Batmon: The command '/bin/ash -o pipefail -c pip3 install -r requirements.txt' returned a non-zero code: 1

wifingalas, 2023-10-07:
get same error in my ha

fl4p, 2023-10-17:
This appears to be the new 'influxdb' requirement. What hardware and OS is it? Inside HA?

legolas3107, 2023-10-17:
Yes inside HA

closed 2023-10-17 by fl4p

legolas3107, 2023-10-17:
Now is OK … thanks