fl4p/batmon-ha #149: Dashboard not showing values even when they appear in LOG

ZS1FJK, 2023-10-07:
Hello.
First of all . Thank you for Batmon for HA. I use your addon for our camping trailer and its been amazing. 

I am having an issue with the values showing on my dash board. As you can see I get all the data in the log file but when I'm looking at the entities card I only get some data and some of it as unavailable. A simple example of this is the cell voltages. Often one of them are unavailable for extended periods of time.  You will see in the picture Bat2 shows in the log but not on the dashboard. Not sure If HA changed something or Batmon. I did just update everything. 

I have tried the rebuild option in the batmon addon section
I have uninstalled and reinstalled.
I have rebooted HA several times.
I have tried multiple config settings and even tried different names. 

I hope I have supplied enough information. Thank you

```
LOG:
BEGIN 
 File "/app/bmslib/sampling.py", line 103, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 293, in __aenter__
    await self.connect()
  File "/app/bmslib/models/jbd.py", line 53, in connect
    await super().connect(**kwargs)
  File "/app/bmslib/bt.py", line 189, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 153, in _connect_client
    await asyncio.wait_for(self.client.connect(timeout=timeout), timeout=timeout + 1)
  File "/usr/lib/python3.11/asyncio/tasks.py", line 489, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bleak/__init__.py", line 531, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 141, in connect
    async with async_timeout(timeout):
  File "/usr/lib/python3.11/asyncio/timeouts.py", line 111, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
22:04:58 INFO [mqtt_util] subscribe homeassistant/switch/Bat1/charge/set
22:04:58 INFO [sampling] Bat1: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:04:58 INFO [sampling] Bat1 volt=[3424,3421,3458,3414] temp=[17.8, 18.3, 18.2]
22:04:58 INFO [sampling] JbdBt(A4:C1:38:E3:7F:9D,Bat1) times: connect=17.00s fetch=0.17s
22:04:58 INFO [sampling] connecting bms JbdBt(A4:C1:38:35:68:81,Bat2)
22:05:20 ERROR [sampling] Bat2 error: <class 'TimeoutError'>
22:05:20 INFO [sampling] Bleak version 0.20.2
22:05:20 INFO [sampling] connecting bms JbdBt(A4:C1:38:35:68:81,Bat2)
22:05:22 INFO [sampling] connected bms JbdBt(A4:C1:38:35:68:81,Bat2)!
22:05:22 INFO [sampling] Bat2 subscribing for {'discharge': True, 'charge': True} switch change
22:05:22 INFO [mqtt_util] subscribe homeassistant/switch/Bat2/discharge/set
22:05:22 INFO [mqtt_util] subscribe homeassistant/switch/Bat2/charge/set
22:05:22 INFO [sampling] Bat2: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:05:23 INFO [sampling] Bat2 volt=[3434,3465,3397,3413] temp=[17.9, 18.5, 18.1]
22:05:23 INFO [sampling] JbdBt(A4:C1:38:35:68:81,Bat2) times: connect=2.77s fetch=0.16s
22:05:23 INFO [sampling] Bat1: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:05:23 INFO [sampling] Bat1 volt=[3425,3420,3458,3416] temp=[17.8, 18.3, 18.2]
22:05:27 INFO [sampling] Bat2: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:05:27 INFO [sampling] Bat2 volt=[3433,3465,3396,3412] temp=[17.9, 18.5, 18.1]
22:05:27 INFO [sampling] Bat1: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:05:27 INFO [sampling] Bat1 volt=[3425,3420,3457,3416] temp=[17.8, 18.3, 18.1]
22:05:31 INFO [sampling] Bat2: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:05:32 INFO [sampling] Bat2 volt=[3433,3465,3397,3413] temp=[17.9, 18.6, 18.1]
22:05:32 INFO [sampling] Bat1: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:05:32 INFO [sampling] Bat1 volt=[3424,3420,3457,3414] temp=[17.8, 18.3, 18.1]
22:05:36 INFO [sampling] Bat2: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:05:36 INFO [sampling] Bat2 volt=[3433,3465,3397,3412] temp=[17.9, 18.6, 18.1]
22:05:36 INFO [sampling] Bat1: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:05:36 INFO [sampling] Bat1 volt=[3425,3421,3457,3414] temp=[17.8, 18.3, 18.2]
22:05:40 INFO [sampling] Bat1: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:05:40 INFO [sampling] Bat1 volt=[3423,3421,3457,3414] temp=[17.8, 18.3, 18.2]
22:05:40 INFO [sampling] Bat2: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:05:41 INFO [sampling] Bat2 volt=[3433,3465,3396,3411] temp=[17.9, 18.5, 18.1]
22:05:45 INFO [sampling] Bat2: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
[... 45 lines cut ...]
22:06:35 INFO [sampling] Bat2: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:06:35 INFO [sampling] Bat2 volt=[3434,3465,3397,3412] temp=[17.9, 18.6, 18.1]
22:06:35 INFO [sampling] Bat1: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:06:35 INFO [sampling] Bat1 volt=[3425,3421,3458,3416] temp=[17.8, 18.3, 18.2]
22:06:39 INFO [sampling] Bat2: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:06:39 INFO [sampling] Bat2 volt=[3434,3465,3397,3412] temp=[17.9, 18.6, 18.1]
22:06:39 INFO [sampling] Bat1: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:06:39 INFO [sampling] Bat1 volt=[3423,3421,3457,3414] temp=[17.8, 18.3, 18.2]
22:06:44 INFO [sampling] Bat1: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:06:44 INFO [sampling] Bat1 volt=[3425,3420,3457,3414] temp=[17.8, 18.3, 18.2]
22:06:44 INFO [sampling] Bat2: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:06:44 INFO [sampling] Bat2 volt=[3434,3465,3396,3412] temp=[17.9, 18.6, 18.1]
22:06:48 INFO [sampling] Bat1: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:06:48 INFO [sampling] Bat1 volt=[3425,3421,3457,3416] temp=[17.8, 18.3, 18.2]
22:06:48 INFO [sampling] Bat2: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:06:48 INFO [sampling] Bat2 volt=[3434,3465,3397,3411] temp=[17.9, 18.6, 18.1]
22:06:48 INFO [sampling] JbdBt(A4:C1:38:35:68:81,Bat2) times: connect=0.00s fetch=0.16s
22:06:53 INFO [sampling] Bat2: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:06:53 INFO [sampling] Bat2 volt=[3433,3465,3396,3412] temp=[17.9, 18.6, 18.1]
22:06:53 INFO [sampling] Bat1: BmsSampl(100.0%,U=13.7V,I=0.00A,P=0W,Q=120.0Ah/120,mos=nan°C)
22:06:53 INFO [sampling] Bat1 volt=[3425,3421,3458,3416] temp=[17.8, 18.3, 18.2]
END
```
[image]
[image]

Qweua, 2023-10-08:
Same problem. Values present after restart but disappear later. Backup restore and reinstall doesn't help :(

fl4p, 2023-10-17:
Try to increase `expire_values_after` in the options (e.g. set it to 300)