fl4p/batmon-ha #151: JDB found but "disconnect because no data has been flowing for some time"

wimg, 2023-10-11:
Attempting to use batmon-ha with a JDB BMS. It seems to be found, but then it times out trying to get data from it.

Tried a number of the steps described in troubleshooting, but none seem to make a difference.

```
Paste log output between BEGIN and END:
BEGIN 
17:10:05 INFO [main] Power cycle bluetooth hardware
17:10:08 INFO [bt] BT Discovery:
17:10:13 INFO [bt] BT Device   xiaoxiang BMS V8.1   address=A4:C1:37:02:F6:B8
17:10:13 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.64
17:10:13 INFO [bt] BMS JbdBt(A4:C1:37:02:F6:B8,jbd1) keep alive enabled
17:10:13 INFO [main] connecting mqtt batmon@localhost
17:10:13 INFO [main] Fetching 1 BMS + 0 others serially, period=1.00s, keep_alive=True
17:10:13 INFO [sampling] connecting bms JbdBt(A4:C1:37:02:F6:B8,jbd1)
17:10:17 INFO [sampling] connected bms JbdBt(A4:C1:37:02:F6:B8,jbd1)!
17:10:33 ERROR [sampling] jbd1 error: <class 'asyncio.exceptions.TimeoutError'>
17:10:33 INFO [sampling] Bleak version 0.13.1a1
17:10:49 ERROR [sampling] jbd1 error: <class 'asyncio.exceptions.TimeoutError'>
17:10:49 INFO [sampling] Bleak version 0.13.1a1
17:10:49 ERROR [main] 1 exceptions occurred fetching BMSs
17:10:49 ERROR [main] Error (num 1, max 0) reading BMS:
17:10:49 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib64/python3.9/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/root/batmon-ha/main.py", line 38, in fetch_loop
    await fn()
  File "/root/batmon-ha/main.py", line 307, in fn
    raise exceptions[0]
  File "/root/batmon-ha/main.py", line 302, in fn
    await t()
  File "/root/batmon-ha/bmslib/sampling.py", line 85, in __call__
    return await self.sample()
  File "/root/batmon-ha/bmslib/sampling.py", line 117, in sample
    sample = await bms.fetch()
  File "/root/batmon-ha/bmslib/models/jbd.py", line 75, in fetch
    buf = await self._q(cmd=0x03)
  File "/root/batmon-ha/bmslib/models/jbd.py", line 69, in _q
    return await self._fetch_futures.wait_for(cmd, self.TIMEOUT)
  File "/root/batmon-ha/bmslib/__init__.py", line 50, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
  File "/usr/lib64/python3.9/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

END
```

wimg, 2023-10-15:
I updated bleak to v0.21.1 but to no avail :
]# python3 main.py
17:07:41 INFO [main] Power cycle bluetooth hardware
17:07:44 INFO [bt] BT Discovery:
17:07:49 INFO [bt] BT Device   CH9141BLE2U   address=84:C2:E4:B2:B4:DD
17:07:49 INFO [bt] BT Device   xiaoxiang BMS V8.1   address=A4:C1:37:02:F6:B8
17:07:49 INFO [bt] BT Device   BSC IP67 12/17 HQ2112VG7Q1   address=F3:4A:4C:1E:4B:29
17:07:49 INFO [bt] BT Device   DL-76670203023E   address=76:67:02:03:02:3E
17:07:49 INFO [main] Bleak version 0.21.1, BtBackend version bluez-v5.64
17:07:49 INFO [bt] BMS JbdBt(A4:C1:37:02:F6:B8,jbd1) keep alive enabled
17:07:49 INFO [main] connecting mqtt batmon@localhost
17:07:49 INFO [main] Fetching 1 BMS + 0 others serially, period=1.00s, keep_alive=True
17:07:49 INFO [sampling] connecting bms JbdBt(A4:C1:37:02:F6:B8,jbd1)
17:07:53 INFO [sampling] connected bms JbdBt(A4:C1:37:02:F6:B8,jbd1)!
17:08:09 ERROR [sampling] jbd1 error: <class 'asyncio.exceptions.TimeoutError'>
17:08:09 INFO [sampling] Bleak version 0.21.1

fl4p, 2023-10-15:
how far is the bms away? did you try to put it closer to your batmon host

wimg, 2023-10-15:
They're about 50cm apart. Connection works fine with my phone from 3m away.

fl4p, 2023-10-16:
did you ever receive any data, current or cell voltages?

wimg, 2023-10-16:
Sadly not, the above error is always the same, with nothing coming through at all.

fl4p, 2023-10-17:
Can you check the HW and SW in the app?

wimg, 2023-10-21:
[image]

[image]

[image]

fl4p, 2023-11-26:
You said it was a JBD, your screenshot show the Daly BMS app.
Is it the same BMS?

closed 2025-02-02 by fl4p