fl4p/batmon-ha #155: SoC calculation for a Supervolt battery group is not correct

Karatzie, 2023-10-30:
Describe the issue here.
I have add a battery group with my both Supervolt batteries.
Battery 1 has a SoC of 77.1 %
Battery 2 has a SoC of 99.8 %

The SoC of the battery group has a value of 44.23%

The sum of SoC of battery 1 and battery 2 is 77.1% + 99.8% = 176.9%.
This sum should be divided by two. 176.9%/2=88.45%.
In Batmon, the sum is divided by four. 176.9%/4=44.225%.

* Include exact BMS model name
Supervolt
* Query HW and SW numbers from the BMS app
* Enable verbose_log

```
Paste log output between BEGIN and END:
BEGIN 
![grafik](https://github.com/fl4p/batmon-ha/assets/98681564/7c62d7a4-7157-43e6-b4f6-91bf2dd7f092)

END
```

Karatzie, 2023-10-30:
The link of my screenshot does not work.

Here is the log but I am charging the batteries at this moment. The values are not the same as you see in post one.
20:10:31 INFO [sampling] battery_mica: BmsSampl(**49.7%,**U=13.6V,I=12.01A,P=163W,Q=198.9Ah/200,mos=15.0°C)
....
20:10:46 INFO [sampling] batt1: BmsSampl(**98.9%,**U=13.6V,I=7.68A,P=104W,Q=98.9Ah/100,mos=15.0°C)
.....
20:10:56 INFO [sampling] batt2: BmsSampl(**100.0%**,U=13.6V,I=4.30A,P=58W,Q=100.0Ah/100,mos=14.0°C)

fl4p, 2023-10-31:
I can reproduce this. The bug is in https://github.com/fl4p/batmon-ha/blob/1ba8f237bdf1459957a6f743f17dffc16478950e/bmslib/group.py#L133

`statistics.mean` need to be replaced with `sum`, to correctly compute the weighted sum of SoCs

fabiomato, 2023-11-09:
I23:01:40 INFO [sampling] JK2: BmsSampl(39.4%,U=24.4V,I=-0.18A,P=-5W,Q=118.1Ah/300,mos=0.0°C)
23:01:40 INFO [sampling] JK2 volt=[3487,3486,3487,3487,3487,3487,3487] temp=[13.2, 13.7]
23:01:40 INFO [sampling] JK1: BmsSampl(38.8%,U=24.5V,I=-0.36A,P=-9W,Q=155.2Ah/400,mos=0.0°C)
23:01:40 INFO [sampling] JK1 volt=[3504,3504,3505,3504,3504,3505,3505] temp=[14.1, 14.3]
23:02:10 INFO [sampling] BATTERIA: BmsSampl(19.5%,U=24.5V,I=-0.55A,P=-13W,Q=273.2Ah/700,mos=0.0°C)

i have the same problem.
****

fl4p, 2023-11-10:
ive pushed 1.73, which should fix this

fl4p, 2023-11-11:
please re-open if needed

closed 2023-11-11 by fl4p

Karatzie, 2023-11-12:
The issue is solved.

Thanks a lot.

BR

Karatzie