fl4p/batmon-ha #157: JK BMS 0 uptime and 0 mos temperature

holywen, 2023-11-02:
Describe the issue here.
uptime is 0, and mos temperature is also 0

JK_B2A2077128

See screenshot and logs:

[image]

```
Paste log output between BEGIN and END:
BEGIN 
13:53 INFO [sampling] jk1: BmsSampl(100.0%,U=54.9V,I=0.00A,P=0W,Q=309.0Ah/309,**mos=0.0°C**)
15:13:53 INFO [sampling] jk1 volt=[3456,3445,3456,3429,3456,3457,3435,3457,3337,3435,3457,3392,3394,3409,3458,3457] temp=[31.9, 31.1]
15:13:55 INFO [sampling] JKBt(C8:47:8C:ED:14:8A,jk1) times: connect=1.97s fetch=0.59s
END
```

fl4p, 2023-11-02:
Uptime zero will work with the next update. Not sure about the mos temp. Can you enable verbose_log and post the add-on logs here? It should contain some hex encoded binary data

fl4p, 2023-11-26:
thanks,
https://github.com/fl4p/batmon-ha/commit/80f654b6c02ba148731854797e3c21527bd43f54 fixes this.
will be in the next update.
please re-open if needed

closed 2023-11-26 by fl4p

holywen, 2023-12-15:
It works now, thanks a lot!