fl4p/batmon-ha #158: DALY BMS out of range values

danbi, 2023-11-04:
Since October 20 2023 I notice that the recording of my DALY BMS shows extreme values. Like SOC -2330%.  Sometimes it is +3200%.. maybe some calculation goes wrong?

Unfortunately, I cannot catch this in the logs.

fl4p, 2023-11-07:
thats odd
can you take screenshot of the soc history graph with the outliers and post it here?

danbi, 2023-11-07:
Here is a snapshot I took when submitting the issue
[image]

Here is another snapshot I took today, for one of the other packs. It apparently had communication issues during the "no data" period.

[image]

Here is the data for all three batteries, from the beginning of the month, together. 
[image]

For some reason  I do not see these out of range values anymore, but they are recorded..

fl4p, 2023-11-07:
Thanks, just noticed that the code for daly bms doesnt verify a checksum. If it did, it would drop these invalid packets. I'll see what I can do

danbi, 2023-11-07:
This might be the issue. I have generally poor bluetooth connectivity at times to the BMSes, and this might be the cause. The bluetooth proxy did not work for me, so am waiting for RS485 options eventually.

fl4p, 2023-11-10:
v1.73 asserts soc range

closed 2023-11-12 by fl4p