fl4p/batmon-ha #159: Victron smart shunt connection error

coil1002, 2023-11-05:
Can not get my smart shunt to connect same error every time. I have GAT enabled and am using the pin in configuration

`20:28:22 ERROR [main] 1 exceptions occurred fetching BMSs
20:28:22 ERROR [main] Error (num 115, max 200) reading BMS: [org.bluez.Error.Failed] le-connection-abort-by-local
20:28:22 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 38, in fetch_loop
    await fn()
  File "/app/main.py", line 315, in fn
    raise exceptions[0]
  File "/app/main.py", line 310, in fn
    await t()
  File "/app/bmslib/sampling.py", line 85, in __call__
    return await self.sample()
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 107, in sample
    async with bms:
  File "/app/bmslib/bt.py", line 316, in __aenter__
    await self.connect()
  File "/app/bmslib/models/victron.py", line 73, in connect
    await super().connect(timeout=timeout)
  File "/app/bmslib/bt.py", line 212, in connect
    await self._connect_client(timeout=timeout)
  File "/app/bmslib/bt.py", line 171, in _connect_client
    await asyncio.wait_for(self.client.connect(timeout=timeout), timeout=timeout + 1)
  File "/usr/lib/python3.11/asyncio/tasks.py", line 489, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 359, in connect
    assert_reply(reply)
  File "/usr/lib/python3.11/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] le-connection-abort-by-local`

coil1002, 2023-11-07:
> Are you running this as a home assistant add-on inside a docker container?

I am running the generic x86 on a pc platform my jk and jbd both connect
just not the victron

fl4p, 2023-11-14:
what bleak version is it (you find it in the logs)

Bear30, 2023-11-19:
I have the same problem. My bleak version is 0.13.1a1

coil1002, 2023-11-19:
Sorry I missed that question I am running bleak-0.20.2

fl4p, 2023-11-22:
does it help if you enable `install_newer_bleak` in the batmon config?

Bear30, 2023-11-23:
with 
- address: XX:XX:XX:XX:XX:XX
  type: victron
  alias: victron shunt
  pin: "000000"
  install_newer_bleak: True
it works. Thank you!!!

coil1002, 2023-11-23:
That did it thank you !!

closed 2023-11-23 by coil1002