fl4p/batmon-ha #160: BM6 battery monitoring support

phoenixckua, 2023-11-05:
It doesn’t support bm6
https://a.aliexpress.com/_EwNjlIf . Is it possible to implement bm6 support?
Android App BM6 working correctly.

phoenixckua, 2023-11-05:
50547B243010 = {"id":"50:54:7B:24:30:10","mac_type":0,"adv_type":0,"name":"BM6","manufacturerdata":"eeaee3c4f8988453578074e7f1677430","rssi":-60,"txpower":0}

fl4p, 2023-11-06:
I think its possible. I dont have such a device.
It doesnt monitor current right?

phoenixckua, 2023-11-07:
Yes, right. Available object: SoC, Voltage, Battery state, Temperature, Chanking Test, Charging Test, GPS Track. I need only SoC, Voltage, Temperature and maybe tracking. Ibeacon.    
https://lgbatterymonitor.com/products/wireless-bluetooth-4-0-battery-monitory-bm6-with-car-battery-health-guardian-app-monitoring

phoenixckua, 2023-11-07:
esp32-atom-lite (build SHA:10d7f7) OMG
50547B24F7B0 = {"id":"50:54:7B:24:F7:B0","mac_type":0,"adv_type":0,"name":"BM6","manufacturerdata":"4c0002153ba29cd9a42c894856badaf2606ef777114c0000cd","rssi":-57,"txpower":0,"brand":"GENERIC","model":"BM6 Battery Monitor","model_id":"BM6","type":"BATT","track":true,"batt":76,"device":"BM6 Tracker"}

0547B24F7B0 = {"id":"50:54:7B:24:F7:B0","mac_type":0,"adv_type":0,"name":"BM6","manufacturerdata":"ee017627cd6db2df636b4f31f0fe5d0f","rssi":-57,"txpower":0}

WolwX, 2024-01-27:
Hello :)

I'm too interested by including my BM6 battery monitor in my Homeassistant.
I'm on a Raspberry Pi and already use Batmon to monitor my BMS JBD lifepo battery.

Do you plan to add BM6 monitoring on Batmon ?

Thanks for your work 👍

fl4p, 2024-01-27:
Its hard to implement without own the actual HW. Maybe I'll buy it.
In the meantime any links to apps or code related to the communication with this BMS are welcome!

Karatzie, 2024-01-27:
Hi Fabian.

I have a BM2 device in my campervan.
I used it with my old lead batteries.
I never connect this device to home assistant.
I use ha-batmon with my new Supervolt batteries.
For the lead starter battery I will uns an INA226 with ESPhome.
That is cheaper than a BM2 or BM6 battery monitor.

Here is a discussion about BM2 and Home Assistant.
The discussion also includes the BM6 battery monitor.
[https://community.home-assistant.io/t/integrate-bluetooth-battery-monitoring-devices-cars-motorbikes-etc/511181](url)

They use the openmqttgateway.
Here is a link to the requirement  The implementation of the voltage object for BM6 battery monitoring is required #1792 
[https://github.com/1technophile/OpenMQTTGateway/issues/1792](url)

BR
Andreas

fl4p, 2024-01-27:
thanks i fixed the links.
will have a look

JeffWDH, 2024-08-13:
The BM6 uses a slightly different encryption key than the BM2. Details (and the key) can be found here: https://www.tarball.ca/posts/reverse-engineering-the-bm6-ble-battery-monitor/

dezanche, 2024-08-29:
that is great news!

jojo01fr, 2024-10-18:
Hi have a look on this nice info:
https://www.tarball.ca/posts/reverse-engineering-the-bm6-ble-battery-monitor/

Rafciq, 2025-02-05:
I have integration for https://github.com/Rafciq/BM6

jojo01fr, 2025-02-05:
Hi, is it compatible with OMG 1.8.x or does it need another BT gateway ?
Regards
jojo

Rafciq, 2025-02-05:
For this integration you only need HA with Bluetooth interface. For example, for HA on Raspberry Pi 5, only this integration is enough.

fl4p, 2025-02-06:
@Rafciq Thanks for posting it. Maybe I will use your code to put it into batmon.

Unfortunately, most of the BMS only support one BLE connection at a time.

Rafciq, 2025-02-06:
> @Rafciq Thanks for posting it. Maybe I will use your code to put it into batmon.
> 
> Unfortunately, most of the BMS only support one BLE connection at a time.
> This is due to the BLE adapter being connected through UART, which doesn't understand anything like sessions.

I did it so that others could benefit. I'd be happy if you used it.

jojo01fr, 2025-02-06:
> For this integration you only need HA with Bluetooth interface. For example, for HA on Raspberry Pi 5, only this integration is enough.
> 
> **Note!** If you have something that already reads from the BM6 device, for example an ESP32 or an app on your phone, the integration will not be able to work well. The BM6 has a flaw and only works well with one reader at the same time.

Hi, you mean raspi 5 can connect with its integrated BT chip ? my HA runs on VM, and my BT Gateway is the OMG 1.8.1, I also have a Shelly PM4 but not yet understood how to use it as gateway or proxy.
cheers
jojo

fl4p, 2026-09-03:
Added `type: bm6` (commit d19e058, in the next release). It speaks the AES-encrypted FFF3/FFF4 protocol from the tarball.ca write-up and Rafciq's integration and publishes voltage, temperature, SOC and the ok/low-voltage/charging state; the BM6 has no current sensor, so no power or Ah meters. The decoder is verified against the captured frames from the write-up but I have no device, so please test and post a verbose_log if anything is off. Remember it serves one connection at a time, close the phone app.

fl4p, 2026-09-03:
please create new issue if neede

closed 2026-09-03 by fl4p

fl4p, 2026-09-03:
2.20 is out with `type: bm6`. Untested on hardware, please try it and post the log if it does not read.