fl4p/batmon-ha #163: Publish HA discovery mqtt data before bms sample data

scotthernandez, 2023-11-22:
Unfortunately my daly bms doesn't quite work with the current code:
'''
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/daly.py", line 144, in _q
    raise TimeoutError(
TimeoutError: timeout awaiting result 95, got 5/6 responses

08:02:48 INFO [sampling] connecting bms DalyBt(18:81:03:03:05:EF,lifepo4-solar-1)
08:02:58 INFO [daly] normal connect failed (<class 'TimeoutError'>), connecting with scanner
'''

And because of the timeout in the last sample I suspect it is never able to publish the discovery mqtt topics that HA needs for creating the sensors.

Can you please check to see if it is easy to rearrange things to publish the discovery data before the sample data?

fl4p, 2023-11-22:
For how many cells did you configure the BMS?

scotthernandez, 2023-11-22:
It is a 15 cell pack - 48v system.

scotthernandez, 2023-11-22:
It does publish data for these topics:
* meter
* switch
* mosfet_status
* soc

If that helps.

fl4p, 2023-11-22:
I found a bug where the code wrongly computes the number of responses (the BMS sends voltages in tuples of 3)
Ill fix this with the next release.

scotthernandez, 2023-11-22:
Great, thanks.

Does it make sense to move the discovery mqtt data check before reading/publishing the samples?

fl4p, 2023-11-22:
Yeah kind of. The discovery function currently needs samples voltages to publish the number of cells and skip the nan fields.
I'll rewrite the code so discovery is sent out even if voltages fetch failed.

scotthernandez, 2023-11-22:
Great, thanks.

fl4p, 2023-11-26:
fix https://github.com/fl4p/batmon-ha/commit/dfd6e4a6da0662911097284ffec8da9910c53ed2 
is in the next release
please re-open if needed

closed 2023-11-26 by fl4p

scotthernandez, 2023-11-26:
> fix dfd6e4a
> is in the next release
> please re-open if needed

Thanks; will do.