fl4p/batmon-ha #164: Don't receive some JK-BMS Values

f-renz, 2023-11-27:
Hello,
I need help with Batmon 1.73 or 1.74 AddOn. Yesterday I install the Update from the V 1.73. to the  V1.74. But I didn't get the SoC value from from my JK-BMS. I have a JK B2A20S20P in use. Connected by BT 
What can I do?  

* JK_B2A20S20P
* sw_version": "11.26", "hw_version": "11.XW"
* 

```
Paste log output between BEGIN and END:
BEGIN 

12:53:19 DEBUG [mqtt_util] discovery msg homeassistant/sensor/batt01/_cell_voltages_11/config: {"unique_id": "batt01__cell_voltages_11", "name": "cell_voltages 11", "device_class": "voltage", "unit_of_measurement": "V", "state_topic": "batt01/cell_voltages/11", "expire_after": 15.0, "device": {"identifiers": ["3020745191"], "manufacturer": "JK", "name": "JK_B2A20S20P (batt01)", "model": "JK_B2A20S20P", "sw_version": "11.26", "hw_version": "11.XW"}}
12:53:19 DEBUG [client] Sending PUBLISH (d0, q0, r0, m70), 'b'homeassistant/sensor/batt01/_cell_voltages_11/config'', ... (357 bytes)

12:53:20 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_ED_09_46/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'%\xc2\x00\x00Q\x87\x00\x00\xb9\x02\x00\x00\xb2\x00\xb0\x00\x00\x00\x08\x00\x00\x00\x00\x01\xbf\x17\x00\x00\x00\xe2\x04\x00A\x00\x00\x00\x19\xf1@\x01d\x00\x00\x00P\n\xfc\x00\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\x01\x00\x00\x00\x8a\x04\x03\x00\x00\x00QEA@\x00\x00\x00\x00j\x13\xfc\x1a\x00\x01\x00\x01\x00\x05\x00\x00?\xd9_\x00\x00\x00\x00\x00\xcd\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'))>}, []]
12:53:20 DEBUG [jikong] bms msg(128) (buf278): 25 c2 0 0 51 87 0 0 b9 2 0 0 b2 0 b0 0 0 0 8 0 0 0 0 1 bf 17 0 0 0 e2 4 0 41 0 0 0 19 f1 40 1 64 0 0 0 50 a fc 0 1 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ff 0 1 0 0 0 8a 4 3 0 0 0 51 45 41 40 0 0 0 0 6a 13 fc 1a 0 1 0 1 0 5 0 0 3f d9 5f 0 0 0 0 0 cd 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0

![grafik](https://github.com/fl4p/batmon-ha/assets/32498861/03533067-1b59-4dd5-8362-755bcfb1a916)

END
```

danner85, 2024-02-07:
I have a :
JK_B2A20S20P
Firmware: 11.25H
Hardware: 11.XW
Have the same issue, i think because they are new models using jk02_32S protocol, vould be nice to be able to read al the values, i have a older model JK BMS and that works fine

semakof, 2024-05-23:
@fl4p I have this exact BMS - JK_B2A20S20P. 
But some values are wrong while others are not showing. hardware: V11.XW, Software: V11.262
Could you help fix this. 
Thanks..
[image]

[image]
[image]

rurenkov, 2024-12-11:
same for me, doesnt work for 7-8 month already

fl4p, 2025-02-03:
Please update to 1.81 and set type = jk_24s or jk_32s .

semakof, 2025-02-03:
> Please update to 1.81 and set type = jk_24s or jk_32s .

Wow. Thanks so much. Everything is working fine now. I am grateful.

[image]

fl4p, 2025-02-04:
good, i also fixed the detection, so leaving type = jk is fine

closed 2025-02-04 by fl4p