fl4p/batmon-ha #166: Random values out of normal range on most items - JBD BMS Bluetooth

karlduffell, 2023-11-27:
Most of the entities suffer random out of range values on some updates44 - shown here a temperature log and also a SOC sample...

I note a similar issue raised for JK but the latest updates have not fixed this issue for me with the JDB.

[image]

[image]

fl4p, 2023-11-28:
Looks like the BLE connection is poor and data is scrambled. checksum verification is currently missing for JBD.
i'll have a look at this

MiyukiDark, 2024-01-13:
I have the same issue, the signal is strong, yet from time to time data seems to be random 
I might look at it and implement it.

fl4p, 2025-11-14:
this should've been fixed, please open another issue if needed