fl4p/batmon-ha #168: Standalone installation of batmon-ha  not coming up after power fail reboot

FFJacot, 2023-11-28:
My recent install of batmon-ha on a standalone Raspberry Pi is working reliably but doesn't survive a reboot after a power fail.

Logging on to the Pi and running:

sudo systemctl status batmon.service

gives

=================================================================
● batmon.service - Batmon
     Loaded: loaded (/etc/systemd/system/batmon.service; enabled; preset: enabl>
     Active: active (running) since Tue 2023-11-28 13:07:12 SAST; 1h 49min ago
   Main PID: 522 (python3)
      Tasks: 2 (limit: 3912)
        CPU: 4min 25.494s
     CGroup: /system.slice/batmon.service
             └─522 python3 main.py

Nov 28 14:57:07 rpib env[522]: 14:57:07 WARNING [mqtt_util] mqtt publish RevovA>
Nov 28 14:57:07 rpib env[522]: 14:57:07 WARNING [mqtt_util] mqtt publish RevovA>
Nov 28 14:57:07 rpib env[522]: 14:57:07 WARNING [mqtt_util] mqtt publish RevovA>
Nov 28 14:57:07 rpib env[522]: 14:57:07 WARNING [mqtt_util] mqtt publish RevovA>
Nov 28 14:57:07 rpib env[522]: 14:57:07 WARNING [mqtt_util] mqtt publish RevovA>
Nov 28 14:57:07 rpib env[522]: 14:57:07 WARNING [mqtt_util] mqtt publish RevovA>
Nov 28 14:57:07 rpib env[522]: 14:57:07 WARNING [mqtt_util] mqtt publish RevovA>
Nov 28 14:57:07 rpib env[522]: 14:57:07 WARNING [mqtt_util] mqtt publish RevovA>
Nov 28 14:57:07 rpib env[522]: 14:57:07 WARNING [mqtt_util] mqtt publish RevovA>
Nov 28 14:57:07 rpib env[522]: 14:57:07 WARNING [mqtt_util] mqtt publish RevovA>
======================================================================

My HA on another Pi is not receiving data

On the command line I enter:

sudo systemctl  stop batmon.service
sudo systemctl start batmon.service

and then a minute or so laterm my HA starts receiving data.

Any ideas what I should be looking for?

Thanks

fl4p, 2023-11-28:
did you `systemctl enable batmon.service` ?

FFJacot, 2023-11-28:
Yes, after getting it working a while back. This worked fine until the first power fail/reboot.

Does it matter if you sudo the systemctl or do it from normal user login?

fl4p, 2023-11-28:
`systemctl` usually needs root permissions and tells you if it it fails

fl4p, 2023-11-28:
Try to add this in the `[Unit]` section:
```
After=network-online.target
Wants=network-online.target
```

FFJacot, 2023-11-28:
OK, added the:

After=network-online.target
Wants=network-online.target

to batmon.service, copied the file, then did:

systemctl enable batmon.service

again, then an sudo reboot and it looks like it's running just fine after coming up again. Hope it's fixed! After an earth leakage trip everything except my lights and geyser go down, specifically my HA box. I'll wait for the next trip to confirm all is well.

Thanks for rapid feed back....

fl4p, 2023-12-01:
good, i"ve added these lines to the Standalone readme file

closed 2023-12-01 by fl4p

FFJacot, 2023-12-01:
https://www.geewiz.co.za/stand-alone-inverters/22813-mecer-axpert-3000va3000w-24v-pure-sine-wave-solar-invertercharger-1200w-pwm.html

It's not quite bottom of the range, but close! 

A lot of people I know have gone the Sunsynk inverter route, usually 5 or 8 kw versions. My 3000W inverter is adequate but "cheap", and it's not really expandable. If I did it again, I'd get the more feature rich Sunsynk 3.6k or 5 kw version and bite the bullet on cost. 

My problem with the Mecer is that it doesn't directly talk to the batteries, even if they had an accessible bms like the Pylontech. The Sunsynks do, and seem to have an excellent monitoring setup, and are good at hybrid mixes of solar+utility.