fl4p/batmon-ha #173: Use Bluetooth Proxy in Home Assistant to extend range?

snipah, 2023-12-10:
No bug, but rather a feature request: Is it possible to use a bluetooth proxy for batmon-ha? I operate a smart camper but to save energy I switched moved my Home Assistant instance to my Unraid-Server and just pull all data from my Camper by using a Wireguard-Tunnel: This way I can get the data from an ESP32. My idea: Could I keep using batmon by using BT proxy (https://esphome.io/components/bluetooth_proxy.html)?

fl4p, 2023-12-11:
technically it is possible to use the esphome BT proxy. however, since batmon uses bleak (which abstracts linux BlueZ stack) and not the HA BT component. I have not verified this.
It needs some changes, which shouldn’t be too hard. The best way I think would be if someone writes a bleak backend for the HA/ESPhome BT component.

Alternatively, run batmon on ESPhome using micropython, which needs implementing a bleak backend for MicroPython BT stack as well.

fl4p, 2023-12-11:
dupe https://github.com/fl4p/batmon-ha/issues/34

closed 2023-12-11 by fl4p

dowermyr, 2024-01-21:
I see this is closed. Does the esp32 bluetooth proxy work? What to fill in the "adapter" row to make it work?