fl4p/batmon-ha #175: Error install addon from HomeAssistant (addon/2af0a32d_batmon)

klugev303, 2023-12-11:
I have this error:
The command '/bin/ash -o pipefail -c pip3 install -r requirements.txt' returned a non-zero code: 1

[image]

jim-bruna, 2023-12-11:
I have the same error on my rasberry pi 3b

jmmitchell81, 2023-12-12:
I have the same error pi3b+

fl4p, 2023-12-12:
can you please install Raspberry PI OS and follows this guide:
https://github.com/fl4p/batmon-ha/blob/master/doc/Standalone.md
does it work?

silviutu, 2023-12-12:
> can you please install Raspberry PI OS and follows this guide: https://github.com/fl4p/batmon-ha/blob/master/doc/Standalone.md does it work?

I have to buy another Raspberry. we need a more detailed guide, I'm not a software programmer.  thank.

fl4p, 2023-12-12:
you don’t need to write code, just use the linux console (many tutorials exist on the internet):
```
git clone https://github.com/fl4p/batmon-ha
cd batmon-ha
pip3 install -r requirements.txt
```

silviutu, 2023-12-13:
> you don’t need to write code, just use the linux console (many tutorials exist on the internet):
> 
> ```
> git clone https://github.com/fl4p/batmon-ha
> cd batmon-ha
> pip3 install -r requirements.txt
> ```

[image]
three hours to get nothing.

jmmitchell81, 2023-12-13:
Any further update on this?

jmmitchell81, 2023-12-13:
I have now tried on a RPi4-4Gb and get the same error. Something isn’t right

k2dp2k, 2023-12-13:
Same here.... The command '/bin/ash -o pipefail -c pip3 install -r requirements.txt' returned a non-zero code: 1
Rpi4 Hassio OS

Kodibuild, 2023-12-13:
Something happened yesterday after working fine for months. It would not start and I hit rebuild and it would not start stating that it was missing an image. Then uninstall and reinstall would give me the missing requirements error.
So I restored an old (same version) copy and hit start. Nothing happens (the start button stays and the integration icon stays greyed out. 
BUT... I am getting results fed to mqtt despite it showing as not started. As someone else said - "something isn't right".

k2dp2k, 2023-12-13:
Yeah but what ? is has to be a change on hassio side, because the code of this addon has not changed since two weeks...

Newspaperman57, 2023-12-13:
Hi, just to weigh in, i'm getting this error on a x86 haos installation too

webbbn, 2023-12-14:
This is due to a change in the base image that prevents installation of pip packages in the system directories. The easiest fix is likely to use a virtual environment.

webbbn, 2023-12-14:
This patch fixes it for me:

```
diff --git a/Dockerfile b/Dockerfile
index c6e9f14..b2676c4 100755
--- a/Dockerfile
+++ b/Dockerfile
@@ -24,7 +24,9 @@ RUN apk add git
 #COPY requirements.txt requirements.txt
 COPY . .

-RUN pip3 install -r requirements.txt
+RUN python3 -m venv venv && \
+    . ./venv/bin/activate && \
+    python -m pip install -r requirements.txt
 RUN chmod a+x addon_main.sh

 CMD [ "./addon_main.sh" ]
diff --git a/addon_main.sh b/addon_main.sh
index 5a3eb10..0a15320 100755
--- a/addon_main.sh
+++ b/addon_main.sh
@@ -1,6 +1,8 @@
 #!/usr/bin/with-contenv bashio

-python3 install_bleak.py
+. /app/venv/bin/activate
+
+python install_bleak.py

 MQTT_HOST=$(bashio::services mqtt "host")
```

jmmitchell81, 2023-12-14:
Thanks for the update. Would you be able to expand on how to implement this and install on home assistant running in a Pi3b+

k2dp2k, 2023-12-14:
[addons.zip](https://github.com/fl4p/batmon-ha/files/13670089/addons.zip)
unzip the file in your addons directory, after that go to addons -> addons store klick update, ther will be a local addon which you can install

jmmitchell81, 2023-12-14:
[image]
I can seem to find the addons folder.

k2dp2k, 2023-12-14:
> ![IMG_0073](https://private-user-images.githubusercontent.com/71808369/290446688-a072ff65-8e5c-471e-b64d-b40fd917e4ac.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI1NDA4ODgsIm5iZiI6MTcwMjU0MDU4OCwicGF0aCI6Ii83MTgwODM2OS8yOTA0NDY2ODgtYTA3MmZmNjUtOGU1Yy00NzFlLWI2NGQtYjQwZmQ5MTdlNGFjLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIxNFQwNzU2MjhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03ZTI1MzliMDg3YzA5NWU1ZmJiMWFiM2MwOTdjODg1NGU3NzQ5MDJlNzcxNzgyZTgyMzFhNzc4ZDU2YmQzNjRmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.3EGR-FigAWfVDylYPU5tZDYUPAPKmuRr-Lzyf7aaMUQ) I can seem to find the addons folder.

you have to go one folder level back at the moment yoou are inside the homeassistant folder... 
[image]

Newspaperman57, 2023-12-14:
@webbbn This fixed it for me too. Consider opening a pull-request with the change, to get this into the official repository ASAP :)

jmmitchell81, 2023-12-14:
@k2dp2k I’m up and running. Thanks for all your help. Much appreciated.

k2dp2k, 2023-12-14:
> @k2dp2k I’m up and running. Thanks for all your help. Much appreciated.

no, we have to thank @webbbn for his fix ;-)

jmmitchell81, 2023-12-14:
Thanks @webbbn

webbbn, 2023-12-14:
I'm glad you all got it working. I tried get all the repos setup to allow temporarily installing from my repo, but it was too late and my brain started shutting down. :-)

If @fl4p doesn't have already have it fixed this evening I'll submit a pull request.

webbbn, 2023-12-14:
I got the repos sorted out, so for now you should be able to use this repository: https://github.com/webbbn/home-assistant-addons.git

Just don't forget to switch back to this repo.

fl4p, 2023-12-14:
thanks everyone @silviutu @webbbn 
grateful for PR

webbbn, 2023-12-15:
PR created: https://github.com/fl4p/batmon-ha/pull/177

closed 2023-12-19 by fl4p

bbeijl, 2023-12-20:
[image]

Temperatures from my jk bms are unavailable, since version 1.75?