fl4p/batmon-ha #178: Can't seem to receive information from SOK BMS - ERROR [sampling] Battery 1 error (#1): timeout waiting for XX command.

Ampakinetic, 2023-12-17:
Hi, I have brand new SOK 24V battery and have no problem connecting to it with the ABC-BMS app.  However the sent commands don't seem to get any responses back with batmon.  I've added a little bit of extra logging and also tried to call the name command before the info command but it's got the same timeout error either way.  The verbose log is below, happy to provide more information if it can be helpful.

```
Paste log output between BEGIN and END:
BEGIN 
soa@orangepizero:~/batmon-ha$ python3 main.py
error reading /data/options.json, trying options.json [Errno 2] No such file or directory: '/data/options.json'
20:40:51 INFO [bt] BT Discovery:
20:40:57 INFO [bt] BT 01:B6:EC:DA:91:54                SOK-24V1247
20:40:57 INFO [main] Verbose logging enabled
20:40:57 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.55
20:40:57 DEBUG [bt] BMS SokBt(01:B6:EC:DA:91:54,Battery 1) keep alive enabled
20:40:57 INFO [main] connecting mqtt mppsolar@msg.ediy.co.nz
20:40:57 DEBUG [client] Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b''
20:40:57 INFO [main] Fetching 1 BMS + 0 virtual + 0 others serially, period=1.00s, keep_alive=True
20:40:57 INFO [sampling] connecting bms SokBt(01:B6:EC:DA:91:54,Battery 1)
20:40:57 INFO [bt] connecting Battery 1 (01:B6:EC:DA:91:54) adapter=default timeout=20
20:40:57 DEBUG [client] Connecting to device @ 01:B6:EC:DA:91:54 with hci0
20:40:57 DEBUG [client] Received CONNACK (0, 0)
20:40:57 DEBUG [scanner] cached devices: {'/org/bluez/hci0/dev_01_B6_EC_DA_91_54': {'Address': '01:B6:EC:DA:91:54', 'AddressType': 'public', 'Name': 'SOK-24V1247', 'Alias': 'SOK-24V1247', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['0000fff0-0000-1000-8000-00805f9b34fb'], 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False}}
20:40:58 DEBUG [scanner] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_next.signature.Variant ('b', True)>}, []]
20:40:58 DEBUG [scanner] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_01_B6_EC_DA_91_54): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -69)>}, []]
20:40:58 DEBUG [scanner] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_01_B6_EC_DA_91_54): ['org.bluez.Device1', {}, ['RSSI']]
20:40:58 DEBUG [scanner] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_next.signature.Variant ('b', False)>}, []]
20:40:58 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_01_B6_EC_DA_91_54): ['org.bluez.Device1', {'Connected': <dbus_next.signature.Variant ('b', True)>}, []]
20:40:58 DEBUG [client] Connection successful (/org/bluez/hci0/dev_01_B6_EC_DA_91_54)
20:40:58 DEBUG [client] Waiting for ServicesResolved (/org/bluez/hci0/dev_01_B6_EC_DA_91_54)
20:40:59 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_01_B6_EC_DA_91_54/service000a', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.GattService1': {'UUID': <dbus_next.signature.Variant ('s', 00001801-0000-1000-8000-00805f9b34fb)>, 'Device': <dbus_next.signature.Variant ('o', /org/bluez/hci0/dev_01_B6_EC_DA_91_54)>, 'Primary': <dbus_next.signature.Variant ('b', True)>, 'Includes': <dbus_next.signature.Variant ('ao', [])>}, 'org.freedesktop.DBus.Properties': {}}]
20:40:59 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_01_B6_EC_DA_91_54/service000b', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.GattService1': {'UUID': <dbus_next.signature.Variant ('s', 0000ffe0-0000-1000-8000-00805f9b34fb)>, 'Device': <dbus_next.signature.Variant ('o', /org/bluez/hci0/dev_01_B6_EC_DA_91_54)>, 'Primary': <dbus_next.signature.Variant ('b', True)>, 'Includes': <dbus_next.signature.Variant ('ao', [])>}, 'org.freedesktop.DBus.Properties': {}}]
20:40:59 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_01_B6_EC_DA_91_54/service000b/char000c', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.GattCharacteristic1': {'UUID': <dbus_next.signature.Variant ('s', 0000ffe2-0000-1000-8000-00805f9b34fb)>, 'Service': <dbus_next.signature.Variant ('o', /org/bluez/hci0/dev_01_B6_EC_DA_91_54/service000b)>, 'Value': <dbus_next.signature.Variant ('ay', b'')>, 'Flags': <dbus_next.signature.Variant ('as', ['write-without-response'])>, 'WriteAcquired': <dbus_next.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
20:40:59 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_01_B6_EC_DA_91_54/service000b/char000e', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.GattCharacteristic1': {'UUID': <dbus_next.signature.Variant ('s', 0000ffe1-0000-1000-8000-00805f9b34fb)>, 'Service': <dbus_next.signature.Variant ('o', /org/bluez/hci0/dev_01_B6_EC_DA_91_54/service000b)>, 'Value': <dbus_next.signature.Variant ('ay', b'')>, 'Notifying': <dbus_next.signature.Variant ('b', False)>, 'Flags': <dbus_next.signature.Variant ('as', ['read', 'notify'])>, 'NotifyAcquired': <dbus_next.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
20:40:59 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_01_B6_EC_DA_91_54/service000b/char000e/desc0010', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.GattDescriptor1': {'UUID': <dbus_next.signature.Variant ('s', 00002902-0000-1000-8000-00805f9b34fb)>, 'Characteristic': <dbus_next.signature.Variant ('o', /org/bluez/hci0/dev_01_B6_EC_DA_91_54/service000b/char000e)>, 'Value': <dbus_next.signature.Variant ('ay', b'')>}, 'org.freedesktop.DBus.Properties': {}}]
20:40:59 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_01_B6_EC_DA_91_54/service0011', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.GattService1': {'UUID': <dbus_next.signature.Variant ('s', 0000180a-0000-1000-8000-00805f9b34fb)>, 'Device': <dbus_next.signature.Variant ('o', /org/bluez/hci0/dev_01_B6_EC_DA_91_54)>, 'Primary': <dbus_next.signature.Variant ('b', True)>, 'Includes': <dbus_next.signature.Variant ('ao', [])>}, 'org.freedesktop.DBus.Properties': {}}]
20:40:59 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_01_B6_EC_DA_91_54/service0011/char0012', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.GattCharacteristic1': {'UUID': <dbus_next.signature.Variant ('s', 00002a29-0000-1000-8000-00805f9b34fb)>, 'Service': <dbus_next.signature.Variant ('o', /org/bluez/hci0/dev_01_B6_EC_DA_91_54/service0011)>, 'Value': <dbus_next.signature.Variant ('ay', b'')>, 'Flags': <dbus_next.signature.Variant ('as', ['read'])>}, 'org.freedesktop.DBus.Properties': {}}]
20:40:59 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_01_B6_EC_DA_91_54/service0011/char0014', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.GattCharacteristic1': {'UUID': <dbus_next.signature.Variant ('s', 00002a24-0000-1000-8000-00805f9b34fb)>, 'Service': <dbus_next.signature.Variant ('o', /org/bluez/hci0/dev_01_B6_EC_DA_91_54/service0011)>, 'Value': <dbus_next.signature.Variant ('ay', b'')>, 'Flags': <dbus_next.signature.Variant ('as', ['read'])>}, 'org.freedesktop.DBus.Properties': {}}]
20:40:59 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_01_B6_EC_DA_91_54/service0011/char0016', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.GattCharacteristic1': {'UUID': <dbus_next.signature.Variant ('s', 00002a25-0000-1000-8000-00805f9b34fb)>, 'Service': <dbus_next.signature.Variant ('o', /org/bluez/hci0/dev_01_B6_EC_DA_91_54/service0011)>, 'Value': <dbus_next.signature.Variant ('ay', b'')>, 'Flags': <dbus_next.signature.Variant ('as', ['read'])>}, 'org.freedesktop.DBus.Properties': {}}]
[... cut ...]
  File "/home/soa/batmon-ha/bmslib/__init__.py", line 84, in wait_for
    raise asyncio.TimeoutError("timeout waiting for %s" % name)
asyncio.exceptions.TimeoutError: timeout waiting for 192

20:42:04 INFO [main] fetch_loop <function main.<locals>.fn at 0xb58c18e0> ends
20:42:04 INFO [main] All fetch loops ended. shutdown is already 1
20:42:04 INFO [main] Disconnecting SokBt(01:B6:EC:DA:91:54,Battery 1)
20:42:04 INFO [main] exit signal handler... (), {}, shutdown was True
soa@orangepizero:~/batmon-ha$

END
```

fl4p, 2023-12-19:
how far is the bms away?
timeouts are usually either protocol issues or poor BLE connection.

Ampakinetic, 2023-12-19:
Thanks for responding.  In this case the BMS is about 30cm away from the BT dongle because I've got an OrangePi right next to it to also talk to the invertor.    Using bluetoothctl I can connect to the battery and query eg the manufacturer name field without problems, so it's probably not a linux BT driver issue.

Ampakinetic, 2023-12-21:
Is there a way I can test for a protocol or connection issue with bluetoothctl ?   Does anyone else with SOK batteries have issues or does it run fine?

fl4p, 2023-12-23:
looks like a protocol issue. someone has to reverse engineer it

rmkilc, 2024-03-09:
I got a SOK 24V 100Ah today and it looks like I am having the same issue. The app works fine but I am getting the same _timeout waiting for 192_ as OP posted. I have tried two different bluetooth dongles with them as close as touching the battery.

KSDaemon, 2024-09-04:
Experiencing the same issue :(