fl4p/batmon-ha #180: ANT BMS dont work

Arnix11, 2023-12-20:
Describe the issue here.
If you experience bluetooth connection problems, follow these steps before opening an issue:
https://github.com/fl4p/batmon-ha#troubleshooting

* Include exact BMS model name
* Query HW and SW numbers from the BMS app
* Enable verbose_log

```
Paste log output between BEGIN and END:
BEGIN 
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
21:30:38 INFO [install_bleak] bleak 0.13.1a1 installed, want 0.20.2, running pip3 install bleak==0.20.2
21:30:38 INFO [install_bleak] influxdb installed=False need=False
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Collecting bleak==0.20.2
  Downloading https://wheels.home-assistant.io/musllinux-index/bleak-0.20.2-py3-none-any.whl (135 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 135.6/135.6 kB 4.6 MB/s eta 0:00:00
Collecting dbus-fast<2.0.0,>=1.83.0 (from bleak==0.20.2)
  Downloading https://wheels.home-assistant.io/musllinux-index/dbus_fast-1.95.2-cp311-cp311-musllinux_1_2_aarch64.whl (1.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.7/1.7 MB 12.2 MB/s eta 0:00:00
Installing collected packages: dbus-fast, bleak
  Attempting uninstall: bleak
    Found existing installation: bleak 0.13.1a1
    Uninstalling bleak-0.13.1a1:
      Successfully uninstalled bleak-0.13.1a1
Successfully installed bleak-0.20.2 dbus-fast-1.95.2
21:30:47 INFO [bt] BT Discovery:
21:30:52 INFO [bt]  - no devices found - 
21:30:52 INFO [main] Bleak version 0.20.2, BtBackend version bluez-v5.70
21:30:52 INFO [main] connecting mqtt mqtt@10.0.1.20
21:30:52 INFO [main] Fetching 1 BMS + 0 virtual + 0 others serially, period=1.00s, keep_alive=True
21:30:52 INFO [sampling] connecting bms AntBt(24:E2:4F:A1:00:03,battery2)
21:30:52 INFO [main] mqtt watchdog loop started with timeout 300.0s
21:30:58 ERROR [bt] Device with address 24:E2:4F:A1:00:03 was not found., starting scanner
21:30:58 INFO [bt] BT Discovery:
21:31:03 INFO [bt]  - no devices found - 
21:31:03 INFO [ant] normal connect failed (Device with address 24:E2:4F:A1:00:03 was not found.), connecting with scanner
21:31:13 ERROR [sampling] AntBt(24:E2:4F:A1:00:03,battery2) device not found, retry in 1 seconds
END
```

Arnix11, 2023-12-20:
stalling bleak-0.13.1a1:
      Successfully uninstalled bleak-0.13.1a1
Successfully installed bleak-0.20.2 dbus-fast-1.95.2
21:32:58 INFO [main] Power cycle bluetooth hardware
Unable to open mgmt_socket
Unable to open mgmt_socket
21:33:01 INFO [bt] BT Discovery:
21:33:07 INFO [bt] BT 74:A0:FA:52:64:A0          74-A0-FA-52-64-A0
21:33:07 INFO [bt] BT 51:B4:77:42:BB:E0          51-B4-77-42-BB-E0
21:33:07 INFO [bt] BT A4:C1:38:A0:52:CE                 ATC_A052CE
21:33:07 INFO [main] Verbose logging enabled
21:33:07 INFO [main] Bleak version 0.20.2, BtBackend version bluez-v5.70
21:33:07 DEBUG [bt] BMS AntBt(24:E2:4F:A1:00:03,battery2) keep alive enabled
21:33:07 INFO [main] connecting mqtt mqtt@10.0.1.20
21:33:07 DEBUG [client] Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b''
21:33:07 INFO [main] Fetching 1 BMS + 0 virtual + 0 others serially, period=1.00s, keep_alive=True
21:33:07 INFO [sampling] connecting bms AntBt(24:E2:4F:A1:00:03,battery2)
21:33:07 INFO [bt] connecting battery2 (24:E2:4F:A1:00:03) adapter=default timeout=6
21:33:07 INFO [main] mqtt watchdog loop started with timeout 300.0s
21:33:07 DEBUG [client] Connecting to device @ 24:E2:4F:A1:00:03
21:33:07 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_fast.signature.Variant ('b', True)>}, []]
21:33:07 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_74_A0_FA_52_64_A0', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_fast.signature.Variant ('s', 74:A0:FA:52:64:A0)>, 'AddressType': <dbus_fast.signature.Variant ('s', random)>, 'Alias': <dbus_fast.signature.Variant ('s', 74-A0-FA-52-64-A0)>, 'Paired': <dbus_fast.signature.Variant ('b', False)>, 'Bonded': <dbus_fast.signature.Variant ('b', False)>, 'Trusted': <dbus_fast.signature.Variant ('b', False)>, 'Blocked': <dbus_fast.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_fast.signature.Variant ('b', False)>, 'RSSI': <dbus_fast.signature.Variant ('n', -57)>, 'Connected': <dbus_fast.signature.Variant ('b', False)>, 'UUIDs': <dbus_fast.signature.Variant ('as', ['0000fe9f-0000-1000-8000-00805f9b34fb'])>, 'Adapter': <dbus_fast.signature.Variant ('o', /org/bluez/hci0)>, 'ServiceData': <dbus_fast.signature.Variant ('a{sv}', {'0000fe9f-0000-1000-8000-00805f9b34fb': <dbus_fast.signature.Variant ('ay', bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'))>})>, 'ServicesResolved': <dbus_fast.signature.Variant ('b', False)>, 'AdvertisingFlags': <dbus_fast.signature.Variant ('ay', bytearray(b'\x00'))>}, 'org.freedesktop.DBus.Properties': {}}]
21:33:07 DEBUG [client] Received CONNACK (0, 0)
21:33:07 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_51_B4_77_42_BB_E0', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_fast.signature.Variant ('s', 51:B4:77:42:BB:E0)>, 'AddressType': <dbus_fast.signature.Variant ('s', random)>, 'Alias': <dbus_fast.signature.Variant ('s', 51-B4-77-42-BB-E0)>, 'Paired': <dbus_fast.signature.Variant ('b', False)>, 'Bonded': <dbus_fast.signature.Variant ('b', False)>, 'Trusted': <dbus_fast.signature.Variant ('b', False)>, 'Blocked': <dbus_fast.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_fast.signature.Variant ('b', False)>, 'RSSI': <dbus_fast.signature.Variant ('n', -62)>, 'Connected': <dbus_fast.signature.Variant ('b', False)>, 'UUIDs': <dbus_fast.signature.Variant ('as', ['0000fef3-0000-1000-8000-00805f9b34fb'])>, 'Adapter': <dbus_fast.signature.Variant ('o', /org/bluez/hci0)>, 'ServicesResolved': <dbus_fast.signature.Variant ('b', False)>, 'AdvertisingFlags': <dbus_fast.signature.Variant ('ay', bytearray(b'\x00'))>}, 'org.freedesktop.DBus.Properties': {}}]
21:33:07 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_74_A0_FA_52_64_A0): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -56)>}, []]

Arnix11, 2024-02-19:
Now its work fine! Thank You!
PS: Can You add the Junctek (KG-F KH-F) battery monitor in a supported models of  this super batmon integration?

fl4p, 2024-02-20:
I dont have a Junctek and without the HW it is difficult.
Do you now any code project that can read this monitor?

closed 2024-02-20 by fl4p

Arnix11, 2024-02-20:
Please try look this:
https://community.home-assistant.io/t/juntek-kg-f-series-on-ha/443741