fl4p/batmon-ha #181: Balance Switch not available in JK BMS

arzaman, 2023-12-22:
I have compelted the integration via BT with my JKBMS and I'm not able to get status of the "balance" switch
just charge/discharge swtich are visible

[image]

bms type and version
[image]

batmon 1.75 on RPI 3 model B 

[image]

let me know if it's possible to add
thanks
Davide

fl4p, 2023-12-23:
Hi, I didn't implement it, for sure it's possible

arzaman, 2023-12-24:
Ok I understood 
could be useful, not essential but in case you can add in the next release is appreciated
thanks
Davide

dougle03, 2023-12-29:
> Ok I understood could be useful, not essential but in case you can add in the next release is appreciated thanks Davide

Hey, sorry to bother you, but do you have a copy of the yaml for the JK-BMS project that UKSA007 has now taken down? I saw in your comment to him and that he's now taken the code down to sell it privately..? Sorry to contact you on this post but I can't find another way to contact you. Thanks

fl4p, 2024-01-08:
added balance switch reading in https://github.com/fl4p/batmon-ha/commit/22dcf47960838b0cbf5b43b1e879198e9015048a

setting the balance switch was already there.

will push this with the next update

fl4p, 2024-01-17:
added with release 1.76

re-open if neede

closed 2024-01-17 by fl4p