fl4p/batmon-ha #182: Missing SOC and Temp info. Voltages are fine

aletmp, 2024-01-02:
Using JK BMS with MQTT and bluetooth. 
It was working fine, no issues, but in the last few weeks I cannot get the all the info. I'm missing SOC and Temps. Tried to restore backups in HA but still the problem persist. I tried different configurations and made no effect. I have no clue what the issue could be. This is from MQTT Explorer:

core-mosquitto.local.hass.io
$SYS
brokerBateria_LIFEPO4
soc
total_voltage = 0.0
current = 0.0
balance_current = 0.0
soc_percent = 0
power = 0.0
capacity = 0.0
cycle_capacity = 357.48
num_cycles = 53555
mosfet_status
capacity_ah = 11534
temperature = 0.0
bms
uptime = 11141290
meter
sample_count = 531
total_charge = -216.6
total_energy = -27.09
total_energy_discharge = 587.2
total_energy_charge = 614.3
total_cycles = 35.74
total_abs_diff_charge = 21750.0
switch
cell_voltages
1 = 3.348
2 = 3.348
3 = 3.348
4 = 3.344
5 = 3.346
6 = 3.346
7 = 3.346
8 = 3.346
9 = 3.344
10 = 3.347
11 = 3.349
12 = 3.346
13 = 3.348
14 = 3.347
15 = 3.347
16 = 3.346
min = 3.344
min_index = 4
max = 3.349
max_index = 11
delta = 0.005
average = 3.347
median = 3.3465
temperatures
1 = 0.0
2 = 0.0

---------------
BEGIN 

09:38:37 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_59_31_42_5A_5F_FB): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -80)>}, []]
09:38:37 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_EE_B6_D7/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xb0\x00\x00\x00\x00\x00'))>}, []]
09:38:37 DEBUG [jikong] bms msg(22) (buf150): 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 b0 0 0 0 0 0

09:38:37 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_5C_47_57_5A_16_9F): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -68)>}, []]
09:38:37 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_EE_B6_D7/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'&\xd1\x00\x00\x10\x94\x03\x00\x1c\x11\x00\x00\xaa\x00\xaa\x00\x00\x00\x00\x00\x00\x00\x00D\xa8(\x03\x00\x80\xa3\x04\x00:\x00\x00\x00\x14\x95\x0f\x01d\x00\x00\x00$\xe8\xe1\x00\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\x01\x00\x00\x00\x13\x04\x15\x00\x00\x00\x8d\xfd@@\x00\x00\x00\x00\xea\x14\x00\x00\x00\x01\x00\x01\x00\x05\x00\x00p\xcb0\x00\x00\x00\x00\x00\xb0\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'))>}, []]
09:38:37 DEBUG [jikong] bms msg(128) (buf278): 26 d1 0 0 10 94 3 0 1c 11 0 0 aa 0 aa 0 0 0 0 0 0 0 0 44 a8 28 3 0 80 a3 4 0 3a 0 0 0 14 95 f 1 64 0 0 0 24 e8 e1 0 1 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ff 0 1 0 0 0 13 4 15 0 0 0 8d fd 40 40 0 0 0 0 ea 14 0 0 0 1 0 1 0 5 0 0 70 cb 30 0 0 0 0 0 b0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0

09:38:37 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_EE_B6_D7/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\xff\x7f\xdc/\x01\x01\x01\x00\x00\x00\x00\x8a'))>}, []]
09:38:37 DEBUG [jikong] bms msg(22) (buf300): 0 0 0 0 0 0 0 0 0 fe ff 7f dc 2f 1 1 1 0 0 0 0 8a
[... 59 lines cut ...]
09:38:38 INFO [sampling] Bateria_LIFEPO4 volt=[3347,3346,3346,3343,3346,3348,3347,3346,3346,3346,3349,3347,3349,3348,3347,3347] temp=[0.0, 0.0]
09:38:38 INFO [sampling] JKBt(C8:47:8C:EE:B6:D7,Bateria_LIFEPO4) times: connect=0.00s fetch=0.15s
09:38:38 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_5C_47_57_5A_16_9F): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -70)>}, []]
09:38:38 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_78_BD_BC_73_FB_D3): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -52)>}, []]
09:38:38 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_59_31_42_5A_5F_FB): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -75)>}, []]
09:38:38 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_EE_B6_D7/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'U\xaa\xeb\x90\x02g\x13\r\x12\r\x13\r\x10\r\x12\r\x14\r\x13\r\x12\r\x12\r\x13\r\x15\r\x12\r\x14\r\x13\r\x13\r\x10\r\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00\x12\r\x05\x00\x05\x0f>\x005\x005\x005\x005\x006\x003\x003\x003\x003\x003\x003\x003\x004\x004\x003\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'))>}, []]
09:38:38 DEBUG [jikong] header, clear buf bytearray(b'')
09:38:38 DEBUG [jikong] bms msg(128) (buf128): 55 aa eb 90 2 67 13 d 12 d 13 d 10 d 12 d 14 d 13 d 12 d 12 d 13 d 15 d 12 d 14 d 13 d 13 d 10 d 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ff ff 0 0 12 d 5 0 5 f 3e 0 35 0 35 0 35 0 35 0 36 0 33 0 33 0 33 0 33 0 33 0 33 0 33 0 34 0 34 0 33 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0

09:38:38 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_EE_B6_D7/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xb0\x00\x00\x00\x00\x00'))>}, []]
09:38:38 DEBUG [jikong] bms msg(22) (buf150): 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 b0 0 0 0 0 0

09:38:38 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_EE_B6_D7/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'"\xd1\x00\x00\x00\x94\x03\x00\x1c\x11\x00\x00\xaa\x00\xaa\x00\x00\x00\x00\x00\x00\x00\x00D\xa9(\x03\x00\x80\xa3\x04\x00:\x00\x00\x00\x14\x95\x0f\x01d\x00\x00\x00%\xe8\xe1\x00\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\x01\x00\x00\x00\x13\x04\x15\x00\x00\x00\x8d\xfd@@\x00\x00\x00\x00\xe9\x14\x00\x00\x00\x01\x00\x01\x00\x05\x00\x00{\xcb0\x00\x00\x00\x00\x00\xb0\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'))>}, []]
09:38:38 DEBUG [jikong] bms msg(128) (buf278): 22 d1 0 0 0 94 3 0 1c 11 0 0 aa 0 aa 0 0 0 0 0 0 0 0 44 a9 28 3 0 80 a3 4 0 3a 0 0 0 14 95 f 1 64 0 0 0 25 e8 e1 0 1 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ff 0 1 0 0 0 13 4 15 0 0 0 8d fd 40 40 0 0 0 0 e9 14 0 0 0 1 0 1 0 5 0 0 7b cb 30 0 0 0 0 0 b0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0

09:38:38 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_EE_B6_D7/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\xff\x7f\xdc/\x01\x01\x01\x00\x00\x00\x00\x85'))>}, []]
09:38:38 DEBUG [jikong] bms msg(22) (buf300): 0 0 0 0 0 0 0 0 0 fe ff 7f dc 2f 1 1 1 0 0 0 0 85

09:38:38 DEBUG [jikong] got response 2 (len300)
09:38:38 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_5C_47_57_5A_16_9F): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -76)>}, []]
09:38:39 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_5C_47_57_5A_16_9F): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -77)>}, []]
09:38:39 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_78_BD_BC_73_FB_D3): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -56)>}, []]

END
```

snascime, 2024-01-14:
Same issue for me.

gianfrdp, 2024-01-17:
Same problem JK_B2A20S20P, 16s 280 Ah
No SoC, no temperatures and no Total Battery Voltage

Using esphome-jk-bms no problem.

[image]

pedrovelaz, 2024-01-23:
Same issue for me. Bluetooth in hasio, debian laptop. No SOC,no voltaje........

Downgrade??

aletmp, 2024-01-23:
Downgrade didn't work for me. No idea what the problem is... 
I'm using Raspberry 3b and also tried with another raspberry 4 with fresh installation and got the same issue straight away

philtechsl, 2024-02-17:
Same problem for me as well :(

[image]

mikeleord, 2024-02-19:
same problem for me too

snascime, 2024-02-22:
@fl4p for me is working now but at the first moment I was getting errors like this. "paho.mqtt.client' has no attribute 'callbackApi Version", so after reading this page https://www.emqx.com/en/blog/how-to-use-mqtt-in-python I replace the following line and added another one. in main.py file.
        #mqtt_client = paho.Client(paho.CallbackAPIVersion.VERSION1) => commented
        client_id = f'python-mqtt-{random.randint(0, 100)}'   => added this new line
        mqtt_client = paho.Client(client_id)  => new value for the varibale

After these changes, I can see all values as expected for JK BMS.

fl4p, 2024-02-22:
@snascime update your paho version to 2.0

philtechsl, 2024-02-24:
> [bd60195](https://github.com/fl4p/batmon-ha/commit/bd601954b6d05013ebe4741915fd82008f8a95a6) should fix it

I still have the problem and I don´t know ho to apply this as I am quite new with home assistant :(

Regards

philtechsl, 2024-03-09:
> Same problem for me as well :(
> 
> ![JK](https://private-user-images.githubusercontent.com/120264840/305675337-f8ec0537-cd1f-4d81-9d53-3cb653f259ef.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk5ODUxMzYsIm5iZiI6MTcwOTk4NDgzNiwicGF0aCI6Ii8xMjAyNjQ4NDAvMzA1Njc1MzM3LWY4ZWMwNTM3LWNkMWYtNGQ4MS05ZDUzLTNjYjY1M2YyNTllZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMwOVQxMTQ3MTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03YzQ0NmViYmZlNGM0MzdjNjkxNjAxMzlkYzI4ZDM3MTA1NTFiZjYzMzVkOGYxNDIzNWI5ODBkYThhZDJmYzJjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.VxUjKbgjBxt5tR8BKVB27rd_nJOWelLr-dEnwqIWeSQ)

Still same problem :(

crunchtime2k, 2024-03-12:
same issue here on a new install of batmon 3/12/2024 with a JBD bms , temperatures populate intermittently , showing correctly and then unavailable then correct , so on and so on.

donulf, 2024-03-21:
hello do you guys have news?

EricGrosfeld, 2024-03-30:
> D bms , temperatures populate intermittently

New to Home assistant. I do not know how to apply the suggested fix... but my system is depending on it. @fl4p , your assistant is much appreciated and i'm willing to reward your help if nessecary.....

donulf, 2024-03-31:
I can't even make it work. I too would be willing to contribute financially for a solution, not to mention the fact that I have already contributed in the past. However, it seems to me that @fl4p  hasn't looked at messages for a while

philtechsl, 2024-05-02:
Have anyone found a solution for this issue? I still have the same problem

alejandropradocabrero, 2024-05-02:
I couldn't find any solution yet. I don't really know how to do the workaround suggested, tbh.

EricGrosfeld, 2024-05-04:
I can verify a work around as also mentioned earlier in this trail. 

Use the JK BMS app on your phone (I use th android version) to connect to the BMS via bluetooth.

Then go to setting. (Enter password if necessary). Change the battery type to an other profile then the current one, and then change it back to the original profile (in my case lifepo4).

Do not forget to update the parameters like cell count and capacity because these setting are changed to the factory default in the step above.

After this, all setting are again received and visible in home assistant.

Cheers,
Eric

philtechsl, 2024-05-15:
> I can verify a work around as also mentioned earlier in this trail.
>
> Use the JK BMS app on your phone (I use th android version) to connect to the BMS via bluetooth.
>
> Then go to setting. (Enter [...]

Thanks, It did work