fl4p/batmon-ha #184: The command '/bin/ash -o pipefail -c apk add python3' returned a non-zero code: 1

Serebrin, 2024-01-17:
Describe the issue here.
If you experience bluetooth connection problems, follow these steps before opening an issue:
https://github.com/fl4p/batmon-ha#troubleshooting

* Include exact BMS model name
* Query HW and SW numbers from the BMS app
* Enable verbose_log

```
Paste log output between BEGIN and END:
BEGIN 

END
```

stanfordcg, 2024-01-17:
similar error with 1.76

Core 2024.1.3
Supervisor 2023.12.1
Operating System 11.4     
Home Assistant Blue

Failed to install add-on
The command '/bin/ash -o pipefail -c apk add py-pip' returned a non-zero code: 1

fl4p, 2024-01-28:
check supervisor logs at http://homeassistant.local:8123/config/logs right after the error happens

gekochtesei, 2024-12-08:
seems to have the same error with 1.78 on OS 13.2
I checked with two different installations. PI 5 with NVMe Hat SSD and PI 3B SD
here the logs as requested:

_# warning in UI ########
Logger: bluetooth_auto_recovery.recover
Quelle: runner.py:154
Erstmals aufgetreten: 00:48:44 (2 Vorkommnisse)
Zuletzt protokolliert: 07:58:44

Could not reset the power state of the Bluetooth adapter hci0 [B8:27:EB:BD:C2:6D] due to timeout after 5 seconds

# error in UI ##########
Logger: homeassistant.components.hassio
Quelle: components/hassio/websocket_api.py:135
Integration: Home Assistant Supervisor (Dokumentation, Probleme)
Erstmals aufgetreten: 00:39:13 (3 Vorkommnisse)
Zuletzt protokolliert: 00:49:52

Failed to to call /addons/2af0a32d_batmon/install - The command '/bin/ash -o pipefail -c venv/bin/pip3 install -r requirements.txt' returned a non-zero code: 1

# log download #########
2024-12-08 00:39:13.148 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /addons/2af0a32d_batmon/install - The command '/bin/ash -o pipefail -c venv/bin/pip3 install -r requirements.txt' returned a non-zero code: 1
2024-12-08 00:45:16.585 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /addons/2af0a32d_batmon/install - The command '/bin/ash -o pipefail -c venv/bin/pip3 install -r requirements.txt' returned a non-zero code: 1
2024-12-08 00:48:44.834 WARNING (MainThread) [bluetooth_auto_recovery.recover] Could not reset the power state of the Bluetooth adapter hci0 [B8:27:EB:BD:C2:6D] due to timeout after 5 seconds
2024-12-08 00:49:52.476 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /addons/2af0a32d_batmon/install - The command '/bin/ash -o pipefail -c venv/bin/pip3 install -r requirements.txt' returned a non-zero code: 1
2024-12-08 07:58:44.265 WARNING (MainThread) [bluetooth_auto_recovery.recover] Could not reset the power state of the Bluetooth adapter hci0 [B8:27:EB:BD:C2:6D] due to timeout after 5 seconds_

fl4p, 2024-12-09:
@gekochtesei you seem to have 2 issues, one with the docker container setup and the other with your bluetooth hardware.
To debug the bluetooth issue, install the terminal add-on and try these commands:
```
bluetoothctl power off
bluetoothctl power on
```

gekochtesei, 2024-12-09:
@fl4p first of all ... many thanks for your support.
Regarding the bluetooth topic ... after powering off and on via shell, this issue dosen't seems to occur anymore.
Regarding the docker topic I can post here the supervisor log - hopefully this helps a bit more:
_2024-12-09 18:04:01.589 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on homeassistant/aarch64-addon-ssh with version 9.15.0
2024-12-09 18:04:06.025 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.1:8099 ssl:default [Connect call failed ('172.30.33.1', 8099)]
2024-12-09 18:04:12.650 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from core_ssh
2024-12-09 18:04:12.660 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from core_ssh
2024-12-09 18:04:12.671 INFO (MainThread) [supervisor.api.middleware.security] /host/info access from core_ssh
2024-12-09 18:04:12.681 INFO (MainThread) [supervisor.api.middleware.security] /core/info access from core_ssh
2024-12-09 18:05:50.243 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2024-12-09 18:07:14.839 INFO (MainThread) [supervisor.addons.addon] No 2af0a32d_batmon addon Docker image 2af0a32d/aarch64-addon-batmon found
2024-12-09 18:07:15.635 INFO (MainThread) [supervisor.docker.addon] Starting build for 2af0a32d/aarch64-addon-batmon:1.78
2024-12-09 18:08:26.965 ERROR (MainThread) [supervisor.docker.addon] Can't build 2af0a32d/aarch64-addon-batmon:1.78: The command '/bin/ash -o pipefail -c venv/bin/pip3 install -r requirements.txt' returned a non-zero code: 1
2024-12-09 18:08:26.969 ERROR (MainThread) [supervisor.docker.addon] Build log: 
Step 1/19 : ARG BUILD_FROM
Step 2/19 : FROM $BUILD_FROM
 ---> f0d9ed03fd87
Step 3/19 : WORKDIR /app
 ---> Using cache
 ---> 445eac8c0b07
Step 4/19 : RUN apk add python3
 ---> Using cache
 ---> 1437d50db10c
Step 5/19 : RUN apk add bluez
 ---> Using cache
 ---> a61d345ffa94
Step 6/19 : RUN apk add py-pip
 ---> Using cache
 ---> 50a13237a168
Step 7/19 : RUN apk add git
 ---> Using cache
 ---> 2e6eb3bf8b2d
Step 8/19 : COPY . .
 ---> Using cache
 ---> bf8ee18a55ef
Step 9/19 : RUN python3 -m venv venv
 ---> Using cache
 ---> 0075052b4953
Step 10/19 : RUN venv/bin/pip3 install -r requirements.txt
 ---> Running in e76e229b8dc6
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Collecting git+https://github.com/jpeters-ml/bleak@feature/windowsPairing (from -r requirements.txt (line 4))
  Cloning https://github.com/jpeters-ml/bleak (to revision feature/windowsPairing) to /tmp/pip-req-build-w2djeezc
  Running command git clone --filter=blob:none --quiet https://github.com/jpeters-ml/bleak /tmp/pip-req-build-w2djeezc
  fatal: unable to access 'https://github.com/jpeters-ml/bleak/': Could not resolve host: github.com
  fatal: could not fetch d4a2c4405ec2e962c521a13af91bf5f7098a62a8 from promisor remote
  warning: Clone succeeded, but checkout failed.
  You can inspect what was checked out with 'git status'
  and retry with 'git restore --source=HEAD :/'
  error: subprocess-exited-with-error
  × git clone --filter=blob:none --quiet https://github.com/jpeters-ml/bleak /tmp/pip-req-build-w2djeezc did not run successfully.
  │ exit code: 128
  ╰─> See above for output.
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error
× git clone --filter=blob:none --quiet https://github.com/jpeters-ml/bleak /tmp/pip-req-build-w2djeezc did not run successfully.
│ exit code: 128
╰─> See above for output.
note: This error originates from a subprocess, and is likely not a problem with pip.
 ---> Removed intermediate container e76e229b8dc6_

thanks in advance

fl4p, 2024-12-09:
looks like sth is preventing internet or DNS access to resolve 'github.com'

gekochtesei, 2024-12-09:
mmhh ... if I just do a wget on "https://github.com/jpeters-ml/bleak/" i got:
_Connecting to github.com (140.82.121.4:443)
saving to index.html_

fl4p, 2024-12-09:
looks like only during docker build DNS look-up fails
maybe this is related https://community.home-assistant.io/t/dns-error-unable-to-resolve-host/524970/33?page=2

its not a batmon issue, thats pretty sure

poligor84, 2025-01-09:
Good afternoon. Did you manage to solve your problem? During the installation, I get the following error: The command '/bin/ash -o pipefail -c venv/bin/pip3 install -r requirements.txt' returned a non-zero code: 1
What do you advise me to do? Thank you.

fl4p, 2025-02-03:
Hi,
please try again and then go to Settings / System / Logs / Supervisor to see more details about the error and post it here

poligor84, 2025-02-03:
> The command '/bin/ash -o pipefail -c venv_bleak_pairing/bin/pip3 install
> returned a non-zero code: 1

Не вдалося встановити доповнення

The command '/bin/ash -o pipefail -c venv_bleak_pairing/bin/pip3 install
***@***.***/windowsPairing'' returned
a non-zero code: 1
[image: image.png]

poligor84, 2025-02-03:
[image: image.png]

poligor84, 2025-02-03:
+++

fl4p, 2025-02-04:
@poligor84  The image didnt go through. Try to reply through the website

poligor84, 2025-02-04:
It worked.