fl4p/batmon-ha #189: BMS JK-B2A20S20P connecting issue

kzajac83, 2024-01-23:
Hello.

I got BMS JK-B2A20S20P  (2A Balancing, up to 20 series, 200A) but cannot connect.

What I understood there are JK02 and JK04 protocols for Jikong BMS, where JK04 is not implemented fully here. First of all how to check does my BMS using JK02 or JK04 protocol? 

Additional not sure about PIN because when I connect via application on smartphone then is need provide PIN 1st time.  Please note I changed ping from default 1234 to 12345 - this is important?

Next question is regarding MAC address.  [scanner] section in logs give me two MAC addresses but both not work, this mean I see lot's errors in py files in logs for both MAC.

General question is possible connect my BMS into batmon? 

Model: JK-B2A20S20P
HW: V10.XG
SW: V10.07

Scanner provide 2 MAC addresses to me:
DC:23:4D:EB:B9:E4 TY
DC:23:4E:0A:8A:62  TY

BTW. Regarding logs - I cannot provide full log because is overwriting in batmon log tab and system -> logs -> batmon too even if I download full logs. 

```
Paste log output between BEGIN and END for DC:23:4D:EB:B9:E4
BEGIN 

  File "/app/main.py", line 355, in fn
    await t()
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 227, in _sample_inner
    async with bms:
  File "/app/bmslib/bt.py", line 345, in __aenter__
    await self.connect()
  File "/app/bmslib/models/jikong.py", line 130, in connect
    await self._connect_with_scanner(timeout=timeout)
  File "/app/bmslib/bt.py", line 273, in _connect_with_scanner
    await self._connect_client(timeout=timeout / 2)
  File "/app/bmslib/bt.py", line 186, in _connect_client
    await asyncio.wait_for(self.client.connect(timeout=timeout), timeout=timeout + 1)
  File "/usr/lib/python3.11/asyncio/tasks.py", line 489, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/app/venv/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 348, in connect
    reply = await asyncio.wait_for(
            ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/tasks.py", line 502, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError

23:50:38 INFO [sampling] connecting bms JKBt(DC:23:4D:EB:B9:E4,battery1)
23:50:38 INFO [bt] connecting battery1 (DC:23:4D:EB:B9:E4) adapter=default timeout=6
23:50:38 DEBUG [client] Connecting to device @ DC:23:4D:EB:B9:E4 with hci0
23:50:43 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_DC_23_4D_EB_B9_E4): ['org.bluez.Device1', {'Connected': <dbus_next.signature.Variant ('b', True)>}, []]
23:50:44 DEBUG [client] Connection successful (/org/bluez/hci0/dev_DC_23_4D_EB_B9_E4)
23:50:44 DEBUG [client] Waiting for ServicesResolved (/org/bluez/hci0/dev_DC_23_4D_EB_B9_E4)
23:50:44 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_DC_23_4D_EB_B9_E4): ['org.bluez.Device1', {'Name': <dbus_next.signature.Variant ('s', COMMON)>, 'Alias': <dbus_next.signature.Variant ('s', COMMON)>}, []]
23:50:45 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_DC_23_4D_EB_B9_E4): ['org.bluez.Device1', {'ServicesResolved': <dbus_next.signature.Variant ('b', True)>}, []]
23:50:45 INFO [bt] [Service] 00001910-0000-1000-8000-00805f9b34fb (Handle: 15): Vendor specific
23:50:45 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_DC_23_4D_EB_B9_E4/service000f/char0014): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')>}, []]
23:50:45 DEBUG [client] Read Characteristic 00002b11-0000-1000-8000-00805f9b34fb | /org/bluez/hci0/dev_DC_23_4D_EB_B9_E4/service000f/char0014: bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
23:50:45 INFO [bt] 	[Characteristic] 00002b11-0000-1000-8000-00805f9b34fb (Handle: 20): Temperature Statistics (read,write-without-response,write), Value: b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
23:50:45 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_DC_23_4D_EB_B9_E4/service000f/char0014/desc0016): ['org.bluez.GattDescriptor1', {'Value': <dbus_next.signature.Variant ('ay', b'Phone->Module\x00')>}, []]
23:50:45 DEBUG [client] Read Descriptor 22 | /org/bluez/hci0/dev_DC_23_4D_EB_B9_E4/service000f/char0014/desc0016: bytearray(b'Phone->Module\x00')
23:50:45 INFO [bt] 		[Descriptor] 00002901-0000-1000-8000-00805f9b34fb (Handle: 22): Characteristic User Description) | Value: b'Phone->Module\x00'
23:50:45 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_DC_23_4D_EB_B9_E4/service000f/char0010): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\x00')>}, []]
23:50:45 DEBUG [client] Read Characteristic 00002b10-0000-1000-8000-00805f9b34fb | /org/bluez/hci0/dev_DC_23_4D_EB_B9_E4/service000f/char0010: bytearray(b'\x00')
23:50:45 INFO [bt] 	[Characteristic] 00002b10-0000-1000-8000-00805f9b34fb (Handle: 16): Temperature Range (read,notify), Value: b'\x00'
23:50:45 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_DC_23_4D_EB_B9_E4/service000f/char0010/desc0013): ['org.bluez.GattDescriptor1', {'Value': <dbus_next.signature.Variant ('ay', b'Module->Phone\x00')>}, []]
[... 142 lines cut ...]

Traceback (most recent call last):
  File "/app/main.py", line 44, in fetch_loop
    if await fn():
       ^^^^^^^^^^
  File "/app/main.py", line 360, in fn
    raise exceptions[0]
  File "/app/main.py", line 355, in fn
    await t()
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 237, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/jikong.py", line 244, in fetch
    await self._fetch_futures.wait_for(0x02, self.TIMEOUT)
  File "/app/bmslib/__init__.py", line 84, in wait_for
    raise asyncio.TimeoutError("timeout waiting for %s" % name)
TimeoutError: timeout waiting for 2

23:53:57 WARNING [sampling] battery1 error fetching device info: 3

END
```

kzajac83, 2024-01-29:
Thank you, is working fine now :)

BTW, maybe you now, what mean number "cell index max" and "cell index min" (without UoM)? Additional "num_cycles", "num_samples" both in UoM [N] and "total cycles meter" with value "0,0 N" ? General what mean UoM [N]? 

PS. All energy [kWh] entities and one capacity [Ah] value display in wrong format but will create other issue for that (or keep here?). Coma is in wrong place.

6bishnoi, 2024-01-30:
good,

cell index max : is the cell number witch cell has the lowest Voltage
cell index min: is the cell with the highest V

If you have VOLTRONIC basis inverter, you can try this new add-on: https://github.com/jean-luc1203/smartphoton-ha-addon/

I don’t now meters…