fl4p/batmon-ha #195: paho-mqtt  issue

peteretep, 2024-02-11:
Hello, and thanks for your work on this.
I downloaded and tried to use this add on today.
It seemed to find my battery but failed with a paho mqtt error.
I suspect it is related to the recent update of paho-mqtt

I suggest you pin paho in the requirements.txt file as `paho-mqtt==1.6.1`

I don't have my battery in the house (it's down at my boat), so I haven't been able to test it.

I'm hoping to be able to monitor a Fogstar Drift battery.

Traceback is here.

```
Exception ignored in: <function Client.__del__ at 0x7f8682ff60>
Traceback (most recent call last):
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 874, in __del__
    self._reset_sockets()
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1133, in _reset_sockets
    self._sock_close()
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1119, in _sock_close
    if not self._sock:
           ^^^^^^^^^^
AttributeError: 'Client' object has no attribute '_sock'
16:23:09 ERROR [main] Main loop exception: Client.__init__() missing 1 required positional argument: 'callback_api_version'
16:23:09 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 400, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/app/main.py", line 235, in main
    mqtt_client = paho.Client()
                  ^^^^^^^^^^^^^
TypeError: Client.__init__() missing 1 required positional argument: 'callback_api_version'

16:23:09 INFO [main] exit signal handler... (), {}, shutdown was False
s6-rc: info: service legacy-services: stopping
```

Igypop78, 2024-02-11:
I have the same problem, I was using this addon for quite a while but after last update I get this errors
`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
00:05:27 INFO [bt] BT Discovery:
00:05:32 INFO [bt] BT 28:12:EB:60:15:C3          28-12-EB-60-15-C3
00:05:32 INFO [bt] BT 7C:87:CE:65:96:9E ShellyPlus1PM-7C87CE65969C
00:05:32 INFO [bt] BT A4:C1:38:8C:56:CE                   MiKetv12
00:05:32 INFO [bt] BT E7:E7:88:0E:A7:86                S14 0C37 LE
00:05:32 INFO [bt] BT 06:8B:E5:A5:B1:75          06-8B-E5-A5-B1-75
00:05:32 INFO [bt] BT 4E:EA:BA:C0:BC:C7          4E-EA-BA-C0-BC-C7
00:05:32 INFO [bt] BT 74:B8:39:58:DB:12         Sc2f3bdce88d87cf9C
00:05:32 INFO [bt] BT A4:C1:37:00:30:A5               Rack Battery
00:05:32 INFO [bt] BT CB:01:93:9A:37:F0 BSC IP67 12/17 HQ2126AHWFT
00:05:32 INFO [bt] BT FF:A9:3D:81:5B:FE          FF-A9-3D-81-5B-FE
00:05:32 INFO [bt] BT 6A:3C:F3:38:F5:FB          6A-3C-F3-38-F5-FB
00:05:32 INFO [bt] BT 6B:62:D2:18:EB:9C          6B-62-D2-18-EB-9C
00:05:32 INFO [bt] BT 5D:AF:02:7B:A3:59          5D-AF-02-7B-A3-59
00:05:32 INFO [bt] BT FC:C1:BD:83:46:77              Amazfit GTR 2
00:05:32 INFO [bt] BT D0:03:DF:F4:C1:AB          D0-03-DF-F4-C1-AB
00:05:32 INFO [bt] BT C8:47:8C:EC:BD:09              Small Battery
00:05:32 INFO [bt] BT D4:A3:BE:07:BE:BD          D4-A3-BE-07-BE-BD
00:05:32 INFO [bt] BT 5E:92:F0:EC:65:A9          5E-92-F0-EC-65-A9
00:05:32 INFO [bt] BT C8:47:8C:E4:56:69               Main Battery
00:05:32 INFO [bt] BT BC:7E:8B:14:85:4F [TV] Samsung Q60 Series (85)
00:05:32 INFO [bt] BT 7F:22:64:A0:AC:EE                  Smart Tag
00:05:32 INFO [bt] BT A4:C1:38:53:2E:D5          A4-C1-38-53-2E-D5
00:05:32 INFO [bt] BT 72:95:FC:34:26:79          72-95-FC-34-26-79
00:05:32 INFO [bt] BT 74:F7:1D:D6:5D:96          74-F7-1D-D6-5D-96
00:05:32 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.70
00:05:32 INFO [main] connecting mqtt mosquitto@core-mosquitto
Exception ignored in: <function Client.__del__ at 0x7f81209b3f60>
Traceback (most recent call last):
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 874, in __del__
    self._reset_sockets()
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1133, in _reset_sockets
    self._sock_close()
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1119, in _sock_close
    if not self._sock:
           ^^^^^^^^^^
AttributeError: 'Client' object has no attribute '_sock'
00:05:32 ERROR [main] Main loop exception: Client.__init__() missing 1 required positional argument: 'callback_api_version'
00:05:32 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 400, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/app/main.py", line 235, in main
    mqtt_client = paho.Client()
                  ^^^^^^^^^^^^^
TypeError: Client.__init__() missing 1 required positional argument: 'callback_api_version'

00:05:32 INFO [main] exit signal handler... (), {}, shutdown was False
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped`

peteretep, 2024-02-12:
I will try to get a pull request organised this week, it's a simple change
that's needed.

Migrations — Eclipse paho-mqtt documentation
<https://eclipse.dev/paho/files/paho.mqtt.python/html/migrations.html>

g1gabyt3, 2024-02-12:
@peteretep hopefully your commit gets merged in quickly I'm having this same issue too.

philtechsl, 2024-02-12:
Me too, I am having the same issue:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
23:26:20 INFO [bt] BT Discovery:
23:26:25 INFO [bt] BT C8:47:80:05:86:DA               JK_B2A24S15P
23:26:25 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.70
23:26:25 INFO [main] connecting mqtt addons@core-mosquitto
Exception ignored in: <function Client.__del__ at 0x7fa605ff60>
Traceback (most recent call last):
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 874, in __del__
    self._reset_sockets()
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1133, in _reset_sockets
    self._sock_close()
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1119, in _sock_close
    if not self._sock:
           ^^^^^^^^^^
AttributeError: 'Client' object has no attribute '_sock'
23:26:25 ERROR [main] Main loop exception: Client.__init__() missing 1 required positional argument: 'callback_api_version'
23:26:25 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 400, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/app/main.py", line 235, in main
    mqtt_client = paho.Client()
                  ^^^^^^^^^^^^^
TypeError: Client.__init__() missing 1 required positional argument: 'callback_api_version'

23:26:25 INFO [main] exit signal handler... (), {}, shutdown was False
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

TroyDL, 2024-02-15:
I see that this PR was merged into master (awesome!); how exactly would I get the newest version of master into my home assistant?  I'm running the HAOS on a pi 4 b.  Do I need to wait for an official release or is there something I can do to jump ahead?

fl4p, 2024-02-16:
I'll push the update tonight