fl4p/batmon-ha #197: TypeError: Client.__init__() missing 1 required positional argument: 'callback_api_version'

asd343, 2024-02-13:
Daly BMS 250A 4S

```
Paste log output between BEGIN and END:
BEGIN 
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
19:00:29 INFO [main] Power cycle bluetooth hardware
Unable to open mgmt_socket
Unable to open mgmt_socket
19:00:33 INFO [bt] BT Discovery:
19:00:38 INFO [bt] BT 0A:89:B1:DD:E2:28          0A-89-B1-DD-E2-28
19:00:38 INFO [bt] BT 64:13:5B:8C:C3:37          64-13-5B-8C-C3-37
19:00:38 INFO [bt] BT 51:EA:50:11:B7:9F          51-EA-50-11-B7-9F
19:00:38 INFO [bt] BT 40:17:08:01:22:67              CamperLifePo4
19:00:38 INFO [bt] BT 6F:14:90:8B:76:EA          6F-14-90-8B-76-EA
19:00:38 INFO [bt] BT 72:78:2E:AA:B0:D6          72-78-2E-AA-B0-D6
19:00:38 INFO [bt] BT 4E:40:C3:F7:13:57          4E-40-C3-F7-13-57
19:00:38 INFO [bt] BT 41:D9:86:80:1A:E4          41-D9-86-80-1A-E4
19:00:38 INFO [bt] BT FA:D7:C0:A1:9E:D2          FA-D7-C0-A1-9E-D2
19:00:38 INFO [main] Verbose logging enabled
19:00:38 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.70
19:00:38 DEBUG [bt] BMS DalyBt(40:17:08:01:22:67,battery1) keep alive enabled
19:00:38 INFO [main] connecting mqtt addons@core-mosquitto
Exception ignored in: <function Client.__del__ at 0x7fa20f3f60>
Traceback (most recent call last):
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 874, in __del__
    self._reset_sockets()
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1133, in _reset_sockets
    self._sock_close()
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1119, in _sock_close
    if not self._sock:
           ^^^^^^^^^^
AttributeError: 'Client' object has no attribute '_sock'
19:00:38 ERROR [main] Main loop exception: Client.__init__() missing 1 required positional argument: 'callback_api_version'
19:00:38 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 400, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/app/main.py", line 235, in main
    mqtt_client = paho.Client()
                  ^^^^^^^^^^^^^
TypeError: Client.__init__() missing 1 required positional argument: 'callback_api_version'

19:00:38 INFO [main] exit signal handler... (), {}, shutdown was False
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
END
```

TroyDL, 2024-02-14:
Same issue here.

akrobots, 2024-02-14:
Downgrade the paho-mqtt < 2. Try, pip install paho-mqtt==1.6.1

peteretep, 2024-02-14:
Duplicate of #195 and sorted in #196

WolwX, 2024-02-14:
Same problem here...
Can we, revert update ?
It was OK before ....

`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
08:10:11 INFO [main] Power cycle bluetooth hardware
Unable to open mgmt_socket
Unable to open mgmt_socket
08:10:15 INFO [bt] BT Discovery:
08:10:20 INFO [bt] BT A4:C1:37:40:BB:14              xiaoxiang bms
08:10:20 INFO [bt] BT D4:AD:FC:78:7B:A6             GBK_H61A0_7BA6
08:10:20 INFO [bt] BT 6C:9A:BC:90:99:EA          6C-9A-BC-90-99-EA
08:10:20 INFO [bt] BT 57:42:E7:C2:44:35          57-42-E7-C2-44-35
08:10:20 INFO [bt] BT D9:2B:91:EA:9E:59           MPPT Geekomobile
08:10:20 INFO [bt] BT 96:69:08:05:29:C3            DL-9669080529C3
08:10:20 INFO [bt] BT 64:90:C1:95:1A:A1          64-90-C1-95-1A-A1
08:10:20 INFO [bt] BT 59:CD:E6:29:F1:69          59-CD-E6-29-F1-69
08:10:20 INFO [bt] BT A4:C1:38:5C:0D:7D                 LYWSD03MMC
08:10:20 INFO [bt] BT 4E:78:E8:34:4F:4D          4E-78-E8-34-4F-4D
08:10:20 INFO [bt] BT A4:C1:38:3C:6E:BB                 LYWSD03MMC
08:10:20 INFO [main] Verbose logging enabled
08:10:20 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.70
08:10:20 DEBUG [bt] BMS DalyBt(96:69:08:05:29:C3,batterie163ah) keep alive enabled
08:10:20 DEBUG [bt] BMS JbdBt(A4:C1:37:40:BB:14,batterie304ah) keep alive enabled
08:10:20 INFO [main] connecting mqtt mqtt-user@core-mosquitto
Exception ignored in: <function Client.__del__ at 0x7fbc9f3f60>
Traceback (most recent call last):
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 874, in __del__
    self._reset_sockets()
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1133, in _reset_sockets
    self._sock_close()
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1119, in _sock_close
    if not self._sock:
           ^^^^^^^^^^
AttributeError: 'Client' object has no attribute '_sock'
08:10:20 ERROR [main] Main loop exception: Client.__init__() missing 1 required positional argument: 'callback_api_version'
08:10:20 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 400, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/app/main.py", line 235, in main
    mqtt_client = paho.Client()
                  ^^^^^^^^^^^^^
TypeError: Client.__init__() missing 1 required positional argument: 'callback_api_version'

08:10:20 INFO [main] exit signal handler... (), {}, shutdown was False
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped`

ZS1FJK, 2024-02-15:
Same Issue here.
`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
02:08:56 INFO [bt] BT Discovery:
02:09:01 INFO [bt] BT A4:C1:38:E3:7F:9D                   R12V-258
02:09:01 INFO [bt] BT E2:D5:DD:68:61:0E          E2-D5-DD-68-61-0E
02:09:01 INFO [bt] BT A4:C1:38:35:68:81                   R12V-264
02:09:01 INFO [bt] BT EA:0F:DE:AC:5A:CE     SmartShunt HQ2113ZAEQN
02:09:01 INFO [bt] BT F4:3F:2C:CE:0A:98          F4-3F-2C-CE-0A-98
02:09:01 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.70
02:09:01 INFO [main] connecting mqtt addons@core-mosquitto
Exception ignored in: <function Client.__del__ at 0x7fbda5bf60>
Traceback (most recent call last):
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 874, in __del__
    self._reset_sockets()
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1133, in _reset_sockets
    self._sock_close()
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1119, in _sock_close
    if not self._sock:
           ^^^^^^^^^^
AttributeError: 'Client' object has no attribute '_sock'
02:09:01 ERROR [main] Main loop exception: Client.__init__() missing 1 required positional argument: 'callback_api_version'
02:09:01 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 400, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/app/main.py", line 235, in main
    mqtt_client = paho.Client()
                  ^^^^^^^^^^^^^
TypeError: Client.__init__() missing 1 required positional argument: 'callback_api_version'

02:09:01 INFO [main] exit signal handler... (), {}, shutdown was False
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped`

esther1112, 2024-02-15:
> Downgrade the paho-mqtt < 2. Try, pip install paho-mqtt==1.6.1

Maybe a silly question but as I'm fairly new to this ,where do I type the above. 
I have the same issue trying to connect jk bms to ha. I can't even get it to show in ha to put it on a dashboard... I've configured mqtt but still can't get it to work

ZS1FJK, 2024-02-15:
> Maybe a silly question but as I'm fairly new to this ,where do I type the above. I have the same issue trying to connect jk bms to ha. I can't even get it to show in ha to put it on a dashboard... [...]

They have opened a pull request. Once the request is closed the code will be updated .  I'm guessing under a new version ?

k2dp2k, 2024-02-15:
Still the same issue even with the changes made...

19:14:24 INFO [main] connecting mqtt addons@core-mosquitto
Exception ignored in: <function Client.__del__ at 0x7fa78fbf60>
Traceback (most recent call last):
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 874, in __del__
    self._reset_sockets()
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1133, in _reset_sockets
    self._sock_close()
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1119, in _sock_close
    if not self._sock:
           ^^^^^^^^^^
AttributeError: 'Client' object has no attribute '_sock'
19:14:24 ERROR [main] Main loop exception: Client.__init__() missing 1 required positional argument: 'callback_api_version'
19:14:24 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 400, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/app/main.py", line 235, in main
    mqtt_client = paho.Client()
                  ^^^^^^^^^^^^^
TypeError: Client.__init__() missing 1 required positional argument: 'callback_api_version'

19:14:24 INFO [main] exit signal handler... (), {}, shutdown was False
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

k2dp2k, 2024-02-15:
the build in the official hassio repo is wrong... that´s the problem

ZS1FJK, 2024-02-15:
SOOOooo I guess no way for us to fix. Just wait?

k2dp2k, 2024-02-16:
sure install the addon manually... @fl4p has to change this in his addon link:
[image]

ZS1FJK, 2024-02-16:
Thank you. Will give it a try.

ZS1FJK, 2024-02-16:
I have successfully failed . HAHA. Ill just have to wait. Thank you for the program @fl4p and thank you @k2dp2k for information and suggestion.

closed 2024-02-16 by fl4p

philtechsl, 2024-02-16:
Still same issue

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
00:06:11 INFO [bt] BT Discovery:
00:06:16 INFO [bt] BT C8:47:80:05:86:DA               JK_B2A24S15P
00:06:16 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.70
00:06:16 INFO [main] connecting mqtt addons@core-mosquitto
Exception ignored in: <function Client.__del__ at 0x7f97f3bf60>
Traceback (most recent call last):
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 874, in __del__
    self._reset_sockets()
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1133, in _reset_sockets
    self._sock_close()
  File "/app/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 1119, in _sock_close
    if not self._sock:
           ^^^^^^^^^^
AttributeError: 'Client' object has no attribute '_sock'
00:06:16 ERROR [main] Main loop exception: Client.__init__() missing 1 required positional argument: 'callback_api_version'
00:06:16 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 400, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/app/main.py", line 235, in main
    mqtt_client = paho.Client()
                  ^^^^^^^^^^^^^
TypeError: Client.__init__() missing 1 required positional argument: 'callback_api_version'

00:06:16 INFO [main] exit signal handler... (), {}, shutdown was False
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

WolwX, 2024-02-17:
Same here but I don't know why I don't have an update notification.

So I tried to uninstall, including removing repository, and reinstalling but still in 1.76, not in 1.77.

Using in HomeAssistant on Raspberry Pi 3.

Anyway thanks again for your work Fl4p 👍

[image]