fl4p/batmon-ha #198: BMS JBD

rintintin17, 2024-02-17:
Describe the issue here.
After update from 1.76 to 1.77, I have this error

I use JBD BMS

```
Paste log output between BEGIN and END:
BEGIN 
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
20:37:20 INFO [install_bleak] bleak 0.13.1a1 installed, want 0.20.2, running pip3 install bleak==0.20.2
20:37:20 INFO [install_bleak] influxdb installed=False need=False
Looking in indexes: https://pypi.org/simple, https://wheels.home-assistant.io/musllinux-index/
Collecting bleak==0.20.2
  Downloading https://wheels.home-assistant.io/musllinux-index/bleak-0.20.2-py3-none-any.whl (135 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 135.6/135.6 kB 2.9 MB/s eta 0:00:00
Collecting dbus-fast<2.0.0,>=1.83.0 (from bleak==0.20.2)
  Downloading https://wheels.home-assistant.io/musllinux-index/dbus_fast-1.95.2-cp311-cp311-musllinux_1_2_aarch64.whl (1.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.7/1.7 MB 12.4 MB/s eta 0:00:00
Installing collected packages: dbus-fast, bleak
  Attempting uninstall: bleak
    Found existing installation: bleak 0.13.1a1
    Uninstalling bleak-0.13.1a1:
      Successfully uninstalled bleak-0.13.1a1
Successfully installed bleak-0.20.2 dbus-fast-1.95.2
20:37:27 INFO [bt] BT Discovery:
20:37:57 ERROR [main] Error discovering devices: 
20:37:57 INFO [main] Bleak version 0.20.2, BtBackend version bluez-v5.70
20:37:57 ERROR [main] Main loop exception: name 'logger' is not defined
20:37:57 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/main.py", line 359, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/app/main.py", line 147, in main
    bms = construct_bms(dev, verbose_log, devices)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/__init__.py", line 47, in construct_bms
    logger.info('Verbose log for %s enabled', addr)
    ^^^^^^
NameError: name 'logger' is not defined

20:37:57 INFO [main] exit signal handler... (), {}, shutdown was False
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
END
```

atre7, 2024-02-18:
I have this with JBD,JK bms

eliesoldini, 2024-02-18:
I have errors with my jbd after updating grom 1.75 to 1.76, i restored back to 1.75 hoping the next version will fix it.
Today, version 1.77 was available and i updated and still all jbd entities are unavailable
The log of batmon have data showing but entities are all unavailable
Here is part of the error, hope it helps

error (#1): timeout waiting for 3
Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/tasks.py", line 500, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/app/bmslib/__init__.py", line 81, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/tasks.py", line 502, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 237, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/jbd.py", line 75, in fetch
    buf = await self._q(cmd=0x03)
          ^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/jbd.py", line 69, in _q
    return await self._fetch_futures.wait_for(cmd, self.TIMEOUT)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/__init__.py", line 84, in wait_for
    raise asyncio.TimeoutError("timeout waiting for %s" % name)
TimeoutError: timeout waiting for 3
13:38:55 INFO [sampling] Bleak version 0.13.1a1

fl4p, 2024-02-18:
@rintintin17 I just pushed an update that fixes your issue

@eliesoldini your error message is different and not related to this issue
Did a downgrade to 1.75 fix it?

closed 2024-02-18 by fl4p

eliesoldini, 2024-02-18:
Yes, i downgraded and it worked like before
I hope future update will fix the issue

Thanks

rintintin17, 2024-02-18:
All is ok, thanks you