fl4p/batmon-ha #203: JBD times out constantly after a few minutes of successful sampling

coconup, 2024-02-19:
I'm running `batmon` in standalone mode through docker as outlined here (on a Raspberry Pi):

https://github.com/fl4p/batmon-ha/issues/25#issuecomment-1952473454

This is my `options.json` file:
```
{
  "devices": [
    {
      "address": "A4:C1:38:0B:79:22",
      "type": "jbd",
      "alias": "jbd1"
    }
  ],
  "sample_period": 15.0,
  "publish_period": 30.0,
  "concurrent_sampling": false
}
```

Everything runs smoothly for a few minutes, then suddenly I start getting timeouts and not a single successful update goes through:
```
15:41:01 INFO [sampling] Bleak version 0.20.2
15:41:01 ERROR [main] 1 exceptions occurred fetching BMSs
15:41:01 ERROR [main] Error (num 17, max 0) reading BMS: timeout waiting for 3
15:41:01 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/tasks.py", line 500, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/app/bmslib/__init__.py", line 81, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/tasks.py", line 502, in wait_for
    raise exceptions.TimeoutError() from exc

TimeoutError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/./main.py", line 36, in fetch_loop
    if await fn():
       ^^^^^^^^^^
  File "/app/./main.py", line 321, in fn
    raise exceptions[0]
  File "/app/./main.py", line 316, in fn
    await t()
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 237, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/jbd.py", line 75, in fetch
    buf = await self._q(cmd=0x03)
          ^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/jbd.py", line 69, in _q
    return await self._fetch_futures.wait_for(cmd, self.TIMEOUT)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/__init__.py", line 84, in wait_for
    raise asyncio.TimeoutError("timeout waiting for %s" % name)
TimeoutError: timeout waiting for 3
```

This is how the same scenario looks with verbose logging enabled:
```

# ... all looking good above here

[... 136 lines cut ...]
           ^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bleak/__init__.py", line 531, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 124, in connect
    device = await BleakScanner.find_device_by_address(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bleak/__init__.py", line 310, in find_device_by_address
    return await cls.find_device_by_filter(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bleak/__init__.py", line 369, in find_device_by_filter
    async with cls(detection_callback=apply_filter, **kwargs):
  File "/usr/lib/python3.11/site-packages/bleak/__init__.py", line 152, in __aenter__
    await self._backend.start()
  File "/usr/lib/python3.11/site-packages/bleak/backends/bluezdbus/scanner.py", line 191, in start
    self._stop = await manager.active_scan(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/bleak/backends/bluezdbus/manager.py", line 368, in active_scan
    assert_reply(reply)
  File "/usr/lib/python3.11/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.InProgress] Operation already in progress
```

The only way out of this last issue is a reboot or running `systemctl restart bluetooth` on the host. I suspect when the container restarts, some child process is not terminated.

coconup, 2024-02-19:
Update: it appears this is only happening when other processes (in my case in a separate container) access other bluetooth devices.

coconup, 2024-02-20:
@fl4p it appears the issue only occurs when some other service is accessing the bluetooth. In my case, I have another container fetching data from my MPPT controller. Both services (the renogy and `batmon`) poll for data every 15 seconds. 

It seems when I turn the renogy container off, `batmon` works fine and without interruption. If the renogy service is running, `batmon` starts showing the behaviour above very quickly. To answer your first question - it never recovers after the timeout and it appears the only way to regain connectivity to the BMS is to restart the bluetooth in my Raspberry Pi. I suspect therefore that some orphan connection is left behind when the timeout occurs, which keeps the BMS's bluetooth busy.

As an additional piece of info, this is the library I am using to communicate with the MPPT controller: https://github.com/cyrils/renogy-bt

That one never seems to lose connectivity because of `batmon` running alongside it.

peteretep, 2024-02-27:
Aah, I have a similar issue that could be due to the same reason - I'm using the victron hacs integration to access my victron mptt controller. Will disable that integration and try it out, not a big deal as the solar isn't producing much at all up in Shetland. But it would be great to have them both working.

ThomasCr, 2024-10-01:
I was searching yesterday for more information`s about the BMS which is build into my [Vatrer Power 51.2v 100ah LiFePo4](https://youtu.be/1n8ZkgtLBt0?si=UM3RFA2l9P-JnBKO&t=627) battery - it should be [Jiabaida / JBD SP16S020](https://jiabaida-bms.com/products/jbd-smart-bms-8s-16s-6-10s-60a-80a-100a-lifepo4-li-ion-for-lithium-iron-battery-pcb-with-uart-rs485-function-bt-module-with-balance-wires)

I also found maybe some useful informations on that [download page](https://jiabaida-bms.com/pages/download-files): 
[JDB RS485-RS232-UART-Bluetooth-Communication Protocol](https://cdn.shopifycdn.net/s/files/1/0606/5199/5298/files/JDB_RS485-RS232-UART-Bluetooth-Communication_Protocol.pdf?v=1682577935)

fl4p, 2026-09-03:
This was two BlueZ clients on one adapter: the other container's scan/connect cycle tore down batmon's notifications mid-poll (the log shows Notifying: False right after the reply) and the old bleak-over-D-Bus path never recovered from that. Since 2.0x batmon defaults to the `bluek` stack on kernel sockets, which shares the adapter with other processes without going through BlueZ discovery, and stuck connections are torn down and retried. If a current version still stalls with a second BLE client running, open a new issue with a verbose_log around the stall.

closed 2026-09-03 by fl4p