fl4p/batmon-ha #205: DALY BMS problem - TimeoutError: timeout awaiting result 93, got 0/1 responses

WolwX, 2024-02-24:
Hello,

Impossible to get my DalyBMS working.
I followed the troubleshoot steps without success.

I use Batmon with another BMS, a JBD one, and that's ok since last update, but with the DALY one, impossible to get any information since long ago.

My DALY BMS is ok in the Android App, data are ok, so I unlink it to be able to appair with Batmon, it appear on the bluetooth scan, but for some reason, impossible to work.

Daly BMS 4S 12v
Serial SH39F003
Software version 210629

BEGIN 
16:36:14 INFO [sampling] batterie163ah bms debug data: {'r': None, 'buf': {}, 'rx': 17, 'tx': 15}
16:36:14 INFO [sampling] Bleak version 0.20.2
16:36:14 WARNING [sampling] disconnecting DalyBt(96:69:08:05:29:C3,batterie163ah) due to too many errors 21
16:36:16 WARNING [bt] BMS DalyBt(96:69:08:05:29:C3,batterie163ah) disconnected after 14.1s!
16:36:16 ERROR [main] 1 exceptions occurred fetching BMSs
16:36:16 ERROR [main] Error (num 146, max 200) reading BMS: timeout awaiting result 93, got 0/1 responses
16:36:17 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/tasks.py", line 500, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/app/bmslib/__init__.py", line 81, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/tasks.py", line 502, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/bmslib/models/daly.py", line 166, in _q
    sample = await self._fetch_futures.wait_for(command, self.TIMEOUT)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/__init__.py", line 84, in wait_for
    raise asyncio.TimeoutError("timeout waiting for %s" % name)
TimeoutError: timeout waiting for 147

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/main.py", line 36, in fetch_loop
    if await fn():
       ^^^^^^^^^^
  File "/app/main.py", line 319, in fn
    raise exceptions[0]
  File "/app/main.py", line 314, in fn
    await t()
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 237, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/daly.py", line 187, in fetch
    status = await self._fetch_status()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/cache/mem.py", line 160, in _mem_cache_wrapper_async
    return await _inner_wrapper_async(cache_key_obj, args, kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/cache/mem.py", line 131, in _inner_wrapper_async
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/daly.py", line 224, in _fetch_status
    response_data = await self._q(0x93)
                    ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/daly.py", line 169, in _q
    raise TimeoutError(
TimeoutError: timeout awaiting result 93, got 0/1 responses
END

Anyway thanks again for your work :)

temakyiv742, 2024-08-10:
Were you able to solve this problem?

omellin, 2024-08-31:
hello everyone,

I have the exact same issue. It gets the first read right ( maybe a bit more ) right after I press the Daly BMS dongle button, so that tells me the config is correct, then I get the same log.
the connection works perfect with the mobile app.
I tried with my pi4 bluetooth and 2 USB dongle, I get the same results. I didnt try with the WIFI disabled ( because I dont know how to change the config.txt and as the result are the same with external dongle, I have doubt this will change my issue  ).

help please ?

PS : If it helps I'm also trying with UART wired connection with ESPhome, but I cant get UART to work (the ESPhome log doesnt even show that it it trying to send a UART command.)

`18:55:23 ERROR [sampling] battery1 error (#1): timeout awaiting result 90, got 0/1 responses
Traceback (most recent call last):
  File "/app/bmslib/__init__.py", line 81, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/bmslib/models/daly.py", line 166, in _q
    sample = await self._fetch_futures.wait_for(command, self.TIMEOUT)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/__init__.py", line 84, in wait_for
    raise asyncio.TimeoutError("timeout waiting for %s" % name)
TimeoutError: timeout waiting for 144

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 237, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/daly.py", line 189, in fetch
    sample = await self.fetch_soc(sample_kwargs=dict(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/daly.py", line 201, in fetch_soc
    resp = await self._q(0x90)
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/daly.py", line 169, in _q
    raise TimeoutError(
TimeoutError: timeout awaiting result 90, got 0/1 responses
18:55:23 INFO [sampling] battery1 bms debug data: {'r': [bytearray(b'\x01\x0c\xb6\x0c\xb2\x0c\xa9\xe8'), bytearray(b'\x02\x0c\xae\x0c\xb2\x0c\xa9\xe8')], 'buf': {149: [bytearray(b'\x01\x0c\xb6\x0c\xb2\x0c\xa9\xe8'), bytearray(b'\x02\x0c\xae\x0c\xb2\x0c\xa9\xe8')]}, 'rx': 17, 'tx': 15}
18:55:23 INFO [sampling] Bleak version 0.20.2
18:55:23 ERROR [main] 1 exceptions occurred fetching BMSs
18:55:23 ERROR [main] Error (num 1, max 200) reading BMS: timeout awaiting result 90, got 0/1 responses
18:55:23 ERROR [main] Stack: Traceback (most recent call last):
  File "/app/bmslib/__init__.py", line 81, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/bmslib/models/daly.py", line 166, in _q
    sample = await self._fetch_futures.wait_for(command, self.TIMEOUT)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/__init__.py", line 84, in wait_for
    raise asyncio.TimeoutError("timeout waiting for %s" % name)
TimeoutError: timeout waiting for 144

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/main.py", line 36, in fetch_loop
    if await fn():
       ^^^^^^^^^^
  File "/app/main.py", line 319, in fn
    raise exceptions[0]
  File "/app/main.py", line 314, in fn
    await t()
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 237, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/daly.py", line 189, in fetch
    sample = await self.fetch_soc(sample_kwargs=dict(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/daly.py", line 201, in fetch_soc
    resp = await self._q(0x90)
           ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/daly.py", line 169, in _q
    raise TimeoutError(
TimeoutError: timeout awaiting result 90, got 0/1 responses

18:55:24 INFO [sampling] connecting bms DalyBt(C7:7C:03:05:0A:5F,battery1)
18:55:34 INFO [daly] normal connect failed (<class 'TimeoutError'>), connecting with scanner
18:55:44 ERROR [sampling] DalyBt(C7:7C:03:05:0A:5F,battery1) device not found, retry in 2 seconds
18:55:50 INFO [sampling] connecting bms DalyBt(C7:7C:03:05:0A:5F,battery1)
18:56:00 INFO [daly] normal connect failed (<class 'TimeoutError'>), connecting with scanner
18:56:10 ERROR [sampling] DalyBt(C7:7C:03:05:0A:5F,battery1) device not found, retry in 5 seconds
18:56:16 INFO [sampling] connecting bms DalyBt(C7:7C:03:05:0A:5F,battery1)
18:56:26 INFO [daly] normal connect failed (<class 'TimeoutError'>), connecting with scanner`

herijaya, 2024-10-05:
i some error 

6-rc: info: service legacy-services successfully started
18:01:10 INFO [bt] BT Discovery:
18:01:15 INFO [bt] BT 40:17:10:01:5C:9B            DL-401710015C9B
18:01:15 INFO [bt] BT 10:5A:17:2A:9A:A0                         TY
18:01:15 INFO [bt] BT 40:18:05:01:1A:26            DL-401805011A26
18:01:15 INFO [main] Bleak version 0.13.1a1, BtBackend version bluez-v5.76
18:01:15 INFO [main] connecting mqtt addons@core-mosquitto
/app/main.py:196: DeprecationWarning: Callback API version 1 is deprecated, update to latest version
  mqtt_client = paho.Client(paho.CallbackAPIVersion.VERSION1)
18:01:15 INFO [main] Fetching 2 BMS + 0 virtual + 0 others serially, period=1.00s, keep_alive=True
18:01:15 INFO [sampling] connecting bms DalyBt(40:17:10:01:5C:9B,battery3)
18:01:15 INFO [main] mqtt watchdog loop started with timeout 300.0s
18:01:17 WARNING [daly] tried rx/tx/sx uuids 17/15/48: Characteristic with UUID 17 could not be found!
18:01:17 INFO [sampling] connected bms DalyBt(40:17:10:01:5C:9B,battery3)!
18:01:29 ERROR [sampling] battery3 error (#1): timeout awaiting result 93, got 0/1 responses
Traceback (most recent call last):
  File "/usr/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/app/bmslib/__init__.py", line 81, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
               ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/bmslib/models/daly.py", line 166, in _q
    sample = await self._fetch_futures.wait_for(command, self.TIMEOUT)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/__init__.py", line 84, in wait_for
    raise asyncio.TimeoutError("timeout waiting for %s" % name)
TimeoutError: timeout waiting for 147

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/sampling.py", line 237, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/daly.py", line 187, in fetch
    status = await self._fetch_status()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/cache/mem.py", line 160, in _mem_cache_wrapper_async
    return await _inner_wrapper_async(cache_key_obj, args, kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/cache/mem.py", line 131, in _inner_wrapper_async
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/daly.py", line 224, in _fetch_status
    response_data = await self._q(0x93)
                    ^^^^^^^^^^^^^^^^^^^
  File "/app/bmslib/models/daly.py", line 169, in _q
    raise TimeoutError(
TimeoutError: timeout awaiting result 93, got 0/1 responses
18:01:29 INFO [sampling] battery3 bms debug data: {'r': None, 'buf': {}, 'rx': '0000fff1-0000-1000-8000-00805f9b34fb', 'tx': '0000fff2-0000-1000-8000-00805f9b34fb'}

kusznier, 2025-03-28:
Same here, i have 2 daly BMS. Newer one works but this is older and i have similar errors:

00:25:56 ERROR [main] 1 exceptions occurred fetching BMSs
00:25:56 ERROR [main] Error (num 1, max 0) reading BMS: timeout awaiting result 93, got 0/1 responses
00:25:56 ERROR [main] Stack: Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/kusz/batmon-ha/bmslib/__init__.py", line 81, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/kusz/batmon-ha/bmslib/models/daly.py", line 166, in _q
    sample = await self._fetch_futures.wait_for(command, self.TIMEOUT)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kusz/batmon-ha/bmslib/__init__.py", line 84, in wait_for
    raise asyncio.TimeoutError("timeout waiting for %s" % name)
TimeoutError: timeout waiting for 147

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/kusz/batmon-ha/main.py", line 36, in fetch_loop
    if await fn():
       ^^^^^^^^^^
  File "/home/kusz/batmon-ha/main.py", line 319, in fn
    raise exceptions[0]
  File "/home/kusz/batmon-ha/main.py", line 314, in fn
    await t()
  File "/home/kusz/batmon-ha/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kusz/batmon-ha/bmslib/sampling.py", line 237, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/home/kusz/batmon-ha/bmslib/models/daly.py", line 187, in fetch
    status = await self._fetch_status()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kusz/batmon-ha/bmslib/cache/mem.py", line 160, in _mem_cache_wrapper_async
    return await _inner_wrapper_async(cache_key_obj, args, kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kusz/batmon-ha/bmslib/cache/mem.py", line 131, in _inner_wrapper_async
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kusz/batmon-ha/bmslib/models/daly.py", line 225, in _fetch_status
    response_data = await self._q(0x93)
                    ^^^^^^^^^^^^^^^^^^^
  File "/home/kusz/batmon-ha/bmslib/models/daly.py", line 169, in _q
    raise TimeoutError(
TimeoutError: timeout awaiting result 93, got 0/1 responses

00:26:20 ERROR [sampling] DL-4664010406B0 error (#4): timeout awaiting result 93, got 0/1 responses
Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/kusz/batmon-ha/bmslib/__init__.py", line 81, in wait_for
    return await asyncio.wait_for(self._futures.get(name), timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/kusz/batmon-ha/bmslib/models/daly.py", line 166, in _q
    sample = await self._fetch_futures.wait_for(command, self.TIMEOUT)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kusz/batmon-ha/bmslib/__init__.py", line 84, in wait_for
    raise asyncio.TimeoutError("timeout waiting for %s" % name)
TimeoutError: timeout waiting for 147

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/kusz/batmon-ha/bmslib/sampling.py", line 155, in __call__
    s = await self._sample_inner()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kusz/batmon-ha/bmslib/sampling.py", line 237, in _sample_inner
    sample = await bms.fetch()
             ^^^^^^^^^^^^^^^^^
  File "/home/kusz/batmon-ha/bmslib/models/daly.py", line 187, in fetch
    status = await self._fetch_status()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kusz/batmon-ha/bmslib/cache/mem.py", line 160, in _mem_cache_wrapper_async
    return await _inner_wrapper_async(cache_key_obj, args, kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kusz/batmon-ha/bmslib/cache/mem.py", line 131, in _inner_wrapper_async
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kusz/batmon-ha/bmslib/models/daly.py", line 225, in _fetch_status
    response_data = await self._q(0x93)
                    ^^^^^^^^^^^^^^^^^^^
  File "/home/kusz/batmon-ha/bmslib/models/daly.py", line 169, in _q
    raise TimeoutError(
TimeoutError: timeout awaiting result 93, got 0/1 responses
00:26:20 INFO [sampling] DL-4664010406B0 bms debug data: {'r': None, 'buf': {}, 'rx': 17, 'tx': 15}
00:26:20 INFO [sampling] Bleak version 0.13.1a1