fl4p/batmon-ha #207: Missing temps

esther1112, 2024-03-04:
I have 6 jk bms in total, 4 new 2 old  connected via Bluetooth.  Since the batmon update everything works as it should until the last HA update, when I had to turn bleak off for it to run again. My issue is tha I can not see any temperatures for any BMS consistently. They appear for around 10secs and disappear for around 40secs. Is this a mqtt throttling issue or a batmon issue. 
Any advice please

Karatzie, 2024-03-11:
I have also a problem with my two Supervolt batteries.
The temperature values switched very often to not available.
The log dispays parsed temperature values but the entity does not display the value.
I observe this behavior with batmon HA 1.78

rurenkov, 2024-04-12:
The same for me as well for about 3 months,  not critical but issue exists

philtechsl, 2024-05-02:
Hi,

I have the same problem for about 4 months now :(

bbeijl, 2024-08-13:
Hi, Has anybody found a solution for this problem? I have the same problem for over a year now. I am using 1.78.

pulec00, 2025-05-17:
Has anyone found a solution to this problem please?

KrilYura, 2025-06-23:
The issue is still present

fl4p, 2026-09-03:
Fixed in 2.08 (July 2026). The cause was what KrilYura found: temperatures were only republished on a 30 s tick while the entity's expire_after defaults to 20 s, so an unchanged reading expired in between. Temperatures are now published every cycle and unchanged values are re-sent every 10 s, so they stay inside any `expire_values_after` >= 20. If you are on 1.78 or older, update.

closed 2026-09-03 by fl4p