fl4p/batmon-ha #209: Missing data from one of the JK BMS i'm using like SOC en Current

EricGrosfeld, 2024-03-09:
Batmon version: 1.78

Accu bank 1 
JK:BMS: JK_B2A20S20P
Firmware: 11.25
Hardware: 11.X

Accu bank 2 
JK BMS: JK_B2A20S20P
Firmware: 11.261
Hardware: 11.XW

Since a couple of weeks I'm missing some data from my second BMS (Accu bank 2) as you cab see also in the logs. SOC en current for example is not shown anymore.

13:43:53 INFO [sampling] Accu bank 1: BmsSampl(80.1%,U=55.1V,I=63.25A,P=3485W,Q=208/260Ah,mos=29°C)
13:43:53 INFO [sampling] Accu bank 1 volt=[3471,3436,3452,3436,3449,3435,3447,3440,3451,3429,3451,3450,3439,3444,3442,3432] temp=[21.6, 16.75, 0.0, 0.0]
**13:44:51 INFO [sampling] Accu bank 2: BmsSampl(0.0%,U=0.0V,I=0.00A,P=0W,Q=21430/0Ah,mos=0°C)**
13:44:51 INFO [sampling] Accu bank 2 volt=[3416,3416,3412,3415,3416,3417,3415,3406,3412,3418,3418,3406,3421,3410,3412,3415] temp=[0.0, 0.0]

When i connect to the BMS with the Androind JK App, everything seems to be fine.

Below the verbose logging

PLEASE HELP!!

Paste log output between BEGIN and END:
BEGIN 
13:52:21 DEBUG [client] Sending PUBLISH (d0, q0, r0, m1528), 'b'Accu_bank_1/switch/balance'', ... (2 bytes)
13:52:21 INFO [sampling] Accu bank 1: BmsSampl(83.5%,U=55.1V,I=62.82A,P=3463W,Q=217/260Ah,mos=29°C)
13:52:21 DEBUG [client] Sending PUBLISH (d0, q0, r0, m1529), 'b'Accu_bank_1/cell_voltages/1'', ... (5 bytes)
13:52:21 DEBUG [mqtt_util] topic Accu_bank_1/cell_voltages/2 data not changed
13:52:21 DEBUG [client] Sending PUBLISH (d0, q0, r0, m1530), 'b'Accu_bank_1/cell_voltages/3'', ... (5 bytes)
13:52:21 DEBUG [client] Sending PUBLISH (d0, q0, r0, m1531), 'b'Accu_bank_1/cell_voltages/4'', ... (5 bytes)
13:52:21 DEBUG [client] Sending PUBLISH (d0, q0, r0, m1532), 'b'Accu_bank_1/cell_voltages/5'', ... (4 bytes)
13:52:21 DEBUG [mqtt_util] topic Accu_bank_1/cell_voltages/6 data not changed
13:52:21 DEBUG [mqtt_util] topic Accu_bank_1/cell_voltages/7 data not changed
13:52:21 DEBUG [mqtt_util] topic Accu_bank_1/cell_voltages/8 data not changed
13:52:21 DEBUG [mqtt_util] topic Accu_bank_1/cell_voltages/9 data not changed
13:52:21 DEBUG [mqtt_util] topic Accu_bank_1/cell_voltages/10 data not changed
13:52:21 DEBUG [mqtt_util] topic Accu_bank_1/cell_voltages/11 data not changed
13:52:21 DEBUG [mqtt_util] topic Accu_bank_1/cell_voltages/12 data not changed
13:52:21 DEBUG [mqtt_util] topic Accu_bank_1/cell_voltages/13 data not changed
13:52:21 DEBUG [mqtt_util] topic Accu_bank_1/cell_voltages/14 data not changed
13:52:21 DEBUG [client] Sending PUBLISH (d0, q0, r0, m1533), 'b'Accu_bank_1/cell_voltages/15'', ... (5 bytes)
13:52:21 DEBUG [mqtt_util] topic Accu_bank_1/cell_voltages/16 data not changed
13:52:21 DEBUG [mqtt_util] topic Accu_bank_1/cell_voltages/min data not changed
13:52:21 DEBUG [client] Sending PUBLISH (d0, q0, r0, m1534), 'b'Accu_bank_1/cell_voltages/min_index'', ... (2 bytes)
13:52:21 DEBUG [client] Sending PUBLISH (d0, q0, r0, m1535), 'b'Accu_bank_1/cell_voltages/max'', ... (5 bytes)
13:52:21 DEBUG [client] Sending PUBLISH (d0, q0, r0, m1536), 'b'Accu_bank_1/cell_voltages/max_index'', ... (1 bytes)
13:52:21 DEBUG [client] Sending PUBLISH (d0, q0, r0, m1537), 'b'Accu_bank_1/cell_voltages/delta'', ... (5 bytes)
13:52:21 DEBUG [mqtt_util] topic Accu_bank_1/cell_voltages/average data not changed
13:52:21 DEBUG [client] Sending PUBLISH (d0, q0, r0, m1538), 'b'Accu_bank_1/cell_voltages/median'', ... (6 bytes)
13:52:21 INFO [sampling] Accu bank 1 volt=[3474,3437,3455,3437,3450,3436,3450,3440,3452,3427,3454,3451,3440,3446,3445,3432] temp=[21.75, 16.8, 0.0, 0.0]
13:52:21 INFO [sampling] JKBt(C8:47:8C:EC:56:E2,Accu bank 1) times: connect=0.00s fetch=0.17s
13:52:21 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_10_13_B8/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'U\xaa\xeb\x90\x02\xb5X\rY\rX\rY\r\\\r]\rX\rO\rW\r\\\r]\rQ\r_\rT\rV\rY\r\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00X\r\x0f\x00\n\x079\x00:\x007\x007\x006\x006\x005\x006\x005\x005\x004\x005\x006\x006\x004\x004\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')>}, []]
13:52:21 DEBUG [jikong] header, clear buf bytearray(b'')
13:52:21 DEBUG [jikong] bms msg(128) (buf128): 55 aa eb 90 2 b5 58 d 59 d 58 d 59 d 5c d 5d d 58 d 4f d 57 d 5c d 5d d 51 d 5f d 54 d 56 d 59 d 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ff ff 0 0 58 d f 0 a 7 39 0 3a 0 37 0 37 0 36 0 36 0 35 0 36 0 35 0 35 0 34 0 35 0 36 0 36 0 34 0 34 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0

13:52:22 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_10_13_B8/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00J\x01\x00\x00\x00\x00')>}, []]
13:52:22 DEBUG [jikong] bms msg(22) (buf150): 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 4a 1 0 0 0 0

13:52:22 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_10_13_B8/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\x85\xd5\x00\x00\x13s5\x00S\xfa\x00\x00\xdb\x00\xbb\x00\x00\x00\x00\x00F\xf8\x02M\x02\x93\x03\x00\xb0\x9b\x04\x00:\x00\x00\x007\x8a\r\x01d\x00\x00\x00\x1cF\x03\x01\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\x01\x00\x00\x00\x1e\x040\x01\x00\x00\x080?@p\x00\x00\x00Z\x15n\x1b\x00\x01\x00\x01\x00\x05\x00\x00\x1f\xbd \n\x00\x00\x00\x00J\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')>}, []]
13:52:22 DEBUG [jikong] bms msg(128) (buf278): 85 d5 0 0 13 73 35 0 53 fa 0 0 db 0 bb 0 0 0 0 0 46 f8 2 4d 2 93 3 0 b0 9b 4 0 3a 0 0 0 37 8a d 1 64 0 0 0 1c 46 3 1 1 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ff 0 1 0 0 0 1e 4 30 1 0 0 8 30 3f 40 70 0 0 0 5a 15 6e 1b 0 1 0 1 0 5 0 0 1f bd 20 a 0 0 0 0 4a 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0

13:52:22 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_10_13_B8/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\xff\x7f\xdc\x0f\x01\x00\x00\x00\x00\x00\x00&')>}, []]
13:52:22 DEBUG [jikong] bms msg(22) (buf300): 0 0 0 0 0 0 0 0 0 fe ff 7f dc f 1 0 0 0 0 0 0 26

13:52:22 DEBUG [jikong] got response 2 (len300)
13:52:22 DEBUG [mqtt_util] topic Accu_bank_2/soc/total_voltage data not changed
13:52:22 DEBUG [mqtt_util] topic Accu_bank_2/soc/current data not changed
[... 32 lines cut ...]
13:52:22 DEBUG [mqtt_util] topic Accu_bank_2/cell_voltages/max data not changed
13:52:22 DEBUG [mqtt_util] topic Accu_bank_2/cell_voltages/max_index data not changed
13:52:22 DEBUG [mqtt_util] topic Accu_bank_2/cell_voltages/delta data not changed
13:52:22 DEBUG [mqtt_util] topic Accu_bank_2/cell_voltages/average data not changed
13:52:22 DEBUG [mqtt_util] topic Accu_bank_2/cell_voltages/median data not changed
13:52:22 INFO [sampling] Accu bank 2 volt=[3416,3417,3416,3417,3420,3421,3416,3407,3415,3420,3421,3409,3423,3412,3414,3417] temp=[0.0, 0.0]
13:52:22 INFO [sampling] JKBt(C8:47:8C:10:13:B8,Accu bank 2) times: connect=0.00s fetch=0.16s
13:52:22 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_EC_56_E2/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'U\xaa\xeb\x90\x02\x9a\x90\rm\r\x7f\rm\rz\rl\rz\rp\r|\rc\r~\r{\rp\rv\ru\rh\r\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00u\r/\x00\x00\t4\x004\x003\x003\x003\x003\x003\x004\x002\x003\x004\x004\x003\x004\x004\x005\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')>}, []]
13:52:22 DEBUG [jikong] header, clear buf bytearray(b'')
13:52:22 DEBUG [jikong] bms msg(128) (buf128): 55 aa eb 90 2 9a 90 d 6d d 7f d 6d d 7a d 6c d 7a d 70 d 7c d 63 d 7e d 7b d 70 d 76 d 75 d 68 d 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ff ff 0 0 75 d 2f 0 0 9 34 0 34 0 33 0 33 0 33 0 33 0 33 0 34 0 32 0 33 0 34 0 34 0 33 0 34 0 34 0 35 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0

[... cut ...] x006\x006\x004\x004\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')>}, []]
13:52:22 DEBUG [jikong] header, clear buf bytearray(b'')
13:52:22 DEBUG [jikong] bms msg(128) (buf128): 55 aa eb 90 2 b6 58 d 59 d 58 d 59 d 5c d 5d d 58 d 4f d 57 d 5c d 5d d 51 d 5f d 54 d 56 d 59 d 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ff ff 0 0 58 d f 0 a 7 39 0 3a 0 37 0 37 0 36 0 36 0 35 0 36 0 35 0 35 0 34 0 35 0 36 0 36 0 34 0 34 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
END

donulf, 2024-03-20:
Same issue here, JK bms. Today suddenly I've lost balance current, capacity, soc, battery voltage and temperature 1 and 2. Other data work as usual.  I had an old version, 0.something, I updated it to the newest but it still doesn't work. I tried to see the data from JK app and it is all OK, so I think it is a batmon app problem. 

[image]

here my log:

20:54:57 INFO [sampling] battery volt=[3310,3311,3311,3311,3311,3311,3311,3311,3311,3313,3310,3314,3311,3310,3316,3313] temp=[0.0, 0.0]
20:54:57 INFO [sampling] JKBt(C8:47:8C:E9:29:34,battery) times: connect=0.00s fetch=0.11s
20:54:57 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_E9_29_34/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'U\xaa\xeb\x90\x02\xd7\xee\x0c\xef\x0c\xef\x0c\xef\x0c\xef\x0c\xef\x0c\xef\x0c\xef\x0c\xef\x0c\xf1\x0c\xee\x0c\xf2\x0c\xef\x0c\xee\x0c\xf4\x0c\xf1\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00\xef\x0c\x06\x00\x0e\x002\x002\x002\x007\x006\x001\x001\x00H\x00G\x002\x002\x001\x008\x008\x001\x001\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'))>}, []]
20:54:57 DEBUG [jikong] header, clear buf bytearray(b'')
20:54:57 DEBUG [jikong] bms msg(128) (buf128): 55 aa eb 90 2 d7 ee c ef c ef c ef c ef c ef c ef c ef c ef c f1 c ee c f2 c ef c ee c f4 c f1 c 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ff ff 0 0 ef c 6 0 e 0 32 0 32 0 32 0 37 0 36 0 31 0 31 0 48 0 47 0 32 0 32 0 31 0 38 0 38 0 31 0 31 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
20:54:57 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_E9_29_34/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xec\x00\x00\x00\x00\x00'))>}, []]
20:54:57 DEBUG [jikong] bms msg(22) (buf150): 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ec 0 0 0 0 0
20:54:57 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_E9_29_34/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\xf8\xce\x00\x00\x7f\xa6\x05\x00\x87\xe5\xff\xff\xbc\x00\xb8\x00\x00\x00\x08\x00\xd4\xf7\x02N\nz\x03\x00\xd0l\x04\x009\x00\x00\x00\x19\x90\x00\x01d\x00\x00\x00\xc2i\xd7\x00\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0f\x00\x01\x00\x00\x00#\x04\x00\x00 \x00}.B@\x00\x00\x00\x00\xb2\x14\x00\x00\x00\x01\x00\x01\x00\x05\x00\x00B\xfc\x80\x01\x00\x00\x00\x00\xec\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'))>}, []]
20:54:57 DEBUG [jikong] bms msg(128) (buf278): f8 ce 0 0 7f a6 5 0 87 e5 ff ff bc 0 b8 0 0 0 8 0 d4 f7 2 4e a 7a 3 0 d0 6c 4 0 39 0 0 0 19 90 0 1 64 0 0 0 c2 69 d7 0 1 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 f 0 1 0 0 0 23 4 0 0 20 0 7d 2e 42 40 0 0 0 0 b2 14 0 0 0 1 0 1 0 5 0 0 42 fc 80 1 0 0 0 0 ec 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
20:54:57 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_E9_29_34/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\xff\x7f\xdc\x0f\x01\x00\x00\x00\x00\x00\x00\xc7'))>}, []]
20:54:57 DEBUG [jikong] bms msg(22) (buf300): 0 0 0 0 0 0 0 0 0 fe ff 7f dc f 1 0 0 0 0 0 0 c7
20:54:57 DEBUG [jikong] got response 2 (len300)
20:54:57 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_E9_29_34/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'U\xaa\xeb\x90\x02\xd8\xee\x0c\xef\x0c\xef\x0c\xef\x0c\xef\x0c\xef\x0c\xef\x0c\xef\x0c\xef\x0c\xf1\x0c\xee\x0c\xf2\x0c\xef\x0c\xee\x0c\xf2\x0c\xf1\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00\xf0\x0c\x06\x00\x0e\x002\x002\x002\x007\x006\x001\x001\x00H\x00G\x002\x002\x001\x008\x008\x001\x001\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'))>}, []]
20:54:57 DEBUG [jikong] header, clear buf bytearray(b'')
20:54:57 DEBUG [jikong] bms msg(128) (buf128): 55 aa eb 90 2 d8 ee c ef c ef c ef c ef c ef c ef c ef c ef c f1 c ee c f2 c ef c ee c f2 c f1 c 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ff ff 0 0 f0 c 6 0 e 0 32 0 32 0 32 0 37 0 36 0 31 0 31 0 48 0 47 0 32 0 32 0 31 0 38 0 38 0 31 0 31 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
20:54:58 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_E9_29_34/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xec\x00\x00\x00\x00\x00'))>}, []]
20:54:58 DEBUG [jikong] bms msg(22) (buf150): 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 ec 0 0 0 0 0
20:54:58 DEBUG [manager] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C8_47_8C_E9_29_34/service000e/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.sign [... cut ...]
20:54:59 DEBUG [client] Sending PUBLISH (d0, q0, r0, m25578), 'b'battery/cell_voltages/max_index'', ... (2 bytes)
20:54:59 DEBUG [client] Sending PUBLISH (d0, q0, r0, m25579), 'b'battery/cell_voltages/delta'', ... (5 bytes)
20:54:59 DEBUG [mqtt_util] topic battery/cell_voltages/average data not changed
20:54:59 DEBUG [mqtt_util] topic battery/cell_voltages/median data not changed
20:54:59 INFO [sampling] battery volt=[3310,3313,3311,3310,3311,3311,3313,3311,3311,3313,3310,3314,3310,3309,3314,3313] temp=[0.0, 0.0]
20:54:59 INFO [sampling] JKBt(C8:47:8C:E9:29:34,battery) times: connect=0.00s fetch=0.10s

daveprue, 2024-03-25:
Same problem here, my logs are comparable.  It seems that after a few months of running, perhaps a field becomes too large and causes an error in the protocol offsets.  Using Bluetooth app everything looks fine, and using 4.2in Display everything looks fine so it undoubtedly is a BatMon issue.

donulf, 2024-03-29:
> Same problem here, my logs are comparable. It seems that after a few months of running, perhaps a field becomes too large and causes an error in the protocol offsets. Using Bluetooth app everything looks fine, and using 4.2in Display everything looks fine so it undoubtedly is a BatMon issue.

I tried uninstalling and reinstalling the add-on but I have the same problems. A few days ago I also lost two other sensors, cell voltage 18 and 19..

esther1112, 2024-04-09:
I have the same issue. I have 6 BMS linked up in a group. It's all been working fine but now soc etc aren't showing for one BMS. Everything else seems fine

rurenkov, 2024-04-12:
The same for me. Lost voltage and soc. Maybe something else, but these 2 a used in HA for automations.  Did you find any way to fix it?

esther1112, 2024-04-23:
Long story short, I ran out of space on my storage device so I had to rebuild on a bigger one.  Also had to hard reset my BMS that wasn't showing figures, all up and running now

same here since a couple of days (JK B2A20S20P)

boernhard, 2024-04-23:
Thanks for this advice :) What exactly do you mean by "hard reset"? Switch off and on with physical switch? Or reset to factory values by app?

> Long story short, I ran out of space on my storage device so I had to rebuild on a bigger one. Also had to hard reset my BMS that wasn't showing figures, all up and running now Sent from Outlook for Android<https://aka.ms/AAb9ysg>
> […](#)

esther1112, 2024-04-23:
We had to reset through the jk app. Wrote down all the figures first as I have 6.

donulf, 2024-04-23:
> We had to reset through the jk app. Wrote down all the figures first as I have 6. 

sorry but I didn't understand. In which way you reset through the jk app? I don't see any way to do it.. maybe uninstall and reinstall it?

esther1112, 2024-04-23:
On the jk bms app you can reset the BMS to it's original settings.

donulf, 2024-04-23:
> On the jk bms app you can reset the BMS to it's original setting

sorry but I can't find nothing similar to a "reset button" in the jk bms app. The only way you can do something similar to a reset is to press on one of the three presets in the main page. However the counter shows the original time, so it isn't a "real" reset... you do it in that way? or maybe you find another way to do it?

[image]

esther1112, 2024-04-24:
Pressed the lifepo4 button as that's the batteries we have. Will reset all the number to what they were originally. Did that re entered all the numbers to our settings and it started working again properly. That's the way we did it.

1.jpg (view on web)<https://github.com/fl4p/batmon-ha/assets/143497112/8f8c7f62-a3da-4278-aace-d94e1fa67635>

donulf, 2024-04-26:
> > Pressed the lifepo4 button as that's the batteries we have. Will reset all the number to what they were originally. Did that re entered all the numbers to our settings and it started working again properly. That's the way we did it.
> 
> Thank you very much, I'll try it!

I tried, it worked for 12 hours and again went down :-( 

[image]

esther1112, 2024-04-26:
How much memory do u have on Ur storage device IE SD card

333.JPG (view on web)<https://github.com/fl4p/batmon-ha/assets/143497112/65a87bdd-c004-4309-939b-1fbed2030950>

donulf, 2024-04-26:
> How much memory do u have on Ur storage device IE SD card 

I have 28GB free on the internal SSD, and this is the CPU and RAM usage. I don't think is a memory issue..
[image]

esther1112, 2024-04-26:
Ah ok. I'm afraid I don't know of any other fixes

donulf, 2024-04-26:
thank you anyway. Maybe some day @fl4p will read this and he will want to help..

daveprue, 2024-05-09:
I gave up and moved over to ESPHome to replace Batmon

Off-Grid-Garage, 2024-05-09:
Is the dev still around and reading these comments, taking care of the many issues?

boernhard, 2024-12-14:
For easier fix see #235

fl4p, 2025-02-02:
Hi, please update to 1.81 and try either `jk_24s` or `jk_32s` instead of `jk` as BMS `type`

rurenkov, 2025-02-03:
Update and selecting TYPE did not help with my batterry. It still reports SOC=0, Wrong data for # of cycles, and battery capacity. Hovewer it cells values reporting works correct.
What is interesting, I've got one more beatery with exactly the same BMS and it works fine.
Maybe there is some data cache or somethig else... no idea.

[image]

[image]

fl4p, 2025-02-03:
Hm that's odd.
What software version is the BMS?

rurenkov, 2025-02-03:
one more update from just now: if on the battery where Batmon works fine I change the type jk_24s it starts to behave in the same wrong way. but type JK - just works fine with it.
More details about the BMS:

[image]

rurenkov, 2025-02-03:
you can reach me any time if you need more details or assistance with testing.

closed 2025-02-04 by fl4p

rurenkov, 2025-02-04:
> your bms is v11, so it needs to be type jk_32s . leaving type "jk" is just fine since I fixed the detection in the last update

nice. changed it to jk_32s and it seems fine. 
Goood!!!
Thanks, a lot.
Will keep monitoring it.