fl4p/batmon-ha #211: Error in log Unable to open mgnt_socket

Karatzie, 2024-03-11:
Describe the issue here.
I had installed the last batmon HA 1.78 in my campervan HA.
I used two Supervolt batteries.

I observer this error messages during startup the addon on the log.

13:27:50 INFO [main] Power cycle bluetooth hardware
Unable to open mgmt_socket
Unable to open mgmt_socket
13:27:54 INFO [bt] BT Discovery:

After some time, the communication is running.

* Include exact BMS model name
* Query HW and SW numbers from the BMS app
* Enable verbose_log

```
Paste log output between BEGIN and END:
BEGIN 

END
```

Karatzie, 2024-03-22:
Hi.

no answer or comment?

The error is still visible.

BR
Andreas

puetsch, 2024-11-27:
I'm also getting `Unable to open mgmt_socket` — none of the `bluetoothctl` commands seem to do anything.

I think this used to work like a week ago, but rebooting the system doesn't fix it, and neither does unplugging and replugging the BT dongle.

fl4p, 2026-09-03:
That line is bluetoothctl complaining it cannot open the kernel mgmt socket, which the add-on container was not allowed to in 1.78. It only appeared during the `bt_power_cycle` power off/on and the actual power command still went through D-Bus, so it was harmless noise. Current versions run with NET_ADMIN and host networking (needed for the bluek stack), so bluetoothctl gets the socket and the message is gone. Closing.

closed 2026-09-03 by fl4p